zephyrproject-rtos/zephyr #74817: [Coverity CID: 392507] Untrusted divisor in subsys/llext/llext_load.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/llext/llext_load.c

Category: Insecure data handling
Function: `do_llext_load`
Component: Other
CID: [392507](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392507)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/llext/llext_load.c

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=392507

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_TAINTED_SCALAR). ([CWE-369](http://cwe.mitre.org/data/definitions/369.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

pillo79, 2024-06-26:
Issue seems related to "data from llext_peek can't be trusted". Coverity specifically mentions `tainted_data_downcast` on section header data read here:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/llext/llext_load.c#L126

... but I don't see a way to improve the checks (mostly done by checking in seek/read functions). 
@teburd Do you think extending `llext_peek` to also check the size would help? It's a basic sanity check that should really be in the API, IMO, but I'm not sure if it's enough for Coverity.

teburd, 2024-06-26:
> Issue seems related to "data from llext_peek can't be trusted". Coverity specifically mentions `tainted_data_downcast` on section header data read here: [...]

Thanks for looking at this, if llext_peek() taking a size allows for validity checks that make coverity happy that makes sense to me.

pillo79, 2024-06-27:
Digged a little bit more, the CID is really made of 2 issues: 
- one in `llext_link` is the proper unchecked division. and is now already fixed by #74568 which tests for a valid reloc `sh_entsize`;
- the one described above is IMO a false positive; Coverity complains about tainted `sh_size` here:
https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/llext/llext_mem.c#L112
but that number was implicitly validated by the allocation that happened just before:
https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/llext/llext_mem.c#L101-L104
(`sect_alloc` is guaranteed to be bigger than `sh_size`).

pillo79, 2024-11-26:
The cast added by my PR did not make Coverity happy; however, as stated [above](https://github.com/zephyrproject-rtos/zephyr/issues/74817#issuecomment-2194426414), the issue looks like a false positive to me since the values are checked indirectly via the allocations failing.

LLEXT API constraints prevent improving the situation:
* the loader API does not expose the size of the data (might not even be available in streaming contexts), and 
* `llext_peek()` does not accept a size and thus can't protect from access after the end of its buffer.

teburd, 2025-01-16:
Marked as false positive in coverity.

commit 5ce6a750 (2024-07-09): llext: add explicit cast to fix Coverity CID: 392507

From https://mails.dpdk.org/archives/dev/2021-December/231212.html:

    Downcasting a void* to struct aesni_gcm_session* caused the session
    data to be treated as tainted. Removing the void* temporary variable
    and adding a cast avoids this issue.

Try the same approach here to prevent the ldr->sect_hdrs pointer from
being treated as tainted.

May fix #74817.
closed 2025-01-16 by teburd