zephyrproject-rtos/zephyr #74822: [Coverity CID: 363713] Out-of-bounds access in subsys/usb/device_next/class/bt_hci.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/usb/device_next/class/bt_hci.c

Category: Memory - corruptions
Function: `bt_hci_preinit`
Component: USB
CID: [363713](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=363713)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/usb/device_next/class/bt_hci.c#L682

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=363713

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

aescolar, 2024-07-09:
Related to https://github.com/zephyrproject-rtos/zephyr/issues/67106

jfischer-no, 2024-07-12:
Marked false positive, see https://github.com/zephyrproject-rtos/zephyr/issues/74823

closed 2024-07-12 by jfischer-no