zephyrproject-rtos/zephyr #74823: [Coverity CID: 363728] Out-of-bounds access in subsys/usb/device_next/class/bt_hci.c

zephyrbot, 2024-06-22:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/dcf42917c550714d2457947538b9e29d083e872e/subsys/usb/device_next/class/bt_hci.c

Category: Memory - corruptions
Function: `bt_hci_preinit`
Component: USB
CID: [363728](https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=363728)

Details:

https://github.com/zephyrproject-rtos/zephyr/blob/dcf42917c550714d2457947538b9e29d083e872e/subsys/usb/device_next/class/bt_hci.c#L674

Please fix or provide comments in coverity using the link:

https://scan9.scan.coverity.com/#/project-view/29271/12996?selectedIssue=363728

For more information about the violation, check the [Coverity Reference](https://scan9.scan.coverity.com/doc/en/cov_checker_ref.html#static_checker_OVERRUN). ([CWE-119](http://cwe.mitre.org/data/definitions/119.html))

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the MAINTAINERS file.

aescolar, 2024-07-09:
Related to https://github.com/zephyrproject-rtos/zephyr/issues/67106

aescolar, 2024-07-09:
Effectively the same as #74822
@andyross @henrikbrixandersen what do you make of this coverity report?

henrikbrixandersen, 2024-07-09:
> Effectively the same as #74822 @andyross @henrikbrixandersen what do you make of this coverity report?

Looks like it's missing a call to `K_KERNEL_STACK_SIZEOF()` (which takes `K_KERNEL_STACK_RESERVED` into account when calculating the stack size).

aescolar, 2024-07-09:
> Looks like it's missing a call to K_KERNEL_STACK_SIZEOF()

Just to clarify as discussed on the side with @henrikbrixandersen `K_KERNEL_STACK_SIZEOF()` is already used
@andyross any clue?

aescolar, 2024-07-09:
@ceolin

andyross, 2024-07-10:
Unfortunately I can't get coverity's site to load at the moment to see the details.  But FWIW: we had something like this in an earlier coverity cycle.  What was going on there was a legitimate stack overflow in thread initialization.   Runtime overflows aren't visible, but during initialization there are cases where we "make room" on the thread stack for various things, and those can interact via kconfig to produce a stack pointer that ends up out of bounds.   IIRC I fixed the case such that it would fail at runtime, but it wasn't clear how coverity was getting a kconfig that would trip this, as it wouldn't happen in any tests I could find.

tl;dr: this might already be fixed

andyross, 2024-07-10:
Found the fix in commit 840db7858e90cb49c1ba253ce037a44c01549b18, once coverity's site comes back we can check the details vs. the comments in that PR and the linked bugs to see if they match.

jfischer-no, 2024-07-12:
@andyross It is in the setup_thread_stack()
```c

408                stack_obj_size = K_KERNEL_STACK_LEN(stack_size);
1. access_dbuff_const: Calling K_KERNEL_STACK_BUFFER indexes array stack at byte position 591.[show details]
409                stack_buf_start = K_KERNEL_STACK_BUFFER(stack);
    /include/zephyr/kernel/thread_stack.h
    286static inline char *K_KERNEL_STACK_BUFFER(k_thread_stack_t *sym)
    287{
    1. element_addr: Forming the address of the element at index 592U requires buffer (char *)sym to have at least 592 elements.
    288        return (char *)sym + K_KERNEL_STACK_RESERVED;
    289}
    290#ifndef CONFIG_USERSPACE

```

andyross, 2024-07-12:
Bingo, that's indeed it.  (And I let it sit and it eventually came up for me, should have just been patient).  Coverity is throwing the error when the index is computed, but the very next step in that code (line 416) is to check the relevant sizes and toss a runtime panic before the overflow can be used.  A panic isn't great, but it's not the security problem imagined.  Unfortunately Zephyr doesn't have an API for getting a runtime failure back from k_thread_create(), it's supposed to be a heapless/failure-free call.  It just happens that in this case you can build Zephyr in such a way that it demands more reserved space than is available in the stack as passed.

So I'm going to make an executive decision to close this one.  It's essentially a false positive, and a duplicate of the original report.  Longer term we should see about re-arranging the code so that coverity doesn't get confused.

closed 2024-07-12 by andyross

jfischer-no, 2024-07-12:
@andyross Thank you.
#69122 and #74822 have the same coverity report, I will mark them as false positive as well.