zephyrproject-rtos/zephyr #74827: Cellular NRF52 linking error, Driver enabled.

derGraph, 2024-06-22:
**Describe the bug**
Getting an `undefined reference to __device_dts_ord_<N>` error when building an NRF52 cellular app. Using the cellular example with its prj.conf where `CONFIG_MODEM=y`

Platform: XIAO BLE (=NRF5280) + SIM800 Module with edited NRF52840dk drivers: changed .dts file, added modem alias, changed &uart
`&uart1 {
	current-speed = <115200>;
	pinctrl-0 = <&uart1_default>;
	pinctrl-1 = <&uart1_sleep>;
	pinctrl-names = "default", "sleep";
	status = "okay";

	modem: modem {
		compatible = "quectel,bg9x";
		mdm-power-gpios  = <&gpio0 5 GPIO_ACTIVE_HIGH>;
		status = "okay";
	};
};`

**Expected behavior**
The [example ](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/net/cellular_modem) should work as expected.

**Error message**
`[9/14] Linking C executable zephyr\zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map C:/Users/paul/zephyrproject/zephyr/samples/net/cellular_modem/build/zephyr/zephyr_pre0.map
C:\Windows\system32\cmd.exe /C "cd . && C:\Users\paul\zephyr-sdk-0.16.8\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe -g -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr\zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=C:/Users/paul/zephyrproject/zephyr/samples/net/cellular_modem/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/lib/posix/options/liblib__posix__options.a  zephyr/lib/os/zvfs/liblib__os__zvfs.a  zephyr/soc/soc/nrf52840/libsoc__nordic.a  zephyr/subsys/fs/libsubsys__fs.a  zephyr/subsys/random/libsubsys__random.a  zephyr/subsys/modem/libsubsys__modem.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ppp/libsubsys__net__l2__ppp.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/lib/dns/libsubsys__net__lib__dns.a  zephyr/subsys/net/conn_mgr/libsubsys__net__conn_mgr.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/entropy/libdrivers__entropy.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/modem/libdrivers__modem.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/hal_nordic/nrfx/libmodules__hal_nordic__nrfx.a  modules/segger/libmodules__segger.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"c:/users/paul/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp"  -LC:/Users/paul/zephyrproject/zephyr/samples/net/cellular_modem/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m4  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  --specs=picolibc.specs  -lc  -lgcc && C:\Windows\system32\cmd.exe /C "cd /D C:\Users\paul\zephyrproject\zephyr\samples\net\cellular_modem\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E true""
c:/users/paul/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: app/libapp.a(main.c.obj):C:/Users/paul/zephyrproject/zephyr/samples/net/cellular_modem/src/main.c:25: undefined reference to `__device_dts_ord_125'
c:/users/paul/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: zephyr/drivers/modem/libdrivers__modem.a(modem_at_shell.c.obj): in function `at_shell_init':
C:/Users/paul/zephyrproject/zephyr/drivers/modem/modem_at_shell.c:219: undefined reference to `__modem_pipelink_125_user_pipe_0'
c:/users/paul/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: zephyr/drivers/modem/libdrivers__modem.a(modem_at_shell.c.obj): in function `at_shell_open_pipe_handler':
C:/Users/paul/zephyrproject/zephyr/drivers/modem/modem_at_shell.c:137: undefined reference to `__modem_pipelink_125_user_pipe_0'
c:/users/paul/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd.exe: zephyr/drivers/modem/libdrivers__modem.a(modem_at_shell.c.obj): in function `at_shell_attach_chat_handler':
C:/Users/paul/zephyrproject/zephyr/drivers/modem/modem_at_shell.c:147: undefined reference to `__modem_pipelink_125_user_pipe_0'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.`

**Environment:**
 - OS: Windows
 - Toolchain Zephyr SDK

bjarki-andreasen, 2024-06-25:
Hi, the compatible should be `quectel-bg95` :)

bjarki-andreasen, 2024-07-11:
Closing due to inactivity

closed 2024-07-11 by bjarki-andreasen