zephyrproject-rtos/zephyr #74829: mDNS Responder Failure when Query Type contains DNS_CLASS_FLUSH

EricNRS, 2024-06-23:
**Describe the bug**
WIth CONFIG_MDNS_RESPONDER enabled, I see the following recv() errors caused by -EINVAL periodically from the network.

```
[00:25:02.278,000] <err> net_dns_dispatcher: DNS recv error (-22)
[00:26:25.237,000] <err> net_dns_dispatcher: DNS recv error (-22)
[00:30:07.945,000] <err> net_dns_dispatcher: DNS recv error (-22)
```

I tracked it down to Android and iPhone devices issuing DNS requests with the class bit DNS_CLASS_FLUSH set.  This causes the `query_class` check in `dns_unpack_query()` to fail:

https://github.com/zephyrproject-rtos/zephyr/blob/e5c05da9cbda7bc68aa84b796b828767b5eea6d7/subsys/net/lib/dns/dns_pack.c#L593-L595

Can the flush bit be masked off for this implementation to prevent the error?  Is there a better way to handle this to prevent the logs from being spammed with error messages?

**To Reproduce**
Should work on any board. I am using an ESP32-S3 board with the following configuration items set:

```
CONFIG_DNS_RESOLVER=y
CONFIG_MDNS_RESPONDER=y
```

**Expected behavior**
The query either works correctly (the DNS_CLASS_FLUSH is masked) or the failure is silently ignored unless INF of DBG logs are enabled.

**Impact**
Simply an annoyance.

**Environment (please complete the following information):**
Zephyr 3.7-rc1 @ e5c05da9cbda7bc68aa84b796b828767b5eea6d7

@jukkar
@rlubos

jukkar, 2024-06-24:
@EricNRS please try the proposal I sent if it fixes your issue.

EricNRS, 2024-06-24:
@jukkar Yes, the patch fixes the Android / iPhone `<err> net_dns_dispatcher: DNS recv error (-22)` error.

PR #74886 (2024-06-26): net: dns: Do not fail if cache flush bit is set in class
Mask the DNS_CLASS_FLUSH value when checking if the DNS_CLASS_IN is set when unpacking a query.

Fixes #74829
commit 0cac7af4 (2024-06-26): net: dns: Do not fail if cache flush bit is set in class

Mask the DNS_CLASS_FLUSH value when checking if the DNS_CLASS_IN
is set when unpacking a query.

Fixes #74829
closed 2024-06-26 by nashif