zephyrproject-rtos/zephyr #74833: driver: serial: gd32: missing `irq_update` in `usart_gd32_driver_api`

chang196700, 2024-06-23:
**Describe the bug**
In the file `drivers/serial/usart_gd32.c`, the `usart_gd32_driver_api` is missing the `irq_update` function implementation. As a result, some functions such as `modem_backend_uart_isr_irq_handler` (`subsys/modem/backends/modem_backend_uart_isr.c#L93`) cannot be executed as expected

**Expected behavior**
Implement `irq_update`, 

**Impact**
Cannot use modem_backend_uart with gd32 serial driver.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK
 - Commit SHA used 14642091730b26abf49e9ca359fe993ac8f56137

**Additional context**
https://github.com/zephyrproject-rtos/zephyr/blob/14642091730b26abf49e9ca359fe993ac8f56137/drivers/serial/usart_gd32.c#L281

https://github.com/zephyrproject-rtos/zephyr/blob/14642091730b26abf49e9ca359fe993ac8f56137/subsys/modem/backends/modem_backend_uart_isr.c#L93

cameled, 2024-06-25:
@chang196700 Looks stm32 return a constant `1`, can you help to try below patch?
 ```
diff --git a/drivers/serial/usart_gd32.c b/drivers/serial/usart_gd32.c
index 22bf9fb31b8..6eeb38202bc 100644
--- a/drivers/serial/usart_gd32.c
+++ b/drivers/serial/usart_gd32.c
@@ -267,6 +267,12 @@ int usart_gd32_irq_is_pending(const struct device *dev)
 		 usart_interrupt_flag_get(cfg->reg, USART_INT_FLAG_TC)));
 }
 
+static int usart_gd32_irq_update(const struct device *dev)
+{
+	ARG_UNUSED(dev);
+	return 1;
+}
+
 void usart_gd32_irq_callback_set(const struct device *dev,
 				 uart_irq_callback_user_data_t cb,
 				 void *user_data)
@@ -295,6 +301,7 @@ static const struct uart_driver_api usart_gd32_driver_api = {
 	.irq_err_enable = usart_gd32_irq_err_enable,
 	.irq_err_disable = usart_gd32_irq_err_disable,
 	.irq_is_pending = usart_gd32_irq_is_pending,
+	.irq_update = usart_gd32_irq_update,
 	.irq_callback_set = usart_gd32_irq_callback_set,
 #endif /* CONFIG_UART_INTERRUPT_DRIVEN */
 };
```

chang196700, 2024-06-26:
@cameled  ok, I'll try it.

nandojve, 2024-08-26:
Hi @chang196700 ,
Is there any update from your side about this issue?

chang196700, 2024-08-30:
Apologies for my recent time constraints. I plan to update and submit a PR within the forthcoming two weeks.

PR #79757 (2024-10-16): driver: serial: gd32: add irq_update
Add missing api irq_update for gd32.

Close #74833
closed 2024-10-16 by henrikbrixandersen