zephyrproject-rtos/zephyr #74841: tests: mimxrt1050_evk : samples/sensor/qdec : build failure

hakehuang, 2024-06-24:
**Describe the bug**
samples/sensor/qdec build failure

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 mimxrt1050_evk
 - What have you tried to diagnose or workaround this issue?
 git bisect
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 
```
e6d8ed2e0d99bb25e9c4002b2a14c8c3a15e9e9b is the first bad commit
commit e6d8ed2e0d99bb25e9c4002b2a14c8c3a15e9e9b
Author: Gerson Fernando Budke <nandojve@gmail.com>
Date:   Wed Jun 19 21:49:29 2024 +0200

    samples: sensor: qdec: Fix wrong harness
    
    The #73619 which fixes the atmel,sam-tc-qdec sensor implementation
    introduced agregression on the sample test removing the harness
    console. That changed the test to use the default ztest behaviour
    which is incompatible with the test outpout. This restore the
    harness to the default console value.
    
    Fixes: #74531
    
    Signed-off-by: Gerson Fernando Budke <nandojve@gmail.com>

 samples/sensor/qdec/sample.yaml | 13 +++++++++++++
 1 file changed, 13 insertions(+)
```

**To Reproduce**
Steps to reproduce the behavior:
```
west twister -p mimxrt1050_evk -s samples/sensor/qdec/sample.sensor.qdec_sensor
```

**Expected behavior**
build pass

**Impact**
sensor supporting

**Logs and console output**

```
-- Board: mimxrt1050_evk, Revision: hyperflash, qualifiers: mimxrt1052
-- Found host-tools: zephyr 0.16.8 (/opt/toolchains/zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/opt/toolchains/zephyr-sdk-0.16.8)
-- Found Dtc: /opt/toolchains/zephyr-sdk-0.16.8/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/boards/nxp/mimxrt1050_evk/mimxrt1050_evk.dts
-- Found devicetree overlay: /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/boards/nxp/mimxrt1050_evk/mimxrt1050_evk_mimxrt1052_hyperflash.overlay
-- Found devicetree overlay: /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/samples/sensor/qdec/boards/mimxrt1050_evk.overlay
devicetree error: /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/samples/sensor/qdec/boards/mimxrt1050_evk.overlay:41 (column 1): parse error: undefined node label 'touch_controller'
-- In: /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/mimxrt1050_evk/samples/sensor/qdec/sample.sensor.qdec_sensor/zephyr, command: /usr/bin/python;/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/scripts/dts/gen_defines.py;--dts;/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/mimxrt1050_evk/samples/sensor/qdec/sample.sensor.qdec_sensor/zephyr/zephyr.dts.pre;--dtc-flags;'';--bindings-dirs;/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/dts/bindings;--header-out;/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/mimxrt1050_evk/samples/sensor/qdec/sample.sensor.qdec_sensor/zephyr/include/generated/zephyr/devicetree_generated.h.new;--dts-out;/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/mimxrt1050_evk/samples/sensor/qdec/sample.sensor.qdec_sensor/zephyr/zephyr.dts.new;--edt-pickle-out;/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/twister-out/mimxrt1050_evk/samples/sensor/qdec/sample.sensor.qdec_sensor/zephyr/edt.pickle;--edtlib-Werror;--vendor-prefixes;/home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at cmake/modules/dts.cmake:296 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  cmake/modules/zephyr_default.cmake:132 (include)
  share/zephyr-package/cmake/ZephyrConfig.cmake:75 (include)
  share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  cmake/package_helper.cmake:126 (find_package)
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v3.7.0-rc1-243-gaf57038bb053

pdgendt, 2024-06-24:
CC @nandojve

nandojve, 2024-06-24:
Hi @hakehuang, @pdgendt ,

On the log there is an specific message about:
`devicetree error: /home/jenkins/agent/workspace/bitbucket_build_platform/zephyr/samples/sensor/qdec/boards/mimxrt1050_evk.overlay:41 (column 1): parse error: undefined node label 'touch_controller'"
`

Since this is a building error I would say that e6d8ed2 is not the root of the problem. The e6d8ed2 enables all devices with `qdec0` to catch any possible build error and maybe this is why you can see this now.

dleach02, 2024-06-24:
@EmilioCBen please add this to the 3.7 bug sweep

dleach02, 2024-06-25:
@hakehuang this test fails in v3.6.0. Something changed in your test environment and I suspect it is running test cases outside of twister or something because the sample.yaml for this test case is only for nucleo_f401re:

```
tests:
  sample.sensor.qdec_sensor:
    platform_allow: nucleo_f401re
    harness_config:
      fixture: fixture_mech_encoder

```

Also, the bisect is wrong, which I suspect is automated given that it fails with v3.6.0.

hakehuang, 2024-06-25:
> Also, the bisect is wrong, which I suspect is automated given that it fails with v3.6.0.
@dleach02 this case is never enabled before e6d8ed2e0d99bb25e9c4002b2a14c8c3a15e9e9b, so bisect detect this issue when NXP platform is enabled, as  @nandojve comments:
> Since this is a building error I would say that [e6d8ed2](https://github.com/zephyrproject-rtos/zephyr/commit/e6d8ed2e0d99bb25e9c4002b2a14c8c3a15e9e9b) is not the root of the problem. The [...]

see below testcase.yml.

```
tests:
  sample.sensor.qdec_sensor:
    filter: dt_alias_exists("qdec0")
    harness_config:
      type: multi_line
      ordered: true
      regex:
        - "Quadrature decoder sensor test"
        - "Position = (.*) degrees"
```

dleach02, 2024-06-25:
Okay, it was originally broken with a PR #56128 when the touch_controller node name was deprecated. But then further, @danieldegrasse removed the node completely since it was associated with a display to go in favor of shields defining the display in PR #72175. Then I guess @nandojve PR to enable devices with qdec0 to be included in the test set things up for twister to find it...

PR #74997 (2024-06-26): samples: sensor: qdec: fix mimxrt1050_evk build failure.
Recent PR turned on test case for twister which revealed a bug with building the test case for RT1050 EVK. The touch_controller DTS node was removed a by commit 57ad325e and further deprecated by bde07beb5 when the platform was converted to tread displays as shields.

Fixes #74841
closed 2024-06-26 by nashif