zephyrproject-rtos/zephyr #74843: Posix API `strnlen` declaration not found in native_posix/native_sim

LingaoM, 2024-06-24:
https://github.com/zephyrproject-rtos/zephyr/actions/runs/9638989345/job/26580638001?pr=74706

``` C
/__w/zephyr/zephyr/subsys/net/lib/http/http_server_http1.c
/__w/zephyr/zephyr/subsys/net/lib/http/http_server_http1.c: In function ‘dynamic_get_req’:
/__w/zephyr/zephyr/subsys/net/lib/http/http_server_http1.c:104:35: error: implicit declaration of function ‘strnlen’; did you mean ‘strlen’? [-Werror=implicit-function-declaration]
  104 |                                   strnlen(http_response,                \
      |                                   ^~~~~~~
/__w/zephyr/zephyr/subsys/net/lib/http/http_server_http1.c:116:15: note: in expansion of macro ‘SEND_RESPONSE’
  116 |         ret = SEND_RESPONSE(RESPONSE_TEMPLATE_CHUNKED,
      |               ^~~~~~~~~~~~~
cc1: all warnings being treated as errors
ninja: build stopped: subcommand failed.
```

I'm also checkout zephyr-3.5.0, this branch build passed, and also compare build command:

**zephyr-3.5.0**
```C
"command": "/usr/bin/gcc -DKERNEL -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -DK_HEAP_MEM_POOL_SIZE=0 -D__ZEPHYR__=1 -I/home/mi/works/github/zephyrproject/zephyr/kernel/include
```
**main:**
```C
"command": "/usr/bin/gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=0 -D__ZEPHYR__=1 -I/home/mi/works/github/zephyrproject/zephyr/kernel/include
```

LingaoM, 2024-06-24:
**ADD:**

The string.h header file also included.
``` C
#include <string.h>
````
And i see the strnlen declared in `/usr/include/string.h`
```C
#ifdef	__USE_XOPEN2K8
/* Find the length of STRING, but scan at most MAXLEN characters.
   If no '\0' terminator is found in that many characters, return MAXLEN.  */
extern size_t strnlen (const char *__string, size_t __maxlen)
     __THROW __attribute_pure__ __nonnull ((1));
#endif
```
It is completely confirmed that `__USE_XOPEN2K8` macro is not defined in the current **main** branch, but it is defined in the `zephyr-3.5.0` branch.

aescolar, 2024-06-24:
Related to #68278
The issue is that the a file that uses `strnlen()` needs to ensure ` _POSIX_C_SOURCE` is set

jukkar, 2024-06-24:
I am wondering why CI is failing now, we have had http_server already in tree for some time.

LingaoM, 2024-06-24:
> Related to #68278 The issue is that a file that uses `strnlen()` needs to ensure ` _POSIX_C_SOURCE` is set as that function is an extension whose prototype only needs to be exposed if that macro is set.

@aescolar Could you push hotfix for this issuer, please don't block CI.

aescolar, 2024-06-24:
> I am wondering why CI is failing now, we have had http_server already in tree for some time.

native_posix is not anymore a default test platform. But that PR which triggered this test built for native_posix was modifying a native_posix board file.

PR #74858 (2024-06-24): Fix implicit function declaration warning for subsys/net/lib/http/http_server_http1.c
    subsys/net/lib/http: Set feature macro as required
    
    This file uses strnlen() but the C library is not require to expose its prototype unless
    _POSIX_C_SOURCE is defined.
    So let's define it to avoid an implicit function declaration warning.

-------

    tests/net/lib/http_server/hpack: Exclude native_posix
    
    This test explicitly enables the POSIX_API which is not compatible with native_posix.
    Let's exclude this target.
    
Fixes #74843
closed 2024-06-24 by aescolar