zephyrproject-rtos/zephyr #74855: AWS IoT error <err> aws: Failed to read MQTT input: -

z3r0g2017, 2024-06-24:
Hi, 
While trying out the AWS iot sample I get the following error.
~/zephyrproject/zephyr$ west build --build-dir /tmp/aws_iot -t run
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version zephyr-v1.0.0-0-g31d4e0e-dirty-20200714_234759-fv-az50-zephyr)
Booting from ROM..
*** Booting Zephyr OS build v3.7.0-rc1-106-g012520d2d7e8 ***
[00:00:00.140,000] <dbg> aws: sntp_sync_time: Acquired time from NTP server: 1719217191
[00:00:00.150,000] <inf> aws: Resolved: 18.195.145.95:8883
[00:00:00.620,000] <dbg> aws: mqtt_event_cb: MQTT event: CONNACK [0] result: 0
[00:00:00.620,000] <inf> aws: Subscribing to 1 topic(s)
[00:00:00.670,000] <dbg> aws: mqtt_event_cb: MQTT event: SUBACK [7] result: 0
[00:00:00.670,000] <inf> aws: PUBLISHED on topic "zephyr_sample/downlink" [ id: 1 qos: 0 ], payload: 13 B
[00:00:00.670,000] <dbg> aws: publish_message: Published payload:
                              7b 22 63 6f 75 6e 74 65  72 22 3a 30 7d          |{"counte r":0}   
[00:00:00.680,000] <dbg> aws: mqtt_event_cb: MQTT event: DISCONNECT [1] result: -128
[00:00:00.680,000] <err> aws: Failed to read MQTT input: -128
[00:00:00.690,000] <inf> aws: mbedTLS heap usage: MAX 47196/65536 (79) CUR 0 (0)
[00:00:01.710,000] <inf> aws: Resolved: 52.57.148.146:8883
[00:00:02.180,000] <dbg> aws: mqtt_event_cb: MQTT event: CONNACK [0] result: 0
[00:00:02.180,000] <inf> aws: Subscribing to 1 topic(s)
[00:00:02.220,000] <dbg> aws: mqtt_event_cb: MQTT event: SUBACK [7] result: 0
[00:00:02.220,000] <inf> aws: PUBLISHED on topic "zephyr_sample/downlink" [ id: 2 qos: 0 ], payload: 13 B
[00:00:02.220,000] <dbg> aws: publish_message: Published payload:
                              7b 22 63 6f 75 6e 74 65  72 22 3a 30 7d          |{"counte r":0}   
[00:00:02.250,000] <dbg> aws: mqtt_event_cb: MQTT event: PUBLISH [2] result: 0
[00:00:02.250,000] <inf> aws: RECEIVED on topic "zephyr_sample/data" [ id: 40688 qos: 0 ] payload: 45 / 4096 B
[00:00:02.250,000] <dbg> aws: handle_published_message: Received payload:
                              7b 0a 20 20 22 6d 65 73  73 61 67 65 22 3a 20 22 |{.  "mes sage": "
                              48 65 6c 6c 6f 20 66 72  6f 6d 20 41 57 53 20 49 |Hello fr om AWS I
                              6f 54 20 63 6f 6e 73 6f  6c 65 22 0a 7d          |oT conso le".}   
[00:00:02.250,000] <inf> aws: PUBLISHED on topic "zephyr_sample/downlink" [ id: 3 qos: 0 ], payload: 13 B
[00:00:02.250,000] <dbg> aws: publish_message: Published payload:
                              7b 22 63 6f 75 6e 74 65  72 22 3a 31 7d          |{"counte r":1}   
[00:00:02.250,000] <dbg> aws: mqtt_event_cb: MQTT event: DISCONNECT [1] result: -128
[00:00:02.250,000] <err> aws: Failed to read MQTT input: -128
[00:00:02.250,000] <inf> aws: mbedTLS heap usage: MAX 47196/65536 (79) CUR 0 (0)
[00:00:03.270,000] <inf> aws: Resolved: 3.74.74.111:8883
[00:00:03.740,000] <dbg> aws: mqtt_event_cb: MQTT event: CONNACK [0] result: 0
[00:00:03.740,000] <inf> aws: Subscribing to 1 topic(s)
[00:00:03.780,000] <dbg> aws: mqtt_event_cb: MQTT event: SUBACK [7] result: 0
[00:00:03.790,000] <inf> aws: PUBLISHED on topic "zephyr_sample/downlink" [ id: 4 qos: 0 ], payload: 13 B
[00:00:03.790,000] <dbg> aws: publish_message: Published payload:
                              7b 22 63 6f 75 6e 74 65  72 22 3a 31 7d          |{"counte r":1}   
[00:00:03.820,000] <dbg> aws: mqtt_event_cb: MQTT event: PUBLISH [2] result: 0
[00:00:03.820,000] <inf> aws: RECEIVED on topic "zephyr_sample/data" [ id: 40688 qos: 0 ] payload: 45 / 4096 B
[00:00:03.820,000] <dbg> aws: handle_published_message: Received payload:
                              7b 0a 20 20 22 6d 65 73  73 61 67 65 22 3a 20 22 |{.  "mes sage": "
                              48 65 6c 6c 6f 20 66 72  6f 6d 20 41 57 53 20 49 |Hello fr om AWS I
                              6f 54 20 63 6f 6e 73 6f  6c 65 22 0a 7d          |oT conso le".}   
[00:00:03.820,000] <inf> aws: PUBLISHED on topic "zephyr_sample/downlink" [ id: 5 qos: 0 ], payload: 13 B
[00:00:03.820,000] <dbg> aws: publish_message: Published payload:
                              7b 22 63 6f 75 6e 74 65  72 22 3a 32 7d          |{"counte r":2}   
[00:00:03.820,000] <dbg> aws: mqtt_event_cb: MQTT event: DISCONNECT [1] result: -128
[00:00:03.820,000] <err> aws: Failed to read MQTT input: -128
[00:00:03.820,000] <inf> aws: mbedTLS heap usage: MAX 47196/65536 (79) CUR 0 (0)

Can you please help me fix this ?
TIA.

rlubos, 2024-06-24:
`mqtt_input()` return -ENOTCONN (-128) in case underlying TCP/TLS connection was closed. If you're using qemu_86, it's fairly simple to monitor network traffic with wireshark, just attach it to the corresponding network interface. That should shed some light on whether is just the peer closing the connection for whatever reason, or some bug in the networking stack.

z3r0g2017, 2024-06-24:
I got it to work with qos1. Which is not the default qos0 when run with qemu_x86. What is the correct config to use in the pro.conf or the KConfig ?

rlubos, 2024-06-25:
> What is the correct config to use for qos1 in the prj.conf or the KConfig ?

There's `CONFIG_AWS_QOS` defined in the sample's Kconfig file.

> by handling the mqtt disconnect event which is a noop in the original source code and using a return instead.

Looking at the sample's code, this change should literally make no difference, the callback returns anyway just after getting `MQTT_EVT_DISCONNECT` event.

rlubos, 2024-06-27:
@z3r0g2017 Do we need to keep this open, given you got the sample to work? I don't know what the bug would be here.

If you still need help here, please provide wireshark logs showing when the client disconnects. Otherwise, please close the issue.

carlescufi, 2024-07-03:
Closing since the issue seemed to be on the reporter's side

closed 2024-07-03 by carlescufi