zephyrproject-rtos/zephyr #74880: stm32u5: Can't run TF-M samples

erwango, 2024-06-24:
**Describe the bug**

It is not possible to run TF-M samples on b_u585i_iot02a due to the 2 following issues:
- ST-SAFE should be disabled
- Regression script should not use HW reset 

**To Reproduce**
`$ west build -p always -b b_u585i_iot02a//ns samples/tfm_integration/tfm_ipc/`

**Expected behavior**
We should be able to flash an run TF-M samples on `b_u585i_iot02a//ns`

**Impact**
Blocker to use TF-M on this target

**Additional context**
Fixes are identified and provided in the following PR: https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/106
For Zephyr 3.7 release, however, these 2 issues should actually be fixed in TF-M LTS (2.1.1) and Zephyr manifest updated to track this version when available.
This ticket is created to track this change.

erwango, 2024-06-24:
@aescolar, @ithinuel FYI

ithinuel, 2024-06-25:
@erwango I see the fixes are actually merged in TF-M.
Would you mind opening a PR on https://github.com/zephyrproject-rtos/trusted-firmware-m/ (or reopening https://github.com/zephyrproject-rtos/trusted-firmware-m/pull/106) cherry picking the commits from TF-M (such that it has the same change & commit message) ?

erwango, 2024-06-26:
@ithinuel Thanks, PR has been reopened and updated

erwango, 2024-07-04:
Closing as https://github.com/zephyrproject-rtos/zephyr/pull/74850 was merged

closed 2024-07-04 by erwango