zephyrproject-rtos/zephyr #74881: drivers: ethernet: lan865x: Does not work with IPv6

spe-ciellt, 2024-06-24:
Or "I don't  get it to work with IPv6"

**Describe the bug**
Enabling and setting up this device driver for an IPv4 network works. Simple test include ping between the device and a PC using the Microchip T1S dongle.
Switching the setup to IPv6 the pinging does not work (no network operation at all). I use link local addresses (which are automatically defined from MAC address). Pinging from the PC host gives no response and no messages on the Zephyr end. Pinging from the Zephyr end gives the following error messages on the console:
```
 [00:00:13.511,000] <err> net_pkt: net_pkt_get_data: Uncontiguous data cannot be linearized
 [00:00:13.511,000] <err> net_ipv6: handle_na_input: DROP: NULL NA header
 [00:00:13.511,000] <err> net_icmpv6: net_icmpv6_input: ICMPv6 handling failure (-5)
```

The Zephyr platform used is Nucleo G474 and Mikroe Two-wire interface.
Zephyr version is 3.6.0.

I have tried various buffer sizes and amounts. One setting that comes from the driver authors example is 
```
CONFIG_NET_BUF_DATA_SIZE=64
```
which makes sense since the buffer sizes to and from the chip is 64 bytes. After a bit of chat on the Discord channel, someone pointed out that an IPv6 header does not fit in 64 bytes, so there must be some gluing of data. Which maybe fails, I do not know?

**To Reproduce**
lan865x.overlay:
```
// SCK=PC10=CN7:1 | MISO=PC11=CN7:2 | MOSI=PC12=CN7:3 | /CS=PA15=CN7:17
&spi3 {
    pinctrl-0 = <&spi3_nss_pa15 &spi3_sck_pc10 &spi3_miso_pc11 &spi3_mosi_pc12>;
    pinctrl-names = "default";
    status = "okay";

    lan8650: lan8650@0 {
        compatible = "microchip,lan865x";
        reg = <0>;
        rst-gpios = <&gpioc 2 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>;
        int-gpios = <&gpioc 3 (GPIO_ACTIVE_LOW | GPIO_PULL_UP)>;
        local-mac-address = [ 00 01 f0 12 34 56 ];
        spi-max-frequency = <25000000>;
        plca-enable;
        plca-node-id = <2>;
        plca-node-count = <8>;
        plca-burst-count = <0>;
        plca-burst-timer = <0x80>;
        plca-to-timer = <0x20>;
        status = "okay";
    };
};
```
The network-overlay.conf
```
##########################
## Enable networking shell
##########################
CONFIG_NET_SHELL=y

###########################
## Networking configuration
###########################
# Generic networking options
CONFIG_NETWORKING=y
CONFIG_NET_L2_ETHERNET=y
CONFIG_NET_L2_ETHERNET_MGMT=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=n
#CONFIG_NET_ARP=y # Part of IPv4
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y

CONFIG_NET_CONNECTION_MANAGER=y

CONFIG_NET_CONFIG_AUTO_INIT=y

# Enable DHCP for IPv4 (IPv6 address must still be defined)
# CONFIG_NET_DHCPV4=n

# IP address options
CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=4
CONFIG_NET_MAX_CONTEXTS=10

# For bigger networks aka stacked version (default is 8)
CONFIG_NET_IPV6_MAX_NEIGHBORS=16

# More connectetions, we have at least 5 in semi idle (default 4)
CONFIG_NET_MAX_CONN=8

# Network buffers
CONFIG_NET_PKT_TX_COUNT=20
CONFIG_NET_BUF_TX_COUNT=50

# Network logging
CONFIG_NET_BUF_LOG=y
CONFIG_NET_LOG=y
CONFIG_NET_STATISTICS=y

#CONFIG_NET_BUF_DATA_SIZE=64

# Network manual settings
CONFIG_NET_CONFIG_SETTINGS=y

####################
## Ethernet settings
####################
CONFIG_ETH_DRIVER=y
CONFIG_ETH_LAN865X=y

CONFIG_ETHERNET_LOG_LEVEL_DBG=y
```

**Expected behavior**
A working IPv6 implementation.

**Impact**
Since this is part of a testing phase, it has no immediate  impact. But if we chose to go for T1S (we have other products using T1S), then this is a show stopper.

**Logs and console output**
```
PROMPT> net ping fe80::7a6b:7fb7:4d8c:3f03
PING fe80::7a6b:7fb7:4d8c:3f03
 [00:00:10.231,000] <err> net_pkt: net_pkt_get_data: Uncontiguous data cannot be linearized
 [00:00:10.231,000] <err> net_ipv6: handle_na_input: DROP: NULL NA header
 [00:00:10.231,000] <err> net_icmpv6: net_icmpv6_input: ICMPv6 handling failure (-5)
```

**Environment (please complete the following information):**
 - OS: Linux Ubuntu
 - Toolchain: Zephyr SDK 0.16.3
 - On the PC end I use the Microchip Dongle on a VM running with the latest 6.6 kernel with the device driver compiled in. Works excellent with our other T1S products,

**Additional context**
I wrote as an experiment about two years ago a driver for this chip that used the Microchip OA TC6 library. That one worked with both IPv4 and IPv6, so it should be possible despite limits in the buffer sizes. But it was not something to be proud of and is today completely scrapped.

aescolar, 2024-06-25:
@lmajewski
CC @pdgendt @decsny

pdgendt, 2024-06-25:
Hi @spe-ciellt can you try with #74951 and set `CONFIG_ETH_LAN865X_CHUNK_SIZE` to a higher value? I haven't tested myself, but I assume the issue is that currently the 64 byte chunk size is insufficient for contiguous ICMPv6 headers.

pdgendt, 2024-06-25:
@spe-ciellt do you enable `CONFIG_NET_HEADERS_ALWAYS_CONTIGUOUS`? If so, try turning it off, as the driver can't guarantee this.

spe-ciellt, 2024-06-25:
> Hi @spe-ciellt can you try with #74951 and set `CONFIG_ETH_LAN865X_CHUNK_SIZE` to a higher value? I haven't tested myself, but I assume the issue is that currently the 64 byte chunk size is insufficient for contiguous ICMPv6 headers.
> 
> EDIT: probably this doesn't help as fixed sizes are used by the controller..

Hi @pdgendt, I tested the patch, and it is, as you say, not working when increasing to for instance 128 bytes. The chip only supports 32 and 64 byte chunks. 64 bytes are default and when you go to 32 bytes chunks there are register bits in the chip you must change as well.
I think there is some gluing together chunks that gets an offset by something error and makes it fail IPv6.

pdgendt, 2024-06-25:
> Hi @pdgendt, I tested the patch, and it is, as you say, not working when increasing to for instance 128 bytes. The chip only supports 32 and 64 byte chunks. 64 bytes are default and when you go to [...]

As expected, I've updated the PR to be able to configure the first buffer size, could you try again with a value for `CONFIG_ETH_LAN865X_HEADER_BUFFER_SIZE` that is sufficient for the ICMPv6 header?

lmajewski, 2024-06-25:
@pdgendt  @spe-ciellt - I would need to check, but IIRC there is configuration of the LAN865x set to have the start of frame always aligned to the beginning of the chunk. Maybe this is the case as IPv6 doesn't fit into 64B chunk?

spe-ciellt, 2024-06-25:
> @pdgendt @spe-ciellt - I would need to check, but IIRC there is configuration of the LAN865x set to have the start of frame always aligned to the beginning of the chunk. Maybe this is the case as IPv6 doesn't fit into 64B chunk?

I have dug up my implementation using the Microchip oatc6 driver. There are some buffer sizes that I have tuned up from the original implementation. You also have the magic value 12 in your implementation. I can not really find any reference in Microchip code.

Is it OK if I refer to some values in their code or are you doing a "clean room" implementation?

spe-ciellt, 2024-06-25:
> As expected, I've updated the PR to be able to configure the first buffer size, could you try again with a value for `CONFIG_ETH_LAN865X_HEADER_BUFFER_SIZE` that is sufficient for the ICMPv6 header?

Tested and did not make a difference.

One interesting thing is that the chip does not generate an IRQ when it receives IPv6 pings, but when I send IPv4 pings an IRQ is triggered and reads out a chunk.

pdgendt, 2024-06-26:
Sorry, I don't have any HW to test this then, maybe @lmajewski can propose a proper fix.

lmajewski, 2024-06-26:
> I have dug up my implementation using the Microchip oatc6 driver. There are some buffer sizes that I have tuned up from the original implementation. You also have the magic value 12 in your [...]

The problem was/is (?) that Microchip's OA_TC6 driver implementation has improper license to be used with Zephyr. I've explicitly reach the author:
https://github.com/MicrochipTech/oa-tc6-lib/issues/13#event-12352950634
and asked if they plan or can release the code on Apache2 license - my original plan was to just "reuse" this library with some Zephyr specific "glue". The reply from October 2023 was that Microchip is working on the "License Adjustment".

As there was no reply - I've decided to do the "clean room" implementation with proper licensing. Moreover, the driver is **very** simple (half-duplex, 64bit chunk transmission at a time) - I had many issues with the HW revision version of LAN8651 and its setup.
Some design decisions were presented/discussed:
https://github.com/MicrochipTech/oa-tc6-lib/issues/13#event-12352950634

Recently there was also discussion about possible improvements:
https://github.com/zephyrproject-rtos/zephyr/issues/74393

Regarding the driver - you may want to check if removing  `OA_CONFIG0_RFA_ZARFE`
from: https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/ethernet/eth_lan865x.c#L437 helps.

This was another simplification - to allow the LAN8651 do as much as possible "work" internally.

spe-ciellt, 2024-07-02:
> Regarding the driver - you may want to check if removing `OA_CONFIG0_RFA_ZARFE` from: https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/ethernet/eth_lan865x.c#L437 helps.
> 
> This was another simplification - to allow the LAN8651 do as much as possible "work" internally.

I tested to remove that bit and it didn't do any difference. IPv4 works, IPv6 doesn't.

bigstefan, 2024-07-12:
Hi, I tried to analyze the problems why IPv6 is not working with lan865x
I see 2 Problems

**1) Fragmented Packet with 64Byte Buffers** 
We use 64 Byte buffer to receive the packets, due to the segmentation of the packet the IPv6 stack is notable to process the neighbor discovery (solicitation / advertisement).
https://github.com/zephyrproject-rtos/zephyr/blob/9c516db69c4d1313f19e1405991353dfe9e0127e/subsys/net/ip/ipv6_nbr.c#L1174

I did a hack to use bigger buffers (4*64 Byte) and concatenated the received data. This showed that neighbor discovery  works and also ping packets can be processed and replied successfully.
https://github.com/zephyrproject-rtos/zephyr/blob/9c516db69c4d1313f19e1405991353dfe9e0127e/drivers/ethernet/oa_tc6.c#L321
The code is not ok for a real driver, we have to find better ways to concat this data or make the IPv6 stack to deal with segmented headers.

**2) Address Filtering for Multicast Address** 
With IPv6 neighbor solicitation uses multicast mac-addr. In the implementation I do not find the required code to to so.
https://github.com/zephyrproject-rtos/zephyr/blob/9c516db69c4d1313f19e1405991353dfe9e0127e/drivers/ethernet/eth_lan865x.c#L62
I compared the code to the linux implementation
https://lore.kernel.org/all/20240418125648.372526-1-Parthiban.Veerasooran@microchip.com/
By mistake the implementation of lan865x_set_specific_multicast_addr(struct net_device *netdev) is not working either.

As a workaround  we can set the interface to promiscuous mode and reception of multicast is working fine.
I tried to find a description or sample code how to set the MAC Hash Register (MAC_HRB/MAC_HRT) but without success.

I expect that fixing this 2 short comings could help

pdgendt, 2024-07-12:
> **1) Fragmented Packet with 64Byte Buffers** We use 64 Byte buffer to receive the packets, due to the segmentation of the packet the IPv6 stack is notable to process the neighbor discovery (solicitation / advertisement).

This is something that will need to be fixed in the driver? It should probably linearize a minimum header size.

> **2) Address Filtering for Multicast Address** With IPv6 neighbor solicitation uses multicast mac-addr. In the implementation I do not find the required code to to so.

You can take a look at other ethernet drivers that implement the `ETHERNET_HW_FILTERING` configuration, many use a hash table to reduce the incoming multicast packets.

lmajewski, 2024-08-06:
@spe-ciellt Please test following PR: https://github.com/zephyrproject-rtos/zephyr/pull/76730

spe-ciellt, 2024-09-17:
Since the pull request was accepted and fixed the problem, this could be closed as well.