zephyrproject-rtos/zephyr #74895: sensors: qdec_stm32 does not handle remapped pins properly

peter-mitsis, 2024-06-24:
**Notes (delete this)**
Github Discussions (https://github.com/zephyrproject-rtos/zephyr/discussions)
are available to first verify that the issue is a genuine Zephyr bug and not a
consequence of Zephyr services misuse.

This issue list is only for bugs in the main Zephyr code base
(https://github.com/zephyrproject-rtos/zephyr/). If the bug is for a project
fork (such as NCS) specific feature, please open an issue in the fork project
instead.

**Describe the bug**
I have recently encountered an out-of-the-box issue with the blue pill board (stm32_min_dev_blue), the qdec sensor (drivers/sensors/st/qdec_stm32) and remapping timer2 pins (to use PA15 and PB3 instead of the default PA0 and PA1). If I am following the manuals correctly, this is not limited to the blue pill board, but applies to boards that use the stm32f1xx family of ARM processors.

With this setup stm32f1xx processor needs the following additional line to have the qdec work.

AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_JTAGDISABLE; // JTAG-DP Disabled, SW-DP Enabled

(Solution to my problem was found on stackoverflow at ... https://stackoverflow.com/questions/45066313/issues-with-timer2-on-stm32f103-in-encoder-interface-mode)

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - What have you tried to diagnose or workaround this issue?
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
 - ...

**To Reproduce**
1. Create a project to use the stm32_min_dev_blue board.
Ensure its board/stm32_min_dev_blue.overlay contains ...

```
/ {
    chosen {
        zephyr,console = &usart3;
        zephyr,timer,qdec = &qdec2;
        zephyr,gpiob = &gpiob;
    };
};

&timers2 {
    status = "okay";
    st,prescaler = <0>;

    qdec2: qdec {
        status = "okay";
        compatible = "st,stm32-qdec";
        pinctrl-0 = <&tim2_ch1_remap3_pwm_in_pa15 &tim2_ch2_remap3_pwm_in_pb3>;
        pinctrl-names = "default";
        st,counts-per-revolution = <360>;
        st,encoder-mode = <1>;
    };
};
```

2. Ensure the prj.conf enables at least ...
```
CONFIG_GPIO=y
CONFIG_SENSOR=y
CONFIG_QDEC_STM32=y.

```

3. Wire PA15 and PB3 to two GPIOB pins that the software will control to simulate a QDEC signal.

4. Simulate the QDEC in signal in software and display the angle using ...
```
    while(1) {
       switch(iterations & 3) {
        case 0:
            gpio_pin_set(gpiob_dev, PIN_PBy, 1);
            break;
        case 1:
            gpio_pin_set(gpiob_dev, PIN_PBx, 1);
            break;
        case 2:
            gpio_pin_set(gpiob_dev, PIN_PBy, 0);
            break;
        case 3:
            gpio_pin_set(gpiob_dev, PIN_PBx, 0);
            break;
        }
        iterations++;

        sensor_sample_fetch(qdec_dev);
        sensor_channel_get(qdec_dev, SENSOR_CHAN_ROTATION, &val);
        printk("QDEC rotation angle is %d\n", val.val1);
        k_sleep(K_MSEC(500));

    }
```

5. Build and flash the project ...
```
    west build -p -b stm32_min_dev_blue <path to project>
    west flash
```

6. Observe the output.

**Expected behavior**

The output should be cycling through the numbers 0..359 (either incrementing or decrementing). However, this will not occur unless the following line is added for this configuration ...

`AFIO->MAPR |= AFIO_MAPR_SWJ_CFG_JTAGDISABLE; // JTAG-DP Disabled, SW-DP Enabled
`
**Impact**

Discovering the solution to this problem has been a pain in the proverbial backside. I expect that I will not be the only Zephyr user to be hit with this. It is also likely that other devices might be hit by this as spi1 can also remap pins PA15 and PB3.

**Environment (please complete the following information):**
  - OS : Ubuntu
  - Toolchain: Zephyr SDK
 - Zephyr version: 3.6 (older versions too)

**Solution I would like to see**

This issue is specific to timer2 (and possibly spi1) when it remaps the pins to use PA15 and PB3. Ideally, I would either like to see the initialization code detect that the remapped pins are in effect and disable the JTAG as shown above so that the use of the remapped timer2 pins are as seamless/smooth as possible. Alternatively, an optional field could be added to the device tree that indicates that the JTAG should be disabled (documenting when to use this optional field would be a 'must' if this approach is taken).

erwango, 2024-06-24:
Can you try to see if the following could also fix the issue ?
```
&pintctrl {
       swj-cfg = jtag-disable;
};
```

If this is the case, then we actually do have a solution but question is how to make it easier to find out.

peter-mitsis, 2024-06-25:
I tried the pinctrl as suggested, and it does the trick.

erwango, 2024-06-25:
> I tried the pinctrl as suggested, and it does the trick.

Ok, thanks for confirming.
Now, based on this experience, is there anything that would have helped you to find it earlier ?

peter-mitsis, 2024-06-25:
One thing that would have made it easier to find would be a note in the sensor binding DTS page for st,stm32-qdec (online docs page = [dts/bindings/sensor/st,stm32-qdec.yaml ](https://docs.zephyrproject.org/latest/build/dts/api/bindings/sensor/st,stm32-qdec.html)

Something like ...

If the chosen configuration remaps pins to use a pin otherwise used for the JTAG, the JTAG may need to be disabled for this sensor to work correctly in that configuration. This can be done by .... <fill in the &pinctrl details here>.

The above sample may not be perfect, but it at least gives a developer some relevant information that may assist in both searches within the manuals, or online help forums.

erwango, 2024-06-26:
> One thing that would have made it easier to find would be a note in the sensor binding DTS page for st,stm32-qdec (online docs page = [dts/bindings/sensor/st,stm32-qdec.yaml](https://docs.zephyrproject.org/latest/build/dts/api/bindings/sensor/st,stm32-qdec.html)

This problem is not specific to qdec but valid to all devices that could make use of impacted pins (likely all devices actually). So this note would have to be duplicated in all related yaml.
Also note that this issue is F1 specific and this binding is used for all stm32, so message would have to be adapted.

So globally, I don't think this is the right direction. What about a note in impacted boards doc ?

erwango, 2024-06-27:
@peter-mitsis I'm going to convert this point as a (doc) enhancement. Is that fine for you ?

peter-mitsis, 2024-06-27:
@erwango - Converting this to a (doc) enhancement works for me.