zephyrproject-rtos/zephyr #74899: ESP32-S2 WiFi does not work (Scan returns no results)

carterghill, 2024-06-24:
**Describe the bug**
The WiFi interface seems altogether non-functional on the S2 chip. Running "wifi scan" in the shell returns no results, and any attempts to connect always fail, through the code or the shell.

`
[00:00:00.165,000] <inf> pca953x: pca9534@20 init ok
[00:00:00.217,000] <err> esp32_wifi: Failed to send packet
[00:00:00.218,000] <err> esp32_wifi: Failed to send packet
[00:00:00.219,000] <err> esp32_wifi: Failed to send packet
[00:00:00.220,000] <err> esp32_wifi: Failed to send packet
*** Booting Zephyr OS build v3.7.0-rc1 ***
[00:00:01.221,000] <err> esp32_wifi: Failed to send packet
[00:00:02.222,000] <err> esp32_wifi: Failed to send packet
uart:~$ wifi scan
Scan requested
Scan request done
[00:00:07.386,000] <inf> esp32_wifi: No Wi-Fi AP found
` 

Sometimes it will actually show one or two of the nearby SSIDs, 

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr/samples/net/wifi
2. west build -p -b esp32s2_v2duck && west flash --esp-device /dev/ttyACM0
3. minicom -b 921600 -D /dev/ttyUSB0
4. "wifi scan"

We are using a custom board that is most similar to the saola board, so for the purposes of reproducing the bug on your end I would point to esp32s2_saola or maybe esp32s2_devkitc. I believe the only difference between our board and the saola is the uart pin output

**Expected behavior**

When compiling the wifi sample code for another esp32 device, the expected output is shown: 

`
uart:~$ wifi scan
Scan requested

Num  | SSID                             (len) | Chan (Band)   | RSSI | Security        | 
1    | *********                       10    | 6    (2.4GHz) | -51  | WPA2-PSK        | 
2    | *********                       10    | 1    (2.4GHz) | -56  | WPA2-PSK        | 
3    | ********                          6     | 5    (2.4GHz) | -64  | UNKNOWN        | 
Scan request done
uart:~$ 

`

**Impact**
This is a show stopper, we need our devices to have functioning wifi.

**Logs and console output**

Given the inconsistent nature of the bug (the fact that it SOMETIMES shows an SSID) has made us think it might be a memory issue, so here's some kernel information that may or may not shed some light on the issue for you. 

Max allocated does go up after the first scan, and then stays the same afterwards.

`uart:~$ kernel heap
free:           42160
allocated:      27536
max. allocated: 27720
uart:~$ wifi scan
Scan requested
Scan request done
[00:00:13.918,000] <inf> esp32_wifi: No Wi-Fi AP found
uart:~$ kernel heap
free:           42084
allocated:      27600
max. allocated: 29592
`

`uart:~$ kernel threads
Scheduler: 71969 since last call
Threads:
 0x3ffcb618 rx_q[0]   
        options: 0x0, priority: -1 timeout: 0
        state: pending, entry: 0x400a0688
        stack size 2048, unused 1784, usage 264 / 2048 (12 %)

 0x3ffcb278 wifi      
        options: 0x8, priority: 2 timeout: 0
        state: pending, entry: 0x40034390
        stack size 3584, unused 2464, usage 1120 / 3584 (31 %)

 0x3ffcb540 net_mgmt  
        options: 0x0, priority: -1 timeout: 0
        state: pending, entry: 0x4002a76c
        stack size 2048, unused 840, usage 1208 / 2048 (58 %)

*0x3ffcb050 shell_uart
        options: 0x0, priority: 14 timeout: 0
        state: queued, entry: 0x40086380
        stack size 2048, unused 456, usage 1592 / 2048 (77 %)

 0x3ffcb1d0 esp_timer 
        options: 0x8, priority: 2 timeout: 0
        state: pending, entry: 0x40089464
        stack size 4096, unused 3680, usage 416 / 4096 (10 %)

 0x3ffcb8c0 sysworkq  
        options: 0x1, priority: -1 timeout: 0
        state: pending, entry: 0x4002a76c
        stack size 1024, unused 256, usage 768 / 1024 (75 %)

 0x3ffcaf70 logging   
        options: 0x0, priority: 14 timeout: 0
        state: pending, entry: 0x4002cfb4
        stack size 1024, unused 744, usage 280 / 1024 (27 %)

 0x3ffcb780 idle      
        options: 0x1, priority: 15 timeout: 0
        state: , entry: 0x4002a240
        stack size 1024, unused 840, usage 184 / 1024 (17 %)
`

sylvioalves, 2024-06-24:
@marekmatej

marekmatej, 2024-06-24:
hi @carterghill I see the same behavior on S2 boards and we are looking for the fix.

closed 2024-07-11 by nashif

carterghill, 2024-07-17:
Unfortunately, this has not fixed things on my end it seems.  After "west update" (and then blobs), I verified that the changes made were present in clock_control_esp32.c, west.yml, and loader.c, yet after building and flashing the wifi sample code, I get the same results as above.

sylvioalves, 2024-07-17:
@carterghill We confirmed that was an issue with ESP32-S2 Wi-Fi exactly as you described. However, the fix provide was good. Is there any other ESP32-S2 devkit board you could give a second try? Do you have any changes in prj.conf?

carterghill, 2024-07-19:
> @carterghill We confirmed that was an issue with ESP32-S2 Wi-Fi exactly as you described. However, the fix provide was good. Is there any other ESP32-S2 devkit board you could give a second try? Do you have any changes in prj.conf?

No changes to prj.conf. I tested several of our devices, and I actually managed to find one that was able to get wifi scan results. This was a bit surprising to me, because wifi had been confirmed to work on our boards a while ago using arduino. I will try to confirm that myself on the boards that are currently not getting results with zephyr.

Might be a silly question, was the fix tested on Saola boards specifically? I know there are different varieties of the S2 but I don't know if the differences would be significant.

In any case, probably the best thing to do is figure out what the difference is with our one functioning board. Thank you for your help!

PR #75680 (2024-07-11): soc: esp32s2: fix wifi clock gate
ESP32-S2 Wi-Fi clock is not initialized properly, causing instability when scanning or connecting to a SSID. In addition, SARADC was kept enabled to feed RND peripheral, adding more instability to Wi-Fi connection.

Fixes #74899
Fixes #74417
commit 43a55908 (2024-07-11): soc: esp32s2: fix wifi clock gate

ESP32-S2 Wi-Fi clock is not initialized properly, causing
instability when scanning or connecting to a SSID.

Fixes #74899
Fixes #74417