zephyrproject-rtos/zephyr #74914: RP2040 Bluetooth HCI Issues

xudongzheng, 2024-06-25:
There are a number of dual chip RP2040 boards that support Bluetooth via an external chip (LILYGO T-PicoC3, Arduino Nano RP2040 Connect, to name a few). RP2040 communicates with the ESP32 chip via Bluetooth HCI UART.

This setup was working with Zephyr 3.6 but is no longer working with Zephyr 3.7 RC1 for two reasons:

- Hardware flow control is missing: This was initially implemented in #58294 but was removed in the #71074 refactoring.
- FIFO is enabled: Even before #71074 was merged, HCI UART has been broken since #70618. In current Zephyr `main`, commenting out `pl011_enable_fifo()` fixes the issue. Disabling FIFO universally doesn't seem to be the right solution but I have no clue why FIFO is causing HCI UART to break.

**Impact**

Currently running a patched version of Zephyr with hardware flow control and `pl011_enable_fifo()` commented out.

protocentralashwin, 2025-01-29:
@xudongzheng , I am also facing exactly the same issue after upgrading from v3.6 to v4.0 . 

Ours is an RP2040 connected to a ESP32C3 with HCI H4. I tried modifying the above files, but still don't get a response from HCI controller. Did you get it working? Just to check, I tried downgrading to 3.6, still works.

xudongzheng, 2025-01-29:
@protocentralashwin If you don't get a response at all, could be a different issue.

I did try briefly with Zephyr 4.0 but didn't use it for very long. For reference, I was using 115200 baud with flow control enabled.

You can also try it with Zephyr 3.7 and see where exactly it breaks.

protocentralashwin, 2025-01-29:
Thanks @xudongzheng !

I've tried with Zephyr 3.7 also, this issue started only when upgrading from 3.6 to 3.7, so it seemed a bit strange. I will dig a little more deeper and will update this issue if I find anything. I'm using 921600 with HW flow control. This baud rate was working fine for a long time.

petejohanson, 2025-01-29:
From playing with enabling the FIFO on RP2040 for some other work of mine, I would *suspect* this is insufficient handling of the RX timeout. I had to play with this when enabling the RP2040 FIFO in ZMK's Zephyr branch (using the old dedicated RP2040 UART driver in Zephyr 3.5)

xudongzheng, 2025-02-13:
The FIFO part of the Bluetooth HCI issue is fixed in #85539. With that, FIFO can be left enabled. Flow control is still needed for running at 115200 baud in my case.

protocentralashwin, 2025-02-13:
I was using a custom fork with the modified pl011 (with FIFO disabled). It works well upto 230400 baud and after that it doesn't seem to communicate. Also, I found that ESP-IDF's hci_h4 example works reliably well as the hci controller on the ESP32C3.

protocentralashwin, 2025-02-13:
Thanks @xudongzheng . I can confirm that after this PR as a patch to branch 3.7, the Bluetooth is back to work well at 921600 baud !

PR #79474 (2025-03-07): drivers: serial: pl011: Add fifo enable configuration
Add a setting to the devicetree for enabling or disabling
the PL011 FIFO.

fix #74914
closed 2025-03-07 by carlescufi