zephyrproject-rtos/zephyr #74925: flash: breaking API changes undocumented, impossible to support storage solutions for multiple zephyr versions

Laczen, 2024-06-25:
**Describe the bug**
PR #71271 introduced a breaking change to the flash API for devices that do not need to be erased prior to writing. Before #71271 devices that have no erase simulated and erase by writing the erase value. After #71271 these devices no longer have an erase and the erase should be done by calling ˋflash_flatten()ˋ.

This breaking API change is not documented in the release notes or migration guide for release 3.7.

The breaking API change makes it impossible for out of tree storage solutions to provide support for zephyr pre and post release 3.7 on devices that do not require erase without knowing the zephyr version they are operating on.

de-nordic, 2024-06-25:
Nothing is broken here. The statement is completely false and malicious.
The erase has not been removed and out of tree storage solutions work as they did previously, not most efficient by the parameters of device but they do. Currently the `flash_flatten` is addition to `flash_erase`.

The plan has been described here: https://github.com/zephyrproject-rtos/zephyr/issues/71270
and stage 1 of the plan is advertisement of the feature and how to adapt.

Laczen, 2024-06-25:
You are right, this bug will appear in stage 1.

closed 2024-06-25 by Laczen