zephyrproject-rtos/zephyr #74931: Adding support for w6100-evb-pico

seyednasermoravej, 2024-06-25:
w6100-evb-pico is a board based on RP2040 and embedded ethernet using W6100 from Wizent.
The board is similar to w55000-evb-pico which is supported by Zephyr RTOS already.
Thanks

kartben, 2025-05-22:
@seyednasermoravej would you be interested in trying to submit this as a pull request?

soburi, 2025-05-23:
Currently the w6100 driver is not implemented.
If someone can do this, adding a board is not difficult.

As far as I can tell from the commit history of the w5500, this is not something that someone from wiznet has done, but rather something that a developer has done with voluntarily effort, and someone needs to accomplish this.

seyednasermoravej, 2025-05-23:
@kartben 
Hi, I have already changed the w5500 driver for the w6100. It's working for my client over 6 month. I haven't collaborate developing driver in Zephyr yet. It would be great if someone helps to do that. Besides the IPv6 is not implemented yet.

soburi, 2025-05-23:
@seyednasermoravej 

That's great.
If so, please create a PR to add the W6100 driver and consult the author of the W5500 driver @parthitce
(https://github.com/zephyrproject-rtos/zephyr/pull/27966)

and Ethernet maintainers and collaborators
https://github.com/zephyrproject-rtos/zephyr/blob/f77e258cb9c2aad7fe81f7489d5fa8ff0f269899/MAINTAINERS.yml#L1451-L1459

Once this is done, adding the board definition will be like dessert.

seyednasermoravej, 2026-01-05:
Now I added only IPv4 support. I'll work on the to support also IPv6  as soon as possible.
https://github.com/zephyrproject-rtos/zephyr/pull/101753