zephyrproject-rtos/zephyr #74955: shell_uart "RX ring buffer full." freezes system

bmulder-innoseis, 2024-06-25:
**Describe the bug**
When sending a lot of messages over shell_uart and the ring buffer gets full (the log message "RX ring buffer full." shows), the system and shell freezes and becomes unresponsive until a reboot.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? NRF52840
 - What have you tried to diagnose or workaround this issue? Decreasing CONFIG_SHELL_BACKEND_SERIAL_RX_RING_BUFFER_SIZE seems to make the problem more frequent. Increasing it seems to make it less frequent. I tried commenting out the log message to see if it was caused by that log message trying to send itself over the UART (and because of that flooding the buffer again), but to no avail. Only long duration runs made the issue apparent and as such, I am not entirely confident that increasing the buffer size will not just postpone the issue to a later time.

**To Reproduce**
Steps to reproduce the behavior:
Use the shell_module example with e.g. CONFIG_SHELL_BACKEND_SERIAL_RX_RING_BUFFER_SIZE=20
I Used a script that keeps sending commands over the serial and read the output in quick succession.
After a while, the "RX ring buffer full" sometimes starts appearing and the system hangs.

**Expected behavior**
If the buffer is full, I expect messages to be dropped, ignored or shortened. Instead, the whole system hangs.

**Impact**
The RX ring buffer full message did not always occur in my custom system, which left me with quite some time debugging the issue and figuring out that increasing the RX buffer size could solve the issue temporarily.
Because I am not certain if the buffer size increase will actually fix the problem rather than make it more rare, something which is quite hard to assess, it means that whenever a user enters a lot of commands in quick succession (a quite likely scenario), it will render the whole system unusable.

**Environment (please complete the following information):**
 - OS: Linux & Windows
 - Toolchain : nrf connect toolchain v2.5.1 with zephyr 3.6.0
 - Commit SHA or Version used: 3.6.0

jakub-uC, 2024-06-27:
Please let me know how did you set: `CONFIG_SHELL_CMD_BUFF_SIZE`?

bmulder-innoseis, 2024-06-27:
> @bmulder-innoseis Please let me know what your value is for `CONFIG_SHELL_CMD_BUFF_SIZE`?

I used the default, which is 256.
According to the description, this value is the maximum size of the command itself. I never sent any long commands to the shell, though.

bmulder-innoseis, 2024-06-27:
The amount of output it sends back can be quite large, though. My suspicion is that because it is busy sending data over the wire (and doing other tasks in the background), the buffer fills itself faster than the incoming input can be handled..

jakub-uC, 2024-06-27:
I am trying to reproduce it. So far I can see a feedback like that:

```
[00:04:16.774,261] <wrn> shell_uart: RX ring buffer full.
--- 299 messages dropped ---
[00:04:16.801,330] <wrn> shell_uart: RX ring buffer full.
--- 368 messages dropped ---
[00:04:16.829,071] <wrn> shell_uart: RX ring buffer full.
--- 215 messages dropped ---
[00:04:16.856,903] <wrn> shell_uart: RX ring buffer full.
--- 225 messages dropped ---
[00:04:16.883,850] <wrn> shell_uart: RX ring buffer full.
--- 227 messages dropped ---
[00:04:16.922,424] <wrn> shell_uart: RX ring buffer full.
--- 218 messages dropped ---
[00:04:16.951,995] <wrn> shell_uart: RX ring buffer full.
--- 192 messages dropped ---
[00:04:16.977,569] <wrn> shell_uart: RX ring buffer full.
[00:04:17.002,960] <wrn> shell_uart: RX ring buffer full.
[00:04:17.027,496] <wrn> shell_uart: RX ring buffer full.
[00:04:17.054,626] <wrn> shell_uart: RX ring buffer full.
[00:04:17.080,139] <wrn> shell_uart: RX ring buffer full.
[00:04:17.106,750] <wrn> shell_uart: RX ring buffer full.
[00:04:17.133,178] <wrn> shell_uart: RX ring buffer full.
[00:04:17.160,125] <wrn> shell_uart: RX ring buffer full.
[00:04:17.199,401] <wrn> shell_uart: RX ring buffer full.
[00:04:17.228,637] <wrn> shell_uart: RX ring buffer full.
[00:04:17.254,455] <wrn> shell_uart: RX ring buffer full
```
So far I didn't see any freeze, but I will keep trying

jakub-uC, 2024-06-27:
I copied several hundred characters to the clipboard and am constantly pasting them into the Picocom console. It looks like Picocom is buffering the messages and continually sending them. After a very short time, the shell starts displaying the message shell_uart: RX ring buffer full. because it cannot keep up with processing the information from Picocom, this is expected because the information is coming in faster than the system can process it.

After pressing the enter key (entering a command), the system becomes fully responsive for me. 
So far, I haven't noticed any freezes that require a reboot. I have an nrf52840 DK somewhere, and I will try to reproduce this on it.

bmulder-innoseis, 2024-06-27:
I have maybe not mentioned that I used a different app which is using the USB next device stack. Sorry for that, I forgot about it. I'm testing the original sample again right now with the old USB stack.

bmulder-innoseis, 2024-06-27:
I have been able to test it with the old USB stack as well as the modified sample which includes the USB next stack. I have been able to reproduce the "RX ring buffer full" messages, but not the freezes. I must conclude that I probably broke something inside my custom app. Sorry for bothering you with it.

bmulder-innoseis, 2024-07-01:
For anyone wondering or having the same kind of issue, in my case I solved it by playing around with thread priorities. It seems that if I have a cooperative thread not being cooperative enough and taking up a lot of time while the USB stack and shell are active, the USB stack crashes after some point, freezing the shell as well.
It seems some drivers in the USB stack have real-time requirements and do not work well if not getting a certain amount of time-share.
Decreasing my "important" thread to only priority zero seems to fix this.