zephyrproject-rtos/zephyr #75004: issue with CONFIG_MMC_VOLUME_NAME

KlimRadostev, 2024-06-25:
**Describe the bug**
observed issue with CONFIG_MMC_VOLUME_NAME, in
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/disk/mmc_subsys.c#L134
could cause issues in a situation where 2 mmc subsystems need to be initialized as it will marry the driver only to 1 label.

would the correct way not be to leverage the device trees Label otherwise we will get collisions when attempting to use
disk_access APIs?

decsny, 2024-07-11:
> > would the correct way not be to leverage the device trees Label otherwise we will get collisions when attempting to use
> > disk_access APIs?
> 
> Yes, I think it would be better to use device tree labels here. We could probably add support for using a devicetree property here. The ramdisk driver does this:
> 
> https://github.com/zephyrproject-rtos/zephyr/blob/a6e672f9b2368e6d011da189afd462fec599fd0d/drivers/disk/ramdisk.c#L164
> . This is a case where we'd be doing software configuration with devicetree though. @decsny, any thoughts on a way to solve this without devicetree?

personally think the DT label property is exactly for this purpose, but clearly our documentation and recent historical efforts have been hostile to using this property, although I'm actually not 100% clear why. I know it was used improperly in the past but I think deprecating it and banning it completely might have gone too far.

@galak

[image]

This is from the devicetree v0.4 spec. It seems like this is the exact situation where it should be used. Do you disagree?

decsny, 2024-07-11:
@danieldegrasse I am assuming that this "name" of a disk is intended to be human readable, otherwise this is a moot discussion. Can you confirm the reason for this. Right now I only see it used in LOG_DBG statements.

jfischer-no, 2024-07-11:
> Yes, I think it would be better to use device tree labels here. We could probably add support for using a devicetree property here. The ramdisk driver does this: [...]

disk_name property refers to disk_info.name, which is used as disk identifier. IMO disk_name is less confusing for users as label property. Only specific names can be used if the filesystem is ELM FATFS, see https://docs.zephyrproject.org/latest/services/file_system/index.html#c.fs_mount. Why cannot zephyr,mmc-disk.yaml aligned with zephyr,ram-disk.yam? Also, it should be move do dts/bindings/disk. (Btw, I described the problem to you in Prague. Same applies to SDMMC)

decsny, 2024-07-11:
> disk_name property refers to disk_info.name, which is used as disk identifier. IMO disk_name is less confusing for users as label property. Only specific names can be used if the filesystem is ELM [...]

As in the screenshot I posted, the label meaning should be specified by the binding. So there should not be any confusion. But I am still not clear if this identifier really needs to be a human readable string, I don't think LOG_DBG is enough justification for that, but maybe I am not seeing something.

danieldegrasse, 2024-07-11:
> As in the screenshot I posted, the label meaning should be specified by the binding. So there should not be any confusion. But I am still not clear if this identifier really needs to be a human [...]

I think it does need to be human readable, or at least able to be set in a way that is deterministic between builds (so we can't just add the instance number or something similar to make it unique)

The reason being that the user's application would access the disk by the disk name if there was a need to make direct I/O calls to the disk itself, without a file system layer

commit 07a8e325 (2024-11-05): drivers: disk: mmc_subsys: remove CONFIG_MMC_VOLUME_NAME

Remove CONFIG_MMC_VOLUME_NAME, and set the disk name based on the
``disk-name`` property. This aligns with other disk drivers, and allows
for multiple instances of the mmc_subsys disk driver to be registered.

Add disk-name properties for all in tree definitions for the
mmc-subsys disk driver, and change all in tree usage of the disk name

Fixes #75004
commit a1dc0b8b (2024-11-05): drivers: disk: sdmmc_subsys: remove CONFIG_SDMMC_VOLUME_NAME

Remove CONFIG_SDMMC_VOLUME_NAME, and set the disk name based on the
``disk-name`` property. This aligns with other disk drivers, and allows
for multiple instances of the sdmmc_subsys disk driver to be registered.

Add disk-name properties for all in tree definitions for the
sdmmc-subsys disk driver, and change all in tree usage of the disk name

Fixes #75004
closed 2024-11-05 by mmahadevan108