zephyrproject-rtos/zephyr #75020: mDNS response is not sent out over STM32U575zi-WiFi

VineetaNarkhede-eaton, 2024-06-26:
### Discussed in https://github.com/zephyrproject-rtos/zephyr/discussions/71540

<div type='discussions-op-text'>

<sup>Originally posted by **VineetaNarkhede-eaton** April 16, 2024</sup>
Hi,

I am facing an issue in running mDNS over STM32U575zi on WiFi interface. Device (Dev board) is in AP mode and connected with Laptop over WiFi. Enabled the mDNS responder in zephyr. Hostname configured is "zephyr"
When I send mdns query from Laptop i.e zephyr.local, the query is successfully reaching the device from the laptop. However, the response from the device does not appear to be reaching the laptop. I have verified that the response is reaching till WiFi driver and the response data is valid. Wireshark logs on the laptop shows query but no response visible. Interestingly, mDNS query-response works sometimes suddenly(very rare) without any code changes. 

I tried the solution provided on this issue https://github.com/zephyrproject-rtos/zephyr/discussions/48327, but no improvement in the result.

**Setup**:
Dev board: NUCLEO-U575ZI-Q
WiFi chip:1YN-CY43439
zephyr OS : v3.2.0
infineon_zephyr_connectivity commit id : 272e2b8da908c7d796ec62757cc8abbe03a91e4c
 
I would appreciate any help with this matter.</div>

jukkar, 2024-06-27:
I was hoping a proper issue instead of pressing the "convert to issue" button as this just copies the first question to an issue. Lot of relevant information missing now.
Anyway, lets start by first figuring out what wifi driver are you using, mentioning chip does not really help here.

jukkar, 2024-06-27:
> Can we replace NET_IF_LOWER_UP with NET_IF_UP in net_if_tx() function to solve this issue?

No, you should not do that. `NET_IF_UP` presents admin state of the interface i.e., it is set when the application or user turns on the interface. The `NET_IF_LOWER_UP` is set when the carrier is turned on. For wifi this means that when the wifi connection is "ready". So it looks like the wifi interface carrier status is turned "ON" too early by the driver when the interface in fact is not yet ready for the connection.

rajolegajendra-eaton, 2024-07-08:
> No, you should not do that. `NET_IF_UP` presents admin state of the interface i.e., it is set when the application or user turns on the interface. The `NET_IF_LOWER_UP` is set when the carrier is [...]

Hello @jukkar,
We have noticed one change in [NET_IF_INIT()](https://github.com/zephyrproject-rtos/zephyr/blob/4769dbb650c88e2946e564b107b926eb09d909f8/include/zephyr/net/net_if.h#L2841) macro which is getting called by [NET_DEVICE_DT_INST_DEFINE()](https://github.com/zephyrproject-rtos/zephyr/blob/4769dbb650c88e2946e564b107b926eb09d909f8/drivers/wifi/infineon/airoc_wifi.c#L771) macro. It is setting flag bit for NET_IF_LOWER_UP as highlighted in following image:
[image]
This sets the flag as carrier ON. As mentioned earlier we are initializing WiFi in application and not before main. So carrier must be turned on only after Network Link is Up and Valid IP is assigned. Since carrier flag is set before wireless network is setup, so iface will not have Valid IP address. When mDNS is initialized from [init_listener()](https://github.com/zephyrproject-rtos/zephyr/blob/4769dbb650c88e2946e564b107b926eb09d909f8/subsys/net/lib/dns/mdns_responder.c#L769) using SYS_INIT() so it finds iface src address as null and gives bus fault.
What is the significance of setting flag to NET_IF_LOWER_UP in NET_IF_INIT()? Found one commit where this change was done. [Link is here](https://github.com/zephyrproject-rtos/zephyr/commit/6271f2fa56067aa491d918044c331558bf44c794#diff-aad12f53aa6f613cd408e3beb066a1bb3f12ff6828e7af7882c9cc20cb155234).
[image]
cc: @VineetaNarkhede-eaton

rajolegajendra-eaton, 2024-07-11:
@jukkar ,
We are waiting for your response.

Above issue can be resolved if we change `NET_IF_LOWER_UP` flag to `NET_IF_NO_AUTO_START`.

jukkar, 2024-07-12:
You have out of tree driver and application, it very difficult to understand what your code is doing.

cc: @rlubos if you have any extra ideas

rlubos, 2024-07-12:
> This sets the flag as carrier ON. As mentioned earlier we are initializing WiFi in application and not before main. So carrier must be turned on only after Network Link is Up and Valid IP is assigned.

The carrier is set to ON by default, for legacy reasons (and we don't push to change it as Linux does the same). If the particular interface is not ready to use after system initialization, it should be changed to off.

Every network interface has possibility to run boot-time initialization by registering `init()` function from `struct net_if_api` (the API struct is statically provided when registering interface with `NET_DEVICE_DT_INST_DEFINE()` or similar). You should set carrier to OFF there.

You can take ESP driver as a reference here, it seems to be a similar case:
https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/wifi/esp32/src/esp_wifi_drv.c#L681

rajolegajendra-eaton, 2024-07-17:
> You can take ESP driver as a reference here, it seems to be a similar case:
> https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/wifi/esp32/src/esp_wifi_drv.c#L681

Hello @rlubos,
Thanks for the reference, we have tried it and it seems Bus Fault issue is resolved. Will add an update here once testing is completed.
cc: @VineetaNarkhede-eaton

carlescufi, 2024-07-31:
Closing since it seems resolved. Please reopen if this is not the case.

closed 2024-07-31 by carlescufi