zephyrproject-rtos/zephyr #75038: doc: Asynchronous Notifications: Public User API sys_notify_get_method() is not documented.

butok, 2024-06-26:
**Describe the bug**
Public User API sys_notify_get_method() function is not documented: https://docs.zephyrproject.org/latest/services/notify.html#c.sys_notify_get_method

**Expected behavior**
All Public User API functions should be documented.

**Environment (please complete the following information):**

 - Commit SHA or Version used

henrikbrixandersen, 2024-06-27:
@butok Could you submit a PR for adding the missing Doxygen?

butok, 2024-06-27:
> @butok Could you submit a PR for adding the missing Doxygen?

It should be done by an author of this function.

kartben, 2024-09-06:
> All Public User API functions should be documented.

This function is actually marked as internal. But since it's not done with the correct Doxygen construct, that makes it show up in the docs making people think it's public, while it's not

closed 2024-11-20 by github-actions[bot]