zephyrproject-rtos/zephyr #75072: drivers: ethernet: stm32h7: DMA errors reported by the ethernet HAL driver

cperera-aud, 2024-06-27:
**Describe the bug**
When running a STM32H7 device (ST Nucleo H753ZI) on an ethernet network, the ethernet HAL driver is detecting RX DMA errors. I have created a minimal example to reproduce the issue and the error detected is "Receive Buffer Unavailable" errors:

<err> net_dhcpv4_client_sample: DMA ERRORS! 00004080

[image]

Are these errors real? Or should they be addressed? The device otherwise seems to be operating normally despite detecting these errors. 

I've only tried the scenario on a STM32H7, I'm not sure if the same errors are seen on other STM32 platforms.

I am also using a custom board that uses an STM32H7 and on this board I'm also seeing the same Receive Buffer Unavailable error. Therefore, it doesn't seem to be an error isolated on a particular board.

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? **STM32H7**
 - What have you tried to diagnose or workaround this issue? 
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit? **Not that I'm aware of**
 - ...

**To Reproduce**
Steps to reproduce the behavior:

1. Checkout eth_rx_dma_errors branch in https://github.com/Audinate/zephyr
2. In zephyr, west build -p always -b nucleo_h753zi samples/net/dhcpv4_client
3. Flash the zephyr.bin or zephyr.elf file to the Nucleo H753ZI board
4. Plug the device in an ethernet network, ideally with other ethernet devices connected on the same network
5. Wait for the device to report DMA ERRORS on the UART console

**Expected behavior**
I would not expect to see DMA/MAC errors being reported unless there are some hardware issues

**Impact**
Otherwise since device is working as expected this is mostly an annoyance, however, that being said, if this is a real error, is there a possibility that it could improve performance?

**Logs and console output**

`[00:00:00.051,000] <inf> phy_mii: PHY (0) ID 7C131

*** Booting Zephyr OS build v3.7.0-rc1-454-g7c64a8b1a7ad ***
[00:00:00.051,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:00.051,000] <inf> net_dhcpv4_client_sample: Start on ethernet@40028000: index=1
[00:00:03.753,000] <inf> phy_mii: PHY (0) Link speed 100 Mb, full duplex

[00:00:05.051,000] <err> net_dhcpv4_client_sample: DMA ERRORS! 00004080
[00:00:06.051,000] <err> net_dhcpv4_client_sample: DMA ERRORS! 00004080
[00:00:07.051,000] <err> net_dhcpv4_client_sample: DMA ERRORS! 00004080
[00:00:09.051,000] <err> net_dhcpv4_client_sample: DMA ERRORS! 00004080
[00:00:10.051,000] <err> net_dhcpv4_client_sample: DMA ERRORS! 00004080
[00:00:11.051,000] <err> net_dhcpv4_client_sample: DMA ERRORS! 00004080
[00:00:12.051,000] <err> net_dhcpv4_client_sample: DMA ERRORS! 00004080
[00:00:13.051,000] <err> net_dhcpv4_client_sample: DMA ERRORS! 00004080
[00:00:14.051,000] <err> net_dhcpv4_client_sample: DMA ERRORS! 00004080
[00:00:15.051,000] <err> net_dhcpv4_client_sample: DMA ERRORS! 00004080
[00:00:15.063,000] <inf> net_dhcpv4_client_sample: DHCP Option 42: 10.13.1.1
[00:00:15.143,000] <inf> net_dhcpv4_client_sample: DHCP Option 42: 10.13.1.1
[00:00:15.143,000] <inf> net_dhcpv4: Received: 10.10.233.59
[00:00:15.144,000] <inf> net_dhcpv4_client_sample:    Address[1]: 10.10.233.59
[00:00:15.144,000] <inf> net_dhcpv4_client_sample:     Subnet[1]: 255.255.255.0
[00:00:15.144,000] <inf> net_dhcpv4_client_sample:     Router[1]: 10.10.233.1
[00:00:15.144,000] <inf> net_dhcpv4_client_sample: Lease time[1]: 43200 seconds
[00:00:16.051,000] <err> net_dhcpv4_client_sample: DMA ERRORS! 00004080
`

**Environment (please complete the following information):**
 - OS: **Linux, Ubuntu 20.04**
 - Toolchain: **GNU ARM**
 - Commit SHA or Version used: 7c64a8b1a7ad7d7330c27affcf0569713c922bcf

cperera-aud, 2024-08-27:
Please remove stale label.

cperera-aud, 2024-10-27:
Please remove the stale label.

nrg83, 2025-01-23:
The problem is this line of code in stm32h7xx_hal_eth.c i think.
`tailidx = (descidx + 1U) % ETH_RX_DESC_CNT;`

The descriptor list is organized as a ring buffer, e.g. desc[0], desc[1], desc[2], desc[3], desc[4], desc[5]

The tail pointer must set "behind" the current descriptor, not in front of it. E.g. current descriptor index = 2, tail pointer must be set to 1, and not 3!

RBU occurs when ETH_DMACRXDTPR = ETH_DMACCARXDR

You can try this fix maybe.
`if (descidx == 0) { tailidx = ETH_RX_DESC_CNT - 1; } else { tailidx = (descidx - 1U) % ETH_RX_DESC_CNT; }`

As general advice, I can only say, don't use this driver. We have been using the H7 since 2021 and the Ethernet driver has never worked reliably. So we developed our own. The Eth peripherals work, except for a few documented errors in the errata, but STM just can't get this driver under control.

The sad thing is that this problem is fixed in such a cheesy way in the code generated from CubeMX. 

[image]

Come on STM, you can do better than that! Get to work on it!

erwango, 2025-01-23:
Thanks @nrg83 for reporting this, we'll have a look. 

> Come on STM, you can do better than that! Get to work on it!

I agree that ideally we should avoid delivering bugs, but that the life of a software project, right?
So, please keep a respectful communication.

nrg83, 2025-01-24:
Sorry, that wasn't meant to sound disrespectful. I actually said that with a winking smiley ;)

marwaiehm-st, 2025-03-18:
Fixed by this PR https://github.com/zephyrproject-rtos/zephyr/pull/84965, the error 'Receive Buffer Unavailable' is no longer detected.

closed 2025-03-18 by marwaiehm-st