zephyrproject-rtos/zephyr #75100: mimxrt1170_evk: USB mass (next) sample triggers Bus Fault

jgenssler-GiN, 2024-06-27:
The USB MSC sample with "next" USB device stack config triggers a bus fault when running on i.MX RT1170 Evaluation Kit (MIMXRT1170-EVK).
Not using the new USB device stack config doesn't show that error.

Steps to reproduce the behavior:
1. Insert SD card into dev board
2. `west build -p -b mimxrt1170_evk/mimxrt1176/cm7 zephyr/samples/subsys/usb/mass/ -DCONF_FILE="usbd_next_prj.conf" -DCONFIG_APP_MSC_STORAGE_SDCARD=y`
3. `west flash`
5. Serial output:
  ```
  *** Booting Zephyr OS build v3.7.0-rc1-465-g08496ff6b449 ***
  Mount /SD:: 0
  /SD:: bsize = 512 ; frsize = 32768 ; blocks = 121456 ; bfree = 121455
  /SD: opendir: 0
  End of files
  [00:00:01.078,000] <err> os: ***** BUS FAULT *****
  [00:00:01.078,000] <err> os:   Precise data bus error
  [00:00:01.078,000] <err> os:   BFAR Address: 0x40430140
  [00:00:01.078,000] <err> os: r0/a1:  0x00000002  r1/a2:  0x800096a8  r2/a3:  0x8000b588
  [00:00:01.078,000] <err> os: r3/a4:  0x40430000 r12/ip:  0x80008b74 r14/lr:  0x30010cd1
  [00:00:01.078,000] <err> os:  xpsr:  0x21000000
  [00:00:01.078,000] <err> os: Faulting instruction address (r15/pc): 0x3000c94a
  [00:00:01.078,000] <err> os: >>> ZEPHYR FATAL ERROR 25: Unknown error on CPU 0
  [00:00:01.078,000] <err> os: Current thread: 0x80009218 (unknown)
  [00:00:01.100,000] <err> os: Halting system
  ```

Zephyr should have booted properly. USB Mass Storage device should show SD card contents when connected to PC.

That is a showstopper as old USB support works unreliably for USB MSC and gives I/O errors sometimes.

When debugged, the fault is triggered when the register in [this line in `usb_device_ehci.c`](https://github.com/zephyrproject-rtos/hal_nxp/blob/384bb4f17245a7e6b2d83149309134597f1b8a69/mcux/middleware/mcux-sdk-middleware-usb/device/usb_device_ehci.c#L1411) is written.

This is the backtrace:
```
0  z_arm_usage_fault () at /workdir/deps/zephyr/arch/arm/core/cortex_m/fault_s.S:80
#1  <signal handler called>
#2  0x3000c94a in USB_DeviceEhciInit (controllerId=2 '\002', handle=0x800096a8 <priv_data_0+4>, ehciHandle=0x800096a8 <priv_data_0+4>) at /workdir/deps/modules/hal/nxp/mcux/middleware/mcux-sdk-middleware-usb/device/usb_device_ehci.c:1421
#3  0x30010ccc in udc_mcux_init (dev=0x3001485c <__device_dts_ord_508>) at /workdir/deps/zephyr/drivers/usb/udc/udc_mcux_ehci.c:698
#4  0x30010a36 in udc_init (dev=0x3001485c <__device_dts_ord_508>, event_cb=event_cb@entry=0x300075a9 <usbd_event_carrier>) at /workdir/deps/zephyr/drivers/usb/udc/udc_common.c:761
#5  0x30007898 in usbd_device_init_core (uds_ctx=uds_ctx@entry=0x80008b2c <sample_usbd>) at /workdir/deps/zephyr/subsys/usb/device_next/usbd_core.c:207
#6  0x300069ca in usbd_init (uds_ctx=uds_ctx@entry=0x80008b2c <sample_usbd>) at /workdir/deps/zephyr/subsys/usb/device_next/usbd_device.c:209
#7  0x30002fec in sample_usbd_init_device (msg_cb=msg_cb@entry=0x0) at /workdir/deps/zephyr/samples/subsys/usb/common/sample_usbd_init.c:173
#8  0x3000306c in enable_usb_device_next () at /workdir/deps/zephyr/samples/subsys/usb/mass/src/main.c:57
#9  main () at /workdir/deps/zephyr/samples/subsys/usb/mass/src/main.c:213
```

 - Zephyr docker image v0.26.11
 - Zephyr SDK 0.16.5-1
 - latest `main` (`08496ff`)

Going back to `49be50c` (April 1st) doesn't produce a successful build:
```
$ west build -p -b mimxrt1170_evk/mimxrt1176/cm7 zephyr/samples/subsys/usb/mass/ -DCONF_FILE="usbd_next_prj.conf" -DCONFIG_APP_MSC_STORAGE_SDCARD=y
-- west build: making build dir /workdir/build pristine
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /workdir/zephyr/samples/subsys/usb/mass
-- CMake version: 3.22.1
-- Found Python3: /usr/bin/python3 (found suitable version "3.10.12", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: /home/user/.cache/zephyr
-- Zephyr version: 3.6.99 (/workdir/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: mimxrt1170_evk, Revision: A, qualifiers: mimxrt1176/cm7
-- Found host-tools: zephyr 0.16.5 (/opt/toolchains/zephyr-sdk-0.16.5-1)
-- Found toolchain: zephyr 0.16.5 (/opt/toolchains/zephyr-sdk-0.16.5-1)
-- Found Dtc: /opt/toolchains/zephyr-sdk-0.16.5-1/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.6.0", minimum required is "1.4.6") 
-- Found BOARD.dts: /workdir/zephyr/boards/nxp/mimxrt1170_evk/mimxrt1170_evk_mimxrt1176_cm7.dts
-- Generated zephyr.dts: /workdir/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /workdir/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /workdir/build/zephyr/dts.cmake
[... 27 lines cut ...]
-- west build: building application
[1/183] Preparing syscall dependency handling

[2/183] Generating include/generated/version.h
-- Zephyr version: 3.6.99 (/workdir/zephyr), build: v3.6.0-1967-g49be50c8ddc0
[178/183] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /workdir/build/zephyr/zephyr_pre0.map 
: && ccache /opt/toolchains/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/workdir/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/arch/arch/arm/core/mpu/libarch__arm__core__mpu.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/boards/nxp/mimxrt1170_evk/libboards__nxp__mimxrt1170_evk.a  zephyr/subsys/fs/libsubsys__fs.a  zephyr/subsys/sd/libsubsys__sd.a  zephyr/subsys/usb/device_next/libsubsys__usb__device_next.a  zephyr/subsys/net/libsubsys__net.a  zephyr/drivers/disk/libdrivers__disk.a  zephyr/drivers/usb/udc/libdrivers__usb__udc.a  zephyr/drivers/clock_control/libdrivers__clock_control.a  zephyr/drivers/console/libdrivers__console.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/hwinfo/libdrivers__hwinfo.a  zephyr/drivers/memc/libdrivers__memc.a  zephyr/drivers/pinctrl/libdrivers__pinctrl.a  zephyr/drivers/sdhc/libdrivers__sdhc.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/timer/libdrivers__timer.a  modules/fatfs/libmodules__fatfs.a  modules/hal_nxp/hal_nxp/lib..__modules__hal__nxp.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/opt/toolchains/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v7e-m/nofp"  -L/workdir/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m7  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  --specs=picolibc.specs  -lc  -lgcc && cd /workdir/build/zephyr && /usr/bin/cmake -E true
/opt/toolchains/zephyr-sdk-0.16.5-1/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(sample_usbd_init.c.obj):(._usbd_contex.static.sample_usbd_+0x18): undefined reference to `__device_dts_ord_504'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /workdir/build
```

The behavior has been verified on a second MIMXRT1170-EVK dev board.

Not using `usbd_next_prj.conf` works fine. The firmware is built and shows Mass Storage device when flashed and connected to PC. Serial output reads:
```
*** Booting Zephyr OS build v3.7.0-rc1-465-g08496ff6b449 ***
Mount /SD:: 0
/SD:: bsize = 512 ; frsize = 32768 ; blocks = 121456 ; bfree = 121455
/SD: opendir: 0
End of files
[00:00:01.077,000] <inf> main: The device is put in USB mass storage mode.
```

jfischer-no, 2024-06-27:
I have observed main stack overflow on some platforms, please check if "CONFIG_MAIN_STACK_SIZE=2048" could fix your issue.
`west build -p -b mimxrt1170_evk/mimxrt1176/cm7 zephyr/samples/subsys/usb/mass/ -DCONF_FILE="usbd_next_prj.conf" -DCONFIG_APP_MSC_STORAGE_SDCARD=y -DCONFIG_MAIN_STACK_SIZE=2048`

jgenssler-GiN, 2024-06-27:
@jfischer-no Unfortunately, that doesn't fix the issue. I'm still getting the same error.

jfischer-no, 2024-06-27:
> @jfischer-no Unfortunately, that doesn't fix the issue. I'm still getting the same error.

okay, btw it works for me on mimxrt1050_evk/mimxrt1052, maybe some platform issue, FYI @MarkWangChinese

MarkWangChinese, 2024-07-03:
The USB next device stack is not enabled on MIMXRT1170-EVK, the clock and device tree (usb phy) is not enabled. I create the pr https://github.com/zephyrproject-rtos/zephyr/pull/75375, but I cannot find any RT1170-EVK to verify it. Thanks @hakehuang to help me to test it.

dleach02, 2024-07-07:
Thanks @MarkWangChinese

PR #75375 (2024-07-08): boards: nxp: enable mcux udc on RT1170-EVK
enable clock and usb phy device tree

Fixes: #75100
closed 2024-07-08 by aescolar