zephyrproject-rtos/zephyr #75119: Doc: Graphviz fontcolor switched between light and dark themes

Thalley, 2024-06-27:
**Is your enhancement proposal related to a problem? Please describe.**
When generating figures using `graphviz`, the fontcolor stays the same when switching between light and dark theme. Since it does not switch, all `graphviz` will have to rely on a fillcolor like white. 

**Describe the solution you'd like**
The color of of the font in `graphviz` should follow the same color as the rest of the documentation, unless specifically specified by the figure. 

**Describe alternatives you've considered**
N/A

**Additional context**

[image]
[image]

kartben, 2025-03-01:
Will probably give https://github.com/astropy/sphinx-automodapi/issues/167#issuecomment-1706737180 a try

kartben, 2025-03-01:
See #86509 -- now looks like:

[image]

PR #86509 (2025-03-05): doc: make graphviz diagrams look good in dark theme
Add CSS filter to invert colors in dark mode and make graphviz diagram more aligned with the overall look&feel of the page.

https://builds.zephyrproject.io/zephyr/pr/86509/docs/connectivity/bluetooth/api/audio/bluetooth-le-audio-arch.html#le-audio-stack
https://builds.zephyrproject.io/zephyr/pr/86509/docs/hardware/peripherals/dma.html#channel-state-machine-expectations

[image]

fixes #75119
PR #86795 (2025-05-09): [Backport v4.0-branch] doc: make graphviz diagrams look good in dark theme
Backport fd919b5160d0c3a3b3722dac798b89c1fac1b804 from #86509.
Fixes #75119.
closed 2025-03-05 by fabiobaltieri