zephyrproject-rtos/zephyr #75120: drivers: flash: common: drivers.flash.common.default fails

katgiadla, 2024-06-27:
**Describe the bug**
The test `tests/drivers/flash/common/drivers.flash.common.default` fails.

Observed for `nrf5340dk/nrf5340/cpuapp`.

**To Reproduce**
Steps to reproduce the behavior:
1. have `nrf5340dk/nrf5340/cpuapp` connected
2. go to your zephyr dir
3. call `./scripts/twister -T tests/drivers/flash/common -p nrf5340dk/nrf5340/cpuapp --device-testing --device-serial /dev/ttyACM1 -v --inline-logs --west-flash="--erase"`
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
```
*** Booting Zephyr OS build v3.7.0-rc1-374-gde69ebd1f888 ***
Running TESTSUITE flash_driver
===================================================================
Test will run on device mx25r6435f@0
START - test_flash_fill
Assertion failed at ../../../../../../../zephyr/tests/drivers/flash/common/src/main.c:210: flash_driver_test_flash_fill: (rc not equal to 0)
Fill failed
FAIL - test_flash_fill in 1.083 seconds
===================================================================
START - test_flash_flatten
Assertion failed at ../../../../../../../zephyr/tests/drivers/flash/common/src/main.c:242: flash_driver_test_flash_flatten: (rc not equal to 0)
Fill failed
FAIL - test_flash_flatten in 1.076 seconds
===================================================================
START - test_read_unaligned_address
Assertion failed at ../../../../../../../zephyr/tests/drivers/flash/common/src/main.c:150: flash_driver_test_read_unaligned_address: (rc not equal to 0)
Cannot write to flash
FAIL - test_read_unaligned_address in 1.083 seconds
===================================================================
TESTSUITE flash_driver failed.
------ TESTSUITE SUMMARY START ------
SUITE FAIL -   0.00% [flash_driver]: pass = 0, fail = 3, skip = 0, total = 3 duration = 3.242 seconds
- FAIL - [flash_driver.test_flash_fill] duration = 1.083 seconds
- FAIL - [flash_driver.test_flash_flatten] duration = 1.076 seconds
- FAIL - [flash_driver.test_read_unaligned_address] duration = 1.083 seconds
------ TESTSUITE SUMMARY END ------
===================================================================
RunID: 53bf18cbf4ae57cd2c50af45a080de8c
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain: Zephyr SDK 0.16.5-1
 - Commit SHA or Version used: `v3.7.0-rc1-374-gde69ebd1f888`

**Additional context**
Probably introduced by 3578753c2ee9bc44e825b358086c0d39eab8daca

katgiadla, 2024-07-04:
The same issue has occurred on `nrf52840dk/nrf52840` on version `v3.7.0-rc2-61-g6700a154be3c`.

de-nordic, 2024-07-10:
At `qspi_nor_write` the `qspi_nor_write_protection_set`->`qspi_send_cmd`->`nrfx_qspi_cinstr_xfer` fails with code `NRFX_ERROR_TIMEOUT` (0x0BAD0007)

closed 2024-09-30 by nordic-piks