zephyrproject-rtos/zephyr #75150: Demo project `sensor_shell` not fitting in `stm32h7sXi` mcu family

AlexFabre, 2024-06-28:
**Describe the bug**

The demo example [sensor_shell](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/sensor/sensor_shell) does not build with eval board `stm32h7s78_dk`

The problem comes from the size of the built binary, which is too big to fit on the flash for that mcu familly. 

**`STM32h7sXi` mcu familly comes with 64KB of flash, and 620kB of RAM.**

**Compiled `sensor_shell` binary for this board is ~79kB**

> In the following extract I manually increased the `flash0` area from 64kB to 96kB, only to output the binary size:
> 
> ~~~sh
> Memory region         Used Size  Region Size  %age Used
>            FLASH:       79564 B        96 KB     80.94%
>              RAM:       18240 B       128 KB     13.92%
>            SRAM1:          0 GB        16 KB      0.00%
>            SRAM2:          0 GB        16 KB      0.00%
>             DTCM:          0 GB       128 KB      0.00%
>             ITCM:          0 GB        64 KB      0.00%
>           EXTMEM:          0 GB       256 MB      0.00%
>         IDT_LIST:          0 GB        32 KB      0.00%
> ~~~

**To Reproduce**

~~~sh
west build -b stm32h7s78_dk samples/sensor/sensor_shell
~~~

**Expected behavior**

Have a demo example that flash and runs.

**Impact**

For anyone that wants to start developing on that target, not having a basic example running can be frustrating.

closed 2024-06-28 by nordicjm

AlexFabre, 2024-06-28:
Hi @nordicjm !

Nice to meet you too.

Before closing this issue, I would have appreciate to explain why I put the "bug" label on this issue.

Zephyr RTOS comes with a wide variety of examples that should be working out-of-the-box. This is what Zephyr promises...
Those examples are not coming from third party vendors, they are actually inside the project's repository.

You are right this is not a bug in the Zephyr RTOS itself, but if you look at it with the eyes of someone new, starting a Zephyr project for the firs time, on that evaluation board, happy to see that Zephyr proposes some example for his board, the frustration of not having something building properly can be really disappointing.

If you still think that this issue has to be closed, I would like to hear why. 

If you only think the problem comes from the issue label, then I invite you to create an issue to discuss why we should have a different "bug" label for Zephyr samples that do not build.

nordicjm, 2024-07-01:
> Zephyr RTOS comes with a wide variety of examples that should be working out-of-the-box. This is what Zephyr promises...
> Those examples are not coming from third party vendors, they are actually inside the project's repository.

They are working

> You are right this is not a bug in the Zephyr RTOS itself, but if you look at it with the eyes of someone new, starting a Zephyr project for the firs time, on that evaluation board, happy to see [...]

The board you are using does not have the resources to run it, there is nothing that can be done about that.