zephyrproject-rtos/zephyr #75164: tfm: Build issues with some tests (posix, newlib)

erwango, 2024-06-28:
**Describe the bug**

Build issues could be seen when building ns targets 

See https://github.com/zephyrproject-rtos/zephyr/actions/runs/9709152909/job/26797411658?pr=74627
```
INFO    - 1) tests/posix/common/portability.posix.common.newlib on b_u585i_iot02a/stm32u585xx/ns error (Build failure)
INFO    - 2) tests/posix/common/portability.posix.common.tls.newlib on b_u585i_iot02a/stm32u585xx/ns error (Build failure)
INFO    - 3) tests/posix/fs/portability.posix.fs.newlib on b_u585i_iot02a/stm32u585xx/ns error (Build failure)
INFO    - 4) tests/posix/fs/portability.posix.fs.tls.newlib on b_u585i_iot02a/stm32u585xx/ns error (Build failure)
INFO    - 5) tests/posix/headers/portability.posix.headers.newlib.with_posix_api on b_u585i_iot02a/stm32u585xx/ns error (Build failure)

INFO    - 1) tests/lib/c_lib/thrd/libraries.libc.c11_threads.newlib on b_u585i_iot02a/stm32u585xx/ns error (Build failure)
INFO    - 2) tests/lib/c_lib/thrd/libraries.libc.c11_threads.newlib_nano on b_u585i_iot02a/stm32u585xx/ns error (Build failure)
```

This is not specific to the board itself as the same errors could be seen with other ns target (`nrf9160dk/nrf9160/ns` for instance)

**To Reproduce**
`west twister -p nrf9160dk/nrf9160/ns -s tests/lib/c_lib/thrd/libraries.libc.c11_threads.newlib`

**Expected behavior**
Compilation ok.

**Impact**
Blocker

**Logs and console output**
<details open>
  <summary>`west twister -p nrf9160dk/nrf9160/ns -s tests/lib/c_lib/thrd/libraries.libc.c11_threads.newlib`</summary>

  In file included from /local/home/frq07517/zephyrproject/zephyr/include/zephyr/posix/time.h:116,
                 from /local/home/frq07517/zephyrproject/zephyr/include/zephyr/posix/time.h:12,
                 from /local/home/frq07517/zephyrproject/zephyr/include/zephyr/net/socket_select.h:22,
                 from /local/home/frq07517/zephyrproject/zephyr/include/zephyr/posix/sys/select.h:10,
                 from /local/home/frq07517/zephyr_sdk/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h:50,
                 from /local/home/frq07517/zephyr_sdk/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:61,
                 from /local/home/frq07517/zephyrproject/zephyr/twister-out/nrf9160dk_nrf9160_ns/tests/lib/c_lib/thrd/libraries.libc.c11_threads.newlib/tfm/api_ns/interface/include/mbedtls/platform.h:58,
                 from /local/home/frq07517/zephyrproject/modules/crypto/mbedtls/library/sha256.c:59:
/local/home/frq07517/zephyr_sdk/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/time.h:56:1: error: unknown type name 'clock_t'
   56 | clock_t    clock (void);
      | ^~~~~~~
/local/home/frq07517/zephyr_sdk/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/time.h:30:1: note: 'clock_t' is defined in header '<time.h>'; did you forget to '#include <time.h>'?
   29 | #include <sys/timespec.h>
  +++ |+#include <time.h>
   30 | 
In file included from /local/home/frq07517/zephyrproject/zephyr/include/zephyr/posix/posix_types.h:15,
                 from /local/home/frq07517/zephyrproject/zephyr/include/zephyr/posix/time.h:61:
/local/home/frq07517/zephyr_sdk/zephyr-sdk-0.16.8/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_pthreadtypes.h:182:3: error: unknown type name 'clock_t'
  182 |   clock_t  clock;             /* specifiy clock for timeouts */
      |   ^~~~~~~
/local/home/frq07517/zephyrproject/zephyr/include/zephyr/posix/posix_types.h:76:9: error: unknown type name 'clockid_t'
   76 |         clockid_t clock;
      |         ^~~~~~~~~
[...]
/local/home/frq07517/zephyrproject/zephyr/include/zephyr/fs/fs.h:409:36: error: unknown type name 'off_t'; did you mean '_off_t'?
  409 | int fs_seek(struct fs_file_t *zfp, off_t offset, int whence);
      |                                    ^~~~~
      |                                    _off_t
[...]
/local/home/frq07517/zephyrproject/zephyr/include/zephyr/sys/fdtable.h:48:17: error: expected specifier-qualifier-list before 'ssize_t'
   48 |                 ssize_t (*read)(void *obj, void *buf, size_t sz);
      |                 ^~~~~~~
[...]

</details>

**Environment (please complete the following information):**
 - OS: Ubuntu, Zephyr CI
 - Toolchain (e.g Zephyr SDK, ...): 0.16.8
 - Commit SHA or Version used: v3.7.0-rc1-524-g5e2fbe6d180

erwango, 2024-06-28:
@aescolar, @ithinuel, @Vge0rge FYI

aescolar, 2024-07-01:
Thanks @erwango 
Same issue as #75205 . Closing on that others issue favor

closed 2024-07-01 by aescolar