zephyrproject-rtos/zephyr #75178: MIMXRT1160_EVK touch screen not working

RafalHalik, 2024-06-28:
**Describe the bug**
When using mimxrt1160_evk with the rk055hdmipi4ma0 display. The I2C setup fails not allowing the touch screen to function.

Currently working with the mimxrt1160_evk board with rk055hdmipi4ma0 display i was testing out the screen and its functionality. I've come across an issue when using the gt911 driver for the touch screen. Using the setup from mimxrt1170_evk as recommedned by the zephyr docs the driver does not get setup correctly,  When the I2C request is sent it seems to fails almost immediately, attached logic analyser output of this fail : image on left is fail image on right is expected result (obtained from MCUXpresso ide lvgl sample).

[image]
[image]

Closer look at green dot: 
[image] [image]

MIPI connector DTS definition (Think the issue might be here). The GPIO are defined correctly as this was checked against MCUXpresso IDE board.h definitions of the same connector/pins. 

	nxp_mipi_connector: mipi-connector {
		compatible = "gpio-nexus";
		#gpio-cells = <2>;
		gpio-map-mask = <0xffffffff 0xffffffc0>;
		gpio-map-pass-thru = <0 0x3f>;
		gpio-map =	<0  0 &gpio9 29 0>,		/* Pin 1, LEDK */
					<21 0 &gpio9 1  0>,		/* Pin 21, RESET */
					<22 0 &gpio9 4  0>,		/* Pin 22, LPTE */
					<26 0 &gpio6 4  0>,		/* Pin 26, CTP_I2C SDA */
					<27 0 &gpio6 5  0>,		/* Pin 27, CTP_I2C SCL */
					<28 0 &gpio9 0  0>,		/* Pin 28, CTP_RST */
					<29 0 &gpio8 31 0>,		/* Pin 29, CTP_INT */
					<32 0 &gpio11 16 0>,	        /* Pin 32, PWR_EN */
					<34 0 &gpio9 29 0>;		/* Pin 34, BL_PWM */
	};

**To Reproduce**
Steps to reproduce the behavior:
1. build lvgl example for mimxrt1160_evk with screen rk055hdmipi4ma0
 `west build -b mimxrt1160_evk/mimxrt1166/cm7 -- -DSHIELD=rk055hdmipi4ma0`
2. Open minicom or putty terminal to dev board
 `minicom -D /dev/ttyACM0`
4. flash device 
 `west flash`
5. watch terminal I2C error will appear
 `[00:00:00.091,000] <err> gt911: Device did not respond to I2C request`

Current work around:
Within the GT911 driver init function I've added the setup from MCUXpresso for the I2C5 pins and clocks. 
file location: zephyr/drivers/input/input_gt911.c

    
    static int gt911_init(const struct device *dev){
 
        IOMUXC_SetPinMux( IOMUXC_GPIO_AD_00_GPIO8_IO31,           /* GPIO_AD_00 is configured as GPIO8_IO31 */
        1U);                                    
        IOMUXC_SetPinMux( IOMUXC_GPIO_AD_01_GPIO9_IO00,           /* GPIO_AD_01 is configured as GPIO9_IO00 */
        0U);                                    
        IOMUXC_SetPinMux( IOMUXC_GPIO_AD_02_GPIO9_IO01,           /* GPIO_AD_02 is configured as GPIO9_IO01 */
        0U);                                  
        IOMUXC_SetPinMux( IOMUXC_GPIO_AD_30_GPIO9_IO29,           /* GPIO_AD_30 is configured as GPIO9_IO29 */
        0U);                                    
        IOMUXC_SetPinMux( IOMUXC_GPIO_DISP_B2_15_GPIO11_IO16,     /* GPIO_DISP_B2_15 is configured as GPIO11_IO16 */
        0U);                                    
        IOMUXC_SetPinMux( IOMUXC_GPIO_LPSR_04_LPI2C5_SDA,         /* GPIO_LPSR_04 is configured as LPI2C5_SDA */
        1U);                                    
        IOMUXC_SetPinMux( IOMUXC_GPIO_LPSR_05_LPI2C5_SCL,         /* GPIO_LPSR_05 is configured as LPI2C5_SCL */
        1U);                                  
        IOMUXC_SetPinConfig( IOMUXC_GPIO_AD_00_GPIO8_IO31,           /* GPIO_AD_00 PAD functional properties : */
        0x00U);     

	    IOMUXC_SetPinConfig( IOMUXC_GPIO_LPSR_04_LPI2C5_SDA, /* GPIO_LPSR_04 PAD functional properties : */
		0x20U);

	    IOMUXC_SetPinConfig( IOMUXC_GPIO_LPSR_05_LPI2C5_SCL, /* GPIO_LPSR_05 PAD functional properties : */
		0x20U);
 

	    lpi2c_master_config_t lpi2cConfig = {0};

    
        lpi2cConfig.debugEnable = false;
        lpi2cConfig.ignoreAck = false;
        lpi2cConfig.pinConfig = kLPI2C_2PinOpenDrain;
        lpi2cConfig.baudRate_Hz = 100000U;
        lpi2cConfig.busIdleTimeout_ns = 0;
        lpi2cConfig.pinLowTimeout_ns = 0;
        lpi2cConfig.sdaGlitchFilterWidth_ns = 0;
        lpi2cConfig.sclGlitchFilterWidth_ns = 0;
    

	LPI2C_Type *pointer = 0x40C34000u;
	LPI2C_MasterInit(pointer, &lpi2cConfig, CLOCK_GetRootClockFreq(kCLOCK_Root_Lpi2c5)); 

**After changes**

With the code changes and setting gt911 to polling rather than interrupt based the touch screen is now functional.

From these changes I concluded that the DTS files are not used in the kernel initialisation process correctly or completely omitted. As far as I can tell the set up I have done manual in the gt911 init function should be identical to what the DTS would do if it were working.

**Impact**

requires changes which are unmaintainable.

**Environment (please complete the following information):**
 - OS: Linux
 - Zephyr SDK
 - version 

**Additional context**
FULL ANALYZER FILES: [zephyr logic anayzler.zip](https://github.com/user-attachments/files/16032758/zephyr.logic.anayzler.zip)

DerekSnell, 2024-06-28:
Hi @RafalHalik ,
Thank you for sharing all the details.  I think I see the issue.  Your workaround to fix this involved setting the pinmux settings.  Looking at the pinctrl settings, I think there is a typo.  Can you try correcting this line in [mimxrt1160_evk.dtsi](https://github.com/zephyrproject-rtos/zephyr/blob/9a08fdc1df5c000b54a1df2ec014207df282768d/boards/nxp/mimxrt1160_evk/mimxrt1160_evk.dtsi#L70C2-L70C31)?  Set to `pinmux_lpi2c5`
```
&lpi2c5 {
	status = "okay";
        pinctrl-0 = <&pinmux_lpi2c1>;   // fix this line
```

I can submit a PR to update this.  Please confirm if this resolves the issue.  Thanks

RafalHalik, 2024-07-01:
Hey @DerekSnell ,
Thanks for the quick reply! Great spot did not even notice this before. Unfortunately this change does not fix the issue, I still need to set up the i2c line manually.

RafalHalik, 2024-07-01:
I have also checked i2c bus using the i2c_shell commands without my additional code but with the typo fix. No devices are found on the i2c@40c34000 bus further making me think the setup commands are omitted by the init phase of zephyr: 
[image]

However with the changes I mentioned previously I can see 2 devices.
[image]

DerekSnell, 2024-07-01:
Thanks, @RafalHalik 
I will take a deeper look.

DerekSnell, 2024-07-01:
I also prepped this draft PR https://github.com/zephyrproject-rtos/zephyr/pull/75294 to enable the display shields for the mimxrt1160_evk.  After https://github.com/zephyrproject-rtos/zephyr/pull/75193 merges, I will submit it.  But since the merge window is currently closed for the v3.7.0 release, likely that PR will not merge until after the release.

PR #75193 (2024-07-04): boards: nxp: mimxrt1160_evk: fix LPI2C pinmux
Fixes typos in pinctrl pinmux nodes

Fixes #75178 .
closed 2024-07-04 by aescolar