zephyrproject-rtos/zephyr #75179: USB-HS not working in stm32h747i_disco with device_next

ddavidebor, 2024-06-28:
**Describe the bug**
USB HS is not working with https://docs.zephyrproject.org/latest/boards/st/stm32h747i_disco/doc/index.html 
CDC-ACM sample for example appears to fail silently somewhere after the USB is initialized.

**To Reproduce**
Run CDC-ACM sample using the device-next overlay. Plug in computer and verify operation. 
The device will not work in your computer.

**Impact**
Currently impossible to develop on the aforementioned platform, makin 

**Environment (please complete the following information):**
 - OS: MacOS and Ubuntu
 - Toolchain: Zephyr SDK 0.16.5-1
 - 3.7.0-rc1

henrikbrixandersen, 2024-06-28:
CC: @erwango

henrikbrixandersen, 2024-06-28:
Please re-test with https://github.com/zephyrproject-rtos/zephyr/pull/75129

ddavidebor, 2024-06-29:
I'm afraid the result is the same

This builds successfully, and works correctly:

```sh
west build --pristine -b stm32h747i_disco/stm32h747xx/m7 samples/subsys/usb/cdc_acm
```

This is not recognised by the computer, and displays no log when the USB host is connected or disconnected, and the device doesn't appear with MacOS system report / lsusb

```sh
west build --pristine -b stm32h747i_disco/stm32h747xx/m7 samples/subsys/usb/cdc_acm -DCONF_FILE=usbd_next_prj.conf
```

erwango, 2024-07-01:
@loicpoulain Would you mind having a look ?

henrikbrixandersen, 2024-07-02:
I have tried to reproduce this error this on the `stm32h735g_disco` board using 768b8bbca30ab78f2fed33272b695434288727d3, but it seems to work fine?

- `west build -b stm32h735g_disco/stm32h735xx samples/subsys/usb/cdc_acm -- -DCONF_FILE="usbd_next_prj.conf"`
- `west flash`

Console log:
```
*** Booting Zephyr OS build v3.7.0-rc2-45-g768b8bbca30a ***
[00:00:00.010,000] <inf> cdc_acm_echo: USB device support enabled
[00:00:00.010,000] <inf> cdc_acm_echo: Wait for DTR
[00:00:00.016,000] <inf> cdc_acm_echo: USBD message: Device suspended
[00:00:00.185,000] <inf> cdc_acm_echo: USBD message: Bus reset
[00:00:07.196,000] <inf> cdc_acm_echo: USBD message: CDC ACM line coding
[00:00:07.196,000] <inf> cdc_acm_echo: Baudrate 9600
```

Linux dmesg logs:
```
[53097.516088] usb 2-3.1: new full-speed USB device number 23 using xhci_hcd
[53097.620789] usb 2-3.1: New USB device found, idVendor=2fe3, idProduct=0001, bcdDevice= 3.07
[53097.620801] usb 2-3.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[53097.620803] usb 2-3.1: Product: USBD CDC ACM sample
[53097.620805] usb 2-3.1: Manufacturer: Zephyr Project
[53097.620807] usb 2-3.1: SerialNumber: 383233313139511900120037
[53097.623414] cdc_acm 2-3.1:1.0: ttyACM1: USB ACM device
```

This is likely specific to the `stm32h747i_disco` board and not the driver?

jfischer-no, 2024-07-02:
> This is likely specific to the `stm32h747i_disco` board and not the driver?

stm32h747i_disco/stm32h747xx/m7 has external ulpi phy, the issue is likely in the driver
https://github.com/zephyrproject-rtos/zephyr/blob/d0474cefde27c5fc115a570e388ceec99db53fbd/drivers/usb/udc/udc_stm32.c#L855-L856

https://github.com/zephyrproject-rtos/zephyr/blob/d0474cefde27c5fc115a570e388ceec99db53fbd/drivers/usb/udc/udc_stm32.c#L1020-L1044

FRASTM, 2024-07-19:
I suspect something around the IRQ: when one global irq is configured in the usbotg_hs node (of the stm32h745.dtsi)
```
			interrupts = <77 0>;
			interrupt-names = "otghs";
```

The plug in the  usb_otg_hs connector is detected :
```
*** Booting Zephyr OS build v3.7.0-rc2-455-g67544a6d2df3 ***                                                         
[00:00:00.010,000] <inf> cdc_acm_echo: USB device support enabled                                                    
[00:00:00.010,000] <inf> cdc_acm_echo: Wait for DTR                                                                  
[00:00:00.016,000] <inf> cdc_acm_echo: USBD message: Device suspended                                                
[00:00:01.726,000] <inf> cdc_acm_echo: USBD message: Bus reset                                                       
[00:00:01.807,000] <inf> cdc_acm_echo: USBD message: Bus reset                                                       
[00:00:01.907,000] <inf> cdc_acm_echo: USBD message: Bus reset                                                       
[00:00:01.986,000] <inf> cdc_acm_echo: USBD message: Bus reset                                                       
[00:00:02.077,000] <wrn> udc: Spurious suspend/resume event                                                          
[00:00:02.077,000] <inf> cdc_acm_echo: USBD message: Device suspended                                                
[00:00:02.402,000] <inf> cdc_acm_echo: USBD message: Bus reset                                                       
[00:00:02.423,000] <inf> cdc_acm_echo: USBD message: Bus reset                                                       
[00:00:02.711,000] <inf> cdc_acm_echo: USBD message: Bus reset                                                       
[00:00:02.802,000] <wrn> udc: Spurious suspend/resume event                                                          
[00:00:02.802,000] <inf> cdc_acm_echo: USBD message: Device suspended 
```

closed 2024-07-30 by fabiobaltieri

ddavidebor, 2024-07-31:
Thank you guys !