zephyrproject-rtos/zephyr #75196: Logger on ESP32C3 does not work

alkurop, 2024-06-28:
`zephyr-sdk-0.16.8`, `Zephyr OS build v3.7.0-rc1-313-g895f246227e2`, `Esp32c3-devkitm`

I am trying to use logger subsys. I have added a flag, and initialised Logger in the module. However, nothing is printed in the serial monitor.

'prj.conf': `CONFIG_LOG=y`
'main.c': 
```
#include <zephyr/logging/log.h>
LOG_MODULE_REGISTER(APP);

int main(void){
 while (1)
  {
        LOG_INF("Hello logger");
   }

}

```

sylvioalves, 2024-06-30:
@alkurop Hi, it does not sounds that there is any issue in there, at least it is what I think at the moment. The suggested sample is flooding the Logging subsystem, make it unable to print anything.  
1) If you replaced LOG_INF to printf, will that work?
2) If you add `k_msleep(10)`, will it work?
3) Finally, making it immediate will force loggind subsystem to output it a bit faster: `CONFIG_LOG_MODE_IMMEDIATE=y`

Please, let me know if any of this is helpful.

alkurop, 2024-07-01:
Hi. I placed the delay, added flag `CONFIG_LOG_MODE_IMMEDIATE=y`, and also tried `printf`. Unfortunately, it did not help. I also cannot see `Booting Zephyr` message with log module enabled

alkurop, 2024-07-10:
I connect to espc3c3_devkitm1 dev board with the USB wire. There are 2 ports on the board - USB port and COM port. I connect to USB port, flash, and then connect monitor 
```
 west espressif monitor --p /dev/cu.usbmodem1114101 
 west espressif monitor --p /dev/tty.usbmodem1114101 
  ```
  Neither printing works

sylvioalves, 2024-07-10:
Would you mind checking changing the default output port to `usb_serial`?
In this file: https://github.com/zephyrproject-rtos/zephyr/blob/main/boards/espressif/esp32c3_devkitm/esp32c3_devkitm.dts#L19-L20

```c
		zephyr,console = &usb_serial;
		zephyr,shell-uart = &usb_serial;
```

marekmatej, 2024-07-10:
@alkurop to use USB serial you need to change the default serial/console driver at the `dts` level as Sylvio suggested! 
If your primary interest is testing the logging subsystem, I suggest to use the USB port labeled as COM, and use it for flashing and for the console/shell. Tip. You might want to enable the shell adding `-DCONFIG_SHELL=y` to your `west build...`  command.

alkurop, 2024-07-10:
After changing the chosen console to `usb_serial`, logger works.

closed 2024-07-10 by alkurop

sylvioalves, 2024-07-10:
@alkurop Nice to hear. Sorry I missed asking/suggesting that before.