zephyrproject-rtos/zephyr #75211: doc: Documentation issue in 'connectivity/networking/api/net_config'

LysandrosAn, 2024-06-30:
Following configuration parameters:
 CONFIG_NET_CONFIG_PEER_IPV4_ADDR,
CONFIG_NET_CONFIG_MY_IPV4_ADDR,
CONFIG_BROKER_IP

return  "_no results_" in the Zephyr documentation. Also, search engine results from these together with the word "Zephyr" doesn’t list the Zephyr website first, but various vendors. It would be good to include such commands in the official documentation, because they seem to be recognized by the software.

<< Please describe the issue here >>
<< You may also want to update the automatically generated issue title above. >>

**Environment**

* Page: `connectivity/networking/api/net_config`
* Version: main
* SHA-1: e3ff9930003cfbbaf1409d4aee51e39057358b0d

jukkar, 2024-07-01:
The `CONFIG_BROKER_IP` is not a config option that is found in upstream Zephyr.

closed 2024-07-03 by aescolar

FERASMANASEER, 2025-12-28:
CONFIG_NET_CONFIG_MY_IPV6_ADDR
CONFIG_NET_CONFIG_PEER_IPV6_ADDR
CONFIG_NET_CONFIG_MY_IPV4_ADDR
CONFIG_NET_CONFIG_MY_IPV4_NETMASK
CONFIG_NET_CONFIG_MY_IPV4_GW
CONFIG_NET_CONFIG_PEER_IPV4_ADDR

PR #75222 (2024-07-03): doc: net: Add info about misc network config options
Add information about these missing Kconfig options:
```
CONFIG_NET_CONFIG_MY_IPV6_ADDR
CONFIG_NET_CONFIG_PEER_IPV6_ADDR
CONFIG_NET_CONFIG_MY_IPV4_ADDR
CONFIG_NET_CONFIG_MY_IPV4_NETMASK
CONFIG_NET_CONFIG_MY_IPV4_GW
CONFIG_NET_CONFIG_PEER_IPV4_ADDR
```
Fixes #75211
commit 9abca618 (2024-07-03): doc: net: Add info about misc network config options

Add information about these missing Kconfig options:

CONFIG_NET_CONFIG_MY_IPV6_ADDR
CONFIG_NET_CONFIG_PEER_IPV6_ADDR
CONFIG_NET_CONFIG_MY_IPV4_ADDR
CONFIG_NET_CONFIG_MY_IPV4_NETMASK
CONFIG_NET_CONFIG_MY_IPV4_GW
CONFIG_NET_CONFIG_PEER_IPV4_ADDR

Fixes #75211