zephyrproject-rtos/zephyr #75247: Bluetooth: Start scanning leads to ZEPHYR FATAL ERROR 3

lukas-fwdev, 2024-07-01:
In a custom app (working fine with v3.6, now trying to use v3.7-rc2) I try to start scanning using the bt_le_scan_start() function.
This leads to a ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0

Full log (with CONFIG_BT_HCI_CORE_LOG_LEVEL_DBG):

`
D: key d68fa14d2f830796807a0e574a3ef42f
D: plaintext 55f95100000000000000000000000000
D: enc_data 41b7503d2c32bd7e96ed6ce8d479183f
D: 51:F9:55:50:B7:41
D: opcode 0x2005 param_len 6
D: buf 0x2004c290
D: buf 0x2004c290 opcode 0x2005 len 9
D: kick TX
ASSERTION FAIL [err == 0] @ WEST_TOPDIR/zephyr/subsys/bluetooth/host/hci_core.c:410
        Controller unresponsive, command opcode 0x2005 timeout with err -11
E: r0/a1:  0x00000003  r1/a2:  0x00000000  r2/a3:  0x00000001
E: r3/a4:  0x00000003 r12/ip:  0x00000000 r14/lr:  0x00021521
E:  xpsr:  0x21000000
E: s[ 0]:  0x00000009  s[ 1]:  0x00073bb1  s[ 2]:  0x2001ee68  s[ 3]:  0x2001ee68
E: s[ 4]:  0x00000000  s[ 5]:  0x00000001  s[ 6]:  0x2001ee78  s[ 7]:  0x2001ee78
E: s[ 8]:  0x0000001c  s[ 9]:  0x2004c290  s[10]:  0x2001eeb8  s[11]:  0x2000018d
E: s[12]:  0x2001ef08  s[13]:  0x000001f4  s[14]:  0x20037f40  s[15]:  0x20037f60
E: fpscr:  0x00000000
E: Faulting instruction address (r15/pc): 0x00021530
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x2001f0b0 (proto_backend)
E: Halting system

`
I am using a Nrf5340 Audio DK. As mentioned everything worked fine with v3.6 but not with the RC2 of v3.7.

hermabe, 2024-07-01:
As the log says, the controller is unresponsive and the command times out. Can you get the logs from the netcore? It is likely that it is stuck somewhere or has asserted.

lukas-fwdev, 2024-07-01:
I am not sure if this is what you are looking for:

`
*** Booting Zephyr OS build v3.7.0-rc2-10-g30cf18ada*** Booting Zephyr OS build v3.7.0-rc2-10-g30cf18adacc1 ***
[00:00:00.000,610] <dbg> hci_ipc: main: Start
[00:00:00.000,640] <dbg> bt_hci_raw: bt_enable_raw:
[00:00:00.005,737] <dbg> bt_ctlr_hci_driver: prio_recv_thread: sem take...
[00:00:00.005,767] <dbg> bt_ctlr_hci_driver: recv_thread: blocking
[00:00:00.005,798] <dbg> bt_ctlr_hci_driver: hci_driver_open: Success.
[00:00:00.005,828] <inf> bt_hci_raw: Bluetooth enabled in RAW mode
[00:00:00.027,282] <inf> hci_ipc: Received message of 4 bytes.
[00:00:00.027,313] <dbg> hci_ipc: hci_ipc_rx: IPC data:
                                  01 03 0c 00                                      |....
[00:00:00.027,343] <dbg> hci_ipc: hci_ipc_cmd_recv: len 0
[00:00:00.027,374] <dbg> bt_hci_raw: bt_send: buf 0x2100f56c len 3
[00:00:00.027,374] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.027,832] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 6 bytes
[00:00:00.027,862] <dbg> bt_hci_raw: bt_hci_recv: buf 0x2100e0a8 len 6
[00:00:00.027,893] <dbg> bt_ctlr_hci_driver: hci_driver_send: exit: 0
--- 7 messages dropped ---
[00:00:00.065,673] <inf> hci_ipc: Received message of 4 bytes.
[00:00:00.065,704] <dbg> hci_ipc: hci_ipc_rx: IPC data:
                                  01 03 10 00                                      |....
[00:00:00.065,734] <dbg> hci_ipc: hci_ipc_cmd_recv: len 0
--- 14 messages dropped ---
[00:00:00.102,691] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.102,722] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 14 bytes
[00:00:00.102,722] <dbg> bt_hci_raw: bt_hci_recv: buf 0x2100e0a8 len 14
[00:00:00.102,752] <dbg> bt_ctlr_hci_driver: hci_driver_send: exit: 0
[00:00:00.102,813] <dbg> hci_ipc: hci_ipc_rx: Final net buffer:
--- 7 messages dropped ---
[00:00:00.139,617] <dbg> bt_hci_raw: bt_send: buf 0x2100f56c len 3
[00:00:00.139,648] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.139,678] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 70 bytes
[00:00:00.139,709] <dbg> bt_hci_raw: bt_hci_recv: buf 0x2100e0a8 len 70
[00:00:00.139,739] <dbg> bt_ctlr_hci_driver: hci_driver_send: exit: 0
--- 6 messages dropped ---
[00:00:00.176,635] <dbg> hci_ipc: hci_ipc_rx: IPC data:
                                  01 18 20 00                                      |.. .
[00:00:00.176,666] <dbg> hci_ipc: hci_ipc_cmd_recv: len 0
[00:00:00.176,696] <dbg> bt_hci_raw: bt_send: buf 0x2100f56c len 3
--- 11 messages dropped ---
[00:00:00.213,806] <dbg> hci_ipc: hci_ipc_cmd_recv: len 0
[00:00:00.213,836] <dbg> bt_hci_raw: bt_send: buf 0x2100f56c len 3
[00:00:00.213,867] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.213,928] <dbg> hci_ipc: hci_ipc_rx: Final net buffer:
[00:00:00.214,233] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 14 bytes
--- 8 messages dropped ---
[00:00:00.249,664] <dbg> hci_ipc: hci_ipc_cmd_recv: len 0
[00:00:00.249,694] <dbg> bt_hci_raw: bt_send: buf 0x2100f56c len 3
[00:00:00.249,725] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.249,786] <dbg> hci_ipc: hci_ipc_rx: Final net buffer:
[00:00:00.250,030] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 14 bytes
--- 8 messages dropped ---
[00:00:00.285,491] <dbg> hci_ipc: hci_ipc_cmd_recv: len 0
[00:00:00.285,522] <dbg> bt_hci_raw: bt_send: buf 0x2100f56c len 3
[00:00:00.285,552] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.285,614] <dbg> hci_ipc: hci_ipc_rx: Final net buffer:
[00:00:00.285,858] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 14 bytes
--- 8 messages dropped ---
[00:00:00.321,319] <dbg> hci_ipc: hci_ipc_cmd_recv: len 0
[00:00:00.321,350] <dbg> bt_hci_raw: bt_send: buf 0x2100f56c len 3
[00:00:00.321,380] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.321,441] <dbg> hci_ipc: hci_ipc_rx: Final net buffer:
[00:00:00.321,624] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 14 bytes
--- 7 messages dropped ---
[00:00:00.358,428] <dbg> hci_ipc: hci_ipc_rx: IPC data:
                                  01 03 20 00                                      |.. .
[00:00:00.358,428] <dbg> hci_ipc: hci_ipc_cmd_recv: len 0
[00:00:00.358,459] <dbg> bt_hci_raw: bt_send: buf 0x2100f56c len 3
[... 67 lines cut ...]
[00:00:00.767,150] <dbg> bt_ctlr_hci_driver: hci_driver_send: exit: 0
[00:00:00.767,211] <dbg> hci_ipc: hci_ipc_rx: Final net buffer:
[00:00:00.767,242] <dbg> hci_ipc: hci_ipc_send: buf 0x2100e0a8 type 1 len 12
[00:00:00.767,272] <dbg> hci_ipc: hci_ipc_send: Controller buffer:
                                  0e 0a 01 09 10 00 00 00  00 00 00 00             |........ ....
[00:00:00.767,303] <dbg> hci_ipc: hci_ipc_send: Final HCI buffer:
                                  04 0e 0a 01 09 10 00 00  00 00 00 00 00          |........ .....
[00:00:00.767,364] <inf> hci_ipc: Sent message of 13 bytes.
[00:00:00.814,086] <inf> hci_ipc: Received message of 10 bytes.
[00:00:00.814,117] <dbg> hci_ipc: hci_ipc_rx: IPC data:
                                  01 05 20 06 a7 24 bb 69  81 f3                   |.. ..$.i ..
[00:00:00.814,147] <dbg> hci_ipc: hci_ipc_cmd_recv: len 6
[00:00:00.814,178] <dbg> bt_hci_raw: bt_send: buf 0x2100f56c len 9
[00:00:00.814,178] <dbg> bt_ctlr_hci_driver: hci_driver_send: enter
[00:00:00.814,208] <dbg> bt_ctlr_hci_driver: cmd_handle: Replying with event of 6 bytes
[... cut ...]
                                  2a 97 03 01 a8 e0                                |*.....
[00:00:00.814,361] <dbg> hci_ipc: hci_ipc_send: buf 0x2100e0a8 type 1 len 6
[00:00:00.814,392] <dbg> hci_ipc: hci_ipc_send: Controller buffer:
                                  0e 04 01 05 20 00                                |.... .
[00:00:00.814,422] <dbg> hci_ipc: hci_ipc_send: Final HCI buffer:
                                  04 0e 04 01 05 20 00                             |..... .
[00:00:00.814,453] <inf> hci_ipc: Sent message of 7 bytes.`

jori-nordic, 2024-07-01:
Did a test with the Bluetooth shell, didn't manage to reproduce the issue, scanning in a noisy environment:
https://github.com/zephyrproject-rtos/zephyr/pull/75277

lukas-fwdev, 2024-08-06:
I maybe have found something. When I try to start scanning, the random address should be set, this is the HCI command which is timing out. I am wondering if it can be related to the changes in PR #72090 ? I am calling bt_le_scan_start() from a user thread.

jori-nordic, 2024-08-06:
Could be related, yes.

> Full log (with CONFIG_BT_HCI_CORE_LOG_LEVEL_DBG)

This is not the full log. Could you post the full logs (attach as .txt) for both app and net cores? You can use 1Mbit in order to not drop packets.

lukas-fwdev, 2024-08-06:
[log_appcore.txt](https://github.com/user-attachments/files/16507897/log_appcore.txt)
[log_netcore.txt](https://github.com/user-attachments/files/16507898/log_netcore.txt)
Thanks for your quick answer. Here are the logs.

jori-nordic, 2024-08-19:
Sorry, got pre-empted for other things.

The logs are lacking time and thread info, could you add those:
```
CONFIG_THREAD_NAME=y
CONFIG_LOG_THREAD_ID_PREFIX=y
# Disable minimal logging, we want time & fn names
CONFIG_LOG_MODE_MINIMAL=n
CONFIG_LOG_MODE_DEFERRED=y
```

In the meantime, there have been some bugfixes that also apply to v3.7. If it's possible, could you try running your app on the latest `origin/main`?

jori-nordic, 2024-08-19:
I am not seeing this log line printed
https://github.com/jori-nordic/zephyr/blob/20e7ef389dc7d3d28adfe1a1d62f035a33f5eb91/subsys/bluetooth/host/hci_core.c#L4650

It's possible that the `tx_processor` work item is not run at all. It may be because the system workqueue is blocked on something else.
Is it possible to get a backtrace for all the threads when the `BT_ASSERT()` fails?

lukas-fwdev, 2024-09-05:
Sorry for my long silence, I was occupied by other things.
I now fixed something in my code, with handling the scan callbacks, and using v3.7_branch. Like this the issue is no longer present. So I will close the ticket.

closed 2024-09-05 by lukas-fwdev