zephyrproject-rtos/zephyr #75249: Bsim: Set log level from command-line argument

jori-nordic, 2024-07-01:
Created from 
https://github.com/zephyrproject-rtos/zephyr/pull/75154#discussion_r1658529168

> Not related to this PR specifically, but have we ever considered setting the log level at runtime for BSIM tests? We don't really care about memory in these, and would allow us to rerun a test with more verbosity without a recompile. The log level can simply be added as a argument in the run.sh (and similar) files.

A starting point would be implementing that for the sample bsim test. 
This way that pattern or library will automatically spread when people make new tests based on this sample.

https://github.com/zephyrproject-rtos/zephyr/blob/v3.7.0-rc2/tests/bsim/bluetooth/host/misc/sample_test/README.rst

00thirdeye00, 2024-07-08:
hej @jori-nordic, I am new to zephyr and would like to work on this issue. can you please guide me and give me some information/references for my understanding.

jori-nordic, 2024-07-09:
Hi, yes.
You should familiarize yourself with BabbleSim and read this page carefully first: https://docs.zephyrproject.org/latest/develop/test/bsim.html

Then also read all the files in the [sample test's directory](https://github.com/zephyrproject-rtos/zephyr/blob/v3.7.0-rc2/tests/bsim/bluetooth/host/misc/sample_test/README.rst).

Then look at how we add arguments to the tests: 
We [register](https://github.com/jori-nordic/zephyr/blob/0c8c28be109cf5a47b2165ef14201c833ec1f972/tests/bsim/bluetooth/audio/src/gmap_ugg_test.c#L1546-L1547) a parsing [function](https://github.com/jori-nordic/zephyr/blob/0c8c28be109cf5a47b2165ef14201c833ec1f972/tests/bsim/bluetooth/audio/src/gmap_ugg_test.c#L1505-L1539) that will set some global variables.
The arguments are passed on the command-line [this way](https://github.com/jori-nordic/zephyr/blob/0c8c28be109cf5a47b2165ef14201c833ec1f972/tests/bsim/bluetooth/audio/test_scripts/gmap_unicast_ac_11_ii.sh#L20)

The idea is to then:
- Use the maximum log level when registering the "main" test modules. Ie replace `CONFIG_APP_LOG_LEVEL` with `LOG_LEVEL_DBG` on [this line](https://github.com/jori-nordic/zephyr/blob/0c8c28be109cf5a47b2165ef14201c833ec1f972/tests/bsim/bluetooth/host/misc/sample_test/src/dut.c#L23-L24)
- Set the log level at runtime, using the global variable that was set by the parsing function. Remember to set a conservative default, e.g. `LOG_LEVEL_INF` if the argument is not specified on the cli. [This is an example](https://github.com/jori-nordic/zephyr/blob/0c8c28be109cf5a47b2165ef14201c833ec1f972/tests/bsim/bluetooth/host/att/long_read/main.c#L127-L129) of how to set the log level at runtime.

The [command line](https://github.com/jori-nordic/zephyr/blob/0c8c28be109cf5a47b2165ef14201c833ec1f972/tests/bsim/bluetooth/host/misc/sample_test/test_scripts/run.sh#L65) for setting e.g., `LOG_LEVEL_DBG` would look something like this then, for the sample test:
`Execute "${test_exe}" -v=${verbosity_level} -s=${simulation_id} -d=0 -rs=420 -testid=dut -argstest log_level 4`.

[Here are the](https://github.com/jori-nordic/zephyr/blob/0c8c28be109cf5a47b2165ef14201c833ec1f972/include/zephyr/logging/log_core.h#L16-L24) log level defines.

Happy hacking 💻

PR #75906 (2024-08-05): tests: bsim: set log level from command-line argument
Implementation to take command-line arguments to set log level in run time for bsim module

Fixes: #75249
closed 2024-08-05 by carlescufi