zephyrproject-rtos/zephyr #75286: OOT board fails to build after going from v3.5.0 to v3.7.0

ycsin, 2024-07-01:
**Describe the bug**
I encountered the following compilation error while testing build on v3.7.0 for OOT boards.

```
are/zephyr/__sources-3.7.0__/out/zephyr/subsys/debug/symtab/symtab_stub.c
riscv64-unknown-elf-gcc: fatal error: cannot read spec file 'picolibc.specs': No such file or directory
compilation terminated.
ninja: build stopped: subcommand failed.
```

This didn't happen in v3.5.0.

There's a similar report in the `#cpp` channel in our discord server, but I'm not sure what board they used.

https://discord.com/channels/720317445772017664/883444430676324382/1255783684322230282

[image]

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
     - OOT RISC-V boards with OOT SoCs
 - What have you tried to diagnose or workaround this issue? 
     - Using `MINIMAL_LIBC` can make the error to go away
 - Is this a regression? If yes, have you been able to "git bisect" it to a
   specific commit?
     - Kinda? It used to build in v3.5

**To Reproduce**
Probably not easy to reproduce with upstream SDK.

**Expected behavior**
I'm expecting my application to build on v3.7.0

**Impact**
Application doesn't build on v3.7.0

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Custom gcc toolchain
 - Commit SHA or Version used: 3.7.0-rc1

ycsin, 2024-07-01:
cc @keith-packard @stephanosio, LMK if you need more info, but it's a OOT board + SOC with a custom compiler so I can only try my best 🥲

keith-packard, 2024-07-01:
It looks like this is using the host compiler (see the reference to `/usr/bin/gcc`) with cross-compilation flags (including `-ffreestanding -fno-builtin --specs=picolibc.specs`) to preprocess something (`-E`). I can't see what it's trying to do beyond that from the included image.

ycsin, 2024-07-01:
> I can't see what it's trying to do beyond that from the included image.

Updated OP with my compilation logs

keith-packard, 2024-07-01:
You're not using the Zephyr SDK? If not, you'll need to make sure to `CONFIG_PICOLIBC_USE_MODULE=y`, which is not the default. If that doesn't work, then there's something messed up with the construction of the compiler flags for that particular compilation step.

PR #75298 (2024-07-05): lib/libc: do not enable `picolibc` if neither the toolchain has it nor its module is present
Currently, `PICOLIBC` is getting enabled even if neither the toolchain has it nor the picolibc module is available to the build system. This can be the case when Zephyr is built in OOT environment, with external toolchain and without `West`.

A possible workaround to the issue is to specify another LIBC (i.e. `CONFIG_MINIMAL_LIBC=y`) explicitly, but in any case, we probably do not want to compile the PICOLIBC interface when we do not have its source.

Fixes #75286
closed 2024-07-05 by aescolar