zephyrproject-rtos/zephyr #75288: [input] Add button gpio to analog axis input.

go2sh, 2024-07-01:
**Is your enhancement proposal related to a problem? Please describe.**
I have a resistive touch controller, which behaves like a two channel ADC. Additional the ADC has a "pressed" output, which goes active if a touch is detected. I would like to use the `analog-axis` input code in conjunction with the ADC, but the pressed information is missing.

**Describe the solution you'd like**
Add a gpio to the `analog-axis` device, which would represent the pressed information and pass this to the input system as well.

**Describe alternatives you've considered**
Creating a custom input driver, but this would result in a lot of repetitive code code.

fabiobaltieri, 2024-07-02:
That's a completely different type of device, it should have its own driver.

fabiobaltieri, 2024-11-06:
Closing down, not really relevant, if anyone wants to develop an adc based resistive touchscreen driver cool but then it's a different issue.

closed 2024-11-06 by fabiobaltieri