zephyrproject-rtos/zephyr #75318: Zephyr 3.7 RC2 BLE on  STM32 Nucleo WB55 does not work

piotrParf, 2024-07-02:
**Describe the bug**
I am using Zephyr 3.7 RC2 but the issue is vissible also with 3.7 RC1.
I am trying if everything still works for the STM32WB55 device. I am using STM32 Nucleo WB55. 
I checked radio firmware stm32wb5x_BLE_HCILayer_fw.bin in version 1.18.0 from STM32WB Copro Wireless Binaries.
The issue is visible with the radio version 1.19.1 and also with the firmware file stm32wb5x_BLE_HCILayer_extended_fw.bin.
Demos with BLE do not work on STM32 Nucleo WB55. I have also my own application but also demo iBeacon does not work.

The issue seems coming from the bt_ipm_ble_init function.

It was working with Zephyr 3.4 when I last checked. 

**To Reproduce**
Have radio firmware loaded with radio fw.
Build the "zephyr/samples/bluetooth/ibeacon" demo for -DBOARD=nucleo_wb55rg flash and run.

**Expected behavior**
Working demo and visible iBeacon node on BLE scanner. 

**Impact**
BLE on STM32 Nucleo WB55 does not work.

**Logs and console output**
When I read the log I got the below report:

```
*** Booting Zephyr OS build v3.7.0-rc1-588-g40810983ead2 ***
Starting iBeacon Demo
E: ***** USAGE FAULT *****
E:   Illegal use of the EPSR
E: r0/a1:  0x080094f4  r1/a2:  0x20002ac8  r2/a3:  0x00080000
E: r3/a4:  0x00000000 r12/ip:  0x20000224 r14/lr:  0x080035b1
E:  xpsr:  0x20000000
E: Faulting instruction address (r15/pc): 0x00000000
E: >>> ZEPHYR FATAL ERROR 35: Unknown error on CPU 0
E: Current thread: 0x20000490 (unknown)
E: Halting system
```

**Environment (please complete the following information):**
Linux Debian 11
CMake version: 3.21.0
Zephyr version: 3.7.0-rc2
west 1.2.0
host-tools: zephyr 0.16.1 
toolchain: zephyr 0.16.1

ajarmouni-st, 2024-07-02:
Hi @piotrParf, have you seen? https://github.com/zephyrproject-rtos/zephyr/blob/6700a154be3ca29be50fb4f1e35d899de69be5ca/boards/st/nucleo_wb55rg/doc/nucleo_wb55rg.rst?plain=1#L194-L196

piotrParf, 2024-07-02:
@ajarmouni-st 
As I mentioned in bug description I tried stm32wb5x_BLE_HCILayer_fw.bin and stm32wb5x_BLE_HCILayer_extended_fw.bin from packages in version 1.18.0 and 1.19.1. Both firmwares are HCI only. Firmware 1.18.0 works with Zephyr 3.4 (our current base which we want to migrate to new version).

erwango, 2024-07-02:
^^@asm5878

asm5878, 2024-07-04:
^^@JoeGitST

asm5878, 2024-07-04:
Thanks to @JoeGitST we have identified the commit causing the regression:
501e7158a8a0d270b032d7b672b313aa8af6d54e

Investigation is on going

asm5878, 2024-07-08:
^^ jhedberg

asm5878, 2024-07-08:
^^ @jhedberg

asm5878, 2024-07-08:
After the update of BLE HCI API drivers we have realized we have an issue in management of TX/RX HCI commands.
The first HCI cmd (any) is causing an usage fault.

Investigation will continue to clearly analyze the root cause.

jhedberg, 2024-07-08:
@asm5878 I took a quick look at the driver code, and it seems like the core issue is likely that the driver attempts to send HCI commands inside `bt_ipm_open()`. No driver should do that, since from the perspective of the host (which owns the `bt_hci_cmd_send*()` APIs) the HCI transport is functional only after the drivers's `open()` callback has successfully returned. What the driver should instead do is to implement a `setup()` callback (which gets called by the host immediately after `open()`) and perform any HCI command sending from within that callback instead.

jhedberg, 2024-07-08:
@asm5878 the exact failure is probably simply that the driver hasn't yet set its `hci->recv` callback pointer before it attempts to send HCI commands (`recv()` is needed to pass the command completion events to the host), however even if you fixed that it's not right for the driver to call any HCI command APIs from `open()` rather those should be moved into `setup()`.

jhedberg, 2024-07-08:
@asm5878 also, this is not the first HCI driver to have such a bug, take a look e.g. at #73711 for a recent similar fix.

jhedberg, 2024-07-10:
@asm5878 @erwango are you able to submit a fix based on my feedback? The fix itself should be fairly quick & simple but at least I'm not able create it as I'm on vacation this and next week. For the fix to make it to 3.7 it needs to be submitted latest today (two days review time and rc3 is coming on Friday).

asm5878, 2024-07-11:
Hello @jhedberg,
I was off for a couple of days. 
We will try our best (according our current priorities) to offer a solution ASAP.

Thanks for your precious feedback

FRASTM, 2024-07-17:
@asm5878 do you have a fix ready  ?

asm5878, 2024-07-22:
@FRASTM we have a potential fix. Unlocking the regression but with warning/errors.
I will check directly and I'll let you know

asm5878, 2024-07-26:
Dear All,

Status update:

We had some regressions due to Zephyr BLE updates.
Now we have a working patch. 
BLE works as expected. 
But still some issue just after connection:

Connected
[00:03:44.083,000] <wrn> bt_hci_core: opcode 0x2016 status 0x3a
[00:03:44.083,000] <err> bt_hci_core: Failed read remote features (-5)

Attached you can find a patch unlocking the BLE behavior (thanks @JoeGitST ).
[STM32WB_BLE_Unlock.patch](https://github.com/user-attachments/files/16393052/STM32WB_BLE_Unlock.patch)

M0 binary FW tested is:
https://github.com/STMicroelectronics/STM32CubeWB/blob/v1.19.1/Projects/STM32WB_Copro_Wireless_Binaries/STM32WB5x/stm32wb5x_BLE_HCILayer_fw.bin

jhedberg, 2024-08-05:
@erwango why did you reopen this?

jhedberg, 2024-08-05:
> @erwango why did you reopen this?

E.g. I don't think the backport requires this to be open, rather it's sufficient that there is evidence (i.e. this original issue) that there was a bug that needed fixing.

erwango, 2024-08-05:
> E.g. I don't think the backport requires this to be open, rather it's sufficient that there is evidence (i.e. this original issue) that there was a bug that needed fixing.

Oh, ok. I'll close it then. Thanks

PR #76577 (2024-08-05): STM32WB55  BLE not working
Implementing and enabling HCI setup function to have a correct and proper
initialization procedure to fix #75318 issue
commit 4fc77248 (2024-08-05): drivers: bluetooth: hci: Fix stm32wb BLE behavior

Enabling BT_HCI_SETUP for STM32WB55 to have a
correct and proper initialization procedure to fix
#75318 issue
commit 7ca2072e (2024-08-05): drivers: bluetooth: hci: Fix stm32wb BLE behavior

Implementing HCI setup function to have a correct and proper
initialization procedure to fix #75318 issue
commit d698d40b (2024-08-06): drivers: bluetooth: hci: Fix stm32wb BLE behavior

Implementing HCI setup function to have a correct and proper
initialization procedure to fix #75318 issue

(cherry picked from commit 7ca2072ed08c1d8def71b7038611a542dd3fe464)
closed 2024-08-05 by erwango