zephyrproject-rtos/zephyr #75319: `fatal error: sys/_timespec.h: No such file or directory` with v3.7.0-rc2

0x6e, 2024-07-02:
I've [started a discussion](https://github.com/zephyrproject-rtos/zephyr/discussions/75212) regarding this issue, but received no feedback so far.

**Describe the bug**
Compiling my application with v3.7.0-rc2 fails to build: 

```
FAILED: zephyr/lib/os/zvfs/CMakeFiles/lib__os__zvfs.dir/zvfs_eventfd.c.obj 
ccache /usr/bin/gcc -DKERNEL -DK_HEAP_MEM_POOL_SIZE=1024 -D__ZEPHYR__=1 -I<path>/build-native/zephyr/include/generated/zephyr -I<path>/zephyr/include -I<path>/build-native/zephyr/include/generated -I<path>/zephyr/soc/native/inf_clock -I<path>/zephyr/boards/native/native_sim -I<path>/zephyr/scripts/native_simulator/common/src/include -I<path>/zephyr/scripts/native_simulator/native/src/include -g -Wshadow -fno-strict-aliasing -Os -imacros <path>/build-native/zephyr/include/generated/zephyr/autoconf.h -fno-common -g -gdwarf-4 -fdiagnostics-color=always -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wdouble-promotion -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=<path>/slint/examples/printerdemo_mcu/zephyr=CMAKE_SOURCE_DIR -fmacro-prefix-map=<path>/zephyr=ZEPHYR_BASE -fmacro-prefix-map=<path>=WEST_TOPDIR -ffunction-sections -fdata-sections -m64 -fPIC -fvisibility=hidden -fno-freestanding -std=c11 -MD -MT zephyr/lib/os/zvfs/CMakeFiles/lib__os__zvfs.dir/zvfs_eventfd.c.obj -MF zephyr/lib/os/zvfs/CMakeFiles/lib__os__zvfs.dir/zvfs_eventfd.c.obj.d -o zephyr/lib/os/zvfs/CMakeFiles/lib__os__zvfs.dir/zvfs_eventfd.c.obj -c <path>/zephyr/lib/os/zvfs/zvfs_eventfd.c
In file included from <path>/zephyr/include/zephyr/toolchain.h:50,
                 from <path>/zephyr/include/zephyr/kernel_includes.h:23,
                 from <path>/zephyr/include/zephyr/kernel.h:17,
                 from <path>/zephyr/lib/os/zvfs/zvfs_eventfd.c:9:
<path>/zephyr/include/zephyr/posix/posix_types.h:51:21: error: ‘pthread_attr_t’ undeclared here (not in a function); did you mean ‘pthread_attr’?
   51 | BUILD_ASSERT(sizeof(pthread_attr_t) >= sizeof(struct pthread_attr));
      |                     ^~~~~~~~~~~~~~
<path>/zephyr/include/zephyr/toolchain/gcc.h:87:52: note: in definition of macro ‘BUILD_ASSERT’
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert((EXPR), "" MSG)
      |                                                    ^~~~
<path>/zephyr/include/zephyr/toolchain/gcc.h:87:51: error: expression in static assertion is not an integer
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert((EXPR), "" MSG)
      |                                                   ^
<path>/zephyr/include/zephyr/posix/posix_types.h:51:1: note: in expansion of macro ‘BUILD_ASSERT’
   51 | BUILD_ASSERT(sizeof(pthread_attr_t) >= sizeof(struct pthread_attr));
      | ^~~~~~~~~~~~
<path>/zephyr/include/zephyr/posix/posix_types.h:70:21: error: ‘pthread_mutexattr_t’ undeclared here (not in a function); did you mean ‘pthread_mutexattr’?
   70 | BUILD_ASSERT(sizeof(pthread_mutexattr_t) >= sizeof(struct pthread_mutexattr));
      |                     ^~~~~~~~~~~~~~~~~~~
<path>/zephyr/include/zephyr/toolchain/gcc.h:87:52: note: in definition of macro ‘BUILD_ASSERT’
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert((EXPR), "" MSG)
      |                                                    ^~~~
<path>/zephyr/include/zephyr/toolchain/gcc.h:87:51: error: expression in static assertion is not an integer
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert((EXPR), "" MSG)
      |                                                   ^
<path>/zephyr/include/zephyr/posix/posix_types.h:70:1: note: in expansion of macro ‘BUILD_ASSERT’
   70 | BUILD_ASSERT(sizeof(pthread_mutexattr_t) >= sizeof(struct pthread_mutexattr));
      | ^~~~~~~~~~~~
<path>/zephyr/include/zephyr/posix/posix_types.h:83:21: error: ‘pthread_condattr_t’ undeclared here (not in a function); did you mean ‘pthread_condattr’?
   83 | BUILD_ASSERT(sizeof(pthread_condattr_t) >= sizeof(struct pthread_condattr));
      |                     ^~~~~~~~~~~~~~~~~~
<path>/zephyr/include/zephyr/toolchain/gcc.h:87:52: note: in definition of macro ‘BUILD_ASSERT’
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert((EXPR), "" MSG)
      |                                                    ^~~~
<path>/zephyr/include/zephyr/toolchain/gcc.h:87:51: error: expression in static assertion is not an integer
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert((EXPR), "" MSG)
      |                                                   ^
<path>/zephyr/include/zephyr/posix/posix_types.h:83:1: note: in expansion of macro ‘BUILD_ASSERT’
   83 | BUILD_ASSERT(sizeof(pthread_condattr_t) >= sizeof(struct pthread_condattr));
      | ^~~~~~~~~~~~
<path>/zephyr/include/zephyr/posix/posix_types.h:107:21: error: ‘pthread_once_t’ undeclared here (not in a function); did you mean ‘pthread_once’?
  107 | BUILD_ASSERT(sizeof(pthread_once_t) >= sizeof(struct pthread_once));
      |                     ^~~~~~~~~~~~~~
<path>/zephyr/include/zephyr/toolchain/gcc.h:87:52: note: in definition of macro ‘BUILD_ASSERT’
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert((EXPR), "" MSG)
      |                                                    ^~~~
<path>/zephyr/include/zephyr/toolchain/gcc.h:87:51: error: expression in static assertion is not an integer
   87 | #define BUILD_ASSERT(EXPR, MSG...) _Static_assert((EXPR), "" MSG)
      |                                                   ^
<path>/zephyr/include/zephyr/posix/posix_types.h:107:1: note: in expansion of macro ‘BUILD_ASSERT’
  107 | BUILD_ASSERT(sizeof(pthread_once_t) >= sizeof(struct pthread_once));
      | ^~~~~~~~~~~~
In file included from <path>/zephyr/include/zephyr/posix/unistd.h:20,
                 from <path>/zephyr/include/zephyr/net/socket.h:1416,
                 from <path>/zephyr/lib/os/zvfs/zvfs_eventfd.c:10:
<path>/zephyr/include/zephyr/posix/sys/stat.h:40:10: fatal error: sys/_timespec.h: No such file or directory
   40 | #include <sys/_timespec.h>
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
```

**To Reproduce**
I wasn't able to find any existing samples that set `CONFIG_POSIX_API=y` which also failed to build.

**Expected behavior**
The applicaiton should compile.

**Impact**
Cannot update to v3.7.0-rc2 (or future versions).

**Environment (please complete the following information):**
 - OS: Ubuntu 24.04
 - Toolchain: Zephyr SDK 0.16.8
 - v3.7.0-rc2

**Additional context**
Git bisect shows the breakage is introduced by the following change: `487a8756c32320f0f1fbe1fbe342849ef89e7440` `posix: eventfd: fix dependency cycle between net and posix`.

ycsin, 2024-07-02:
Would you be able to share the full compile log & `autoconf.h`?

ycsin, 2024-07-02:
I'm not sure what is the root cause yet, it compiles fine if this line is disabled:

https://github.com/zephyrproject-rtos/zephyr/blob/6700a154be3ca29be50fb4f1e35d899de69be5ca/include/zephyr/net/socket.h#L1416

Can't really understand the magic here, cc @jukkar @cfriedt @rlubos

cfriedt, 2024-07-02:
@ycsin - can you make a PR?

I think there likely won't be a problem if that line is removed

0x6e, 2024-07-03:
Removing the specified line allows my project to build.

0x6e, 2024-07-12:
Wouldn't it be better to report this as an unsupported configuration then, rather just failing to compile with some obscure message?

aescolar, 2024-07-12:
> Wouldn't it be better to report this as an unsupported configuration then, rather just failing to compile with some obscure message?

Of course

0x6e, 2024-07-13:
Is there an appropriate place to add such a check, and a place to add a test for this?

0x6e, 2024-07-13:
I'm a bit frustrated that this has been closed as completed. The compiler error has just changed now that the #75351 has been merged.

I've tried to various configuration options, but as far as I can tell `CONFIG_REQUIRES_FULL_LIBCPP=y` and `CONFIG_POSIX_API=y` are incompatible options. Is that really the case? That seems to be quite a regression considering that this configuration previously worked...

cfriedt, 2024-07-13:
It's unfortunately still necessary to workaround the existing dependency cycle between networking and posix (`CONFIG_NET_SOCKETS_POSIX_NAMES`) and also avoid pulling in host headers when building for native_sim.

Fix is just going through CI.

0x6e, 2024-07-20:
Thank you very much @cfriedt your changes work for me.

PR #75351 (2024-07-12): net: socket: fix `sys/_timespec.h` not found error
fixes #75319
PR #75849 (2024-07-27): net + posix: avoid host includes when building for native_sim
A corner case involving C++, posix, networking, and native_sim was causing problems.

Even though C and C++ builds should include zephyr/posix/.. in the default search path with `CONFIG_POSIX_API=y`, for some reason, the native compiler pulls in /usr/include first anyway.

The stat.h header pulled in <sys/types.h> (which is normally fine) but due to the native build, it was pulling in /usr/include/sys/types.h, from the host toolchain.

Explicitly include <zephyr/posix/posix_types.h> instead of <sys/types.h> from stat.h, and continue [... cut]
closed 2024-07-27 by nashif