zephyrproject-rtos/zephyr #75345: LittleFS with Winbond qspi-nor-flash w25q128 not working

MSeesko, 2024-07-02:
**Describe the bug**
I am trying to get the [LittleFS sample](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/fs/littlefs) to run with an external qspi flash W25Q128FV on my nucleo_h753zi board. But i keep running into the same error (Superblock 0x0 has become unwritable).

I already tried setting writeoc to PP_1_1_4 (Since it seems that Winbond devices don't always support PP_1_4_4)

Also i tried to use the zephyr flash shell commands, with mixed results/success.

Do you have any ideas what could cause this? 

**To Reproduce**
target: nucleo_h753zi board
qspi-flash: W25Q128FV
using the [Zephyr LittleFS sample project](https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/subsys/fs/littlefs) i adjusted the following:

My devicetree looks like this:
```
/delete-node/ &storage_partition;

/ {
  aliases {
  };
};

&quadspi {
  pinctrl-0 = <&quadspi_clk_pf10 &quadspi_bk2_ncs_pc11
  &quadspi_bk2_io0_pe7 &quadspi_bk2_io1_pe8
  &quadspi_bk2_io2_pe9 &quadspi_bk2_io3_pe10>;

  pinctrl-names = "default";

  flash-id = <2>;
  status = "okay";

  w25q128: qspi-nor-flash@90000000 {
    compatible = "st,stm32-qspi-nor";
    label = "W25Q128";
    reg = <0x90000000 DT_SIZE_M(16)>; /* 128 Mbits */
    qspi-max-frequency = <50000000>;
    spi-bus-width = <4>;
    writeoc = "PP_1_1_4";
    status = "okay";

    partitions {
      compatible = "fixed-partitions";
      #address-cells = <1>;
      #size-cells = <1>;

      storage_partition: partition@0 {
      label = "storage";
      reg = <0 DT_SIZE_M(16)>;
      };
    };
  };
};
```

To my proj.conf i added this:
```
#Filesystem
CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_LITTLEFS=y
CONFIG_FS_LITTLEFS_CACHE_SIZE=4096
CONFIG_FS_LITTLEFS_PROG_SIZE=256
CONFIG_FS_LITTLEFS_READ_SIZE=256
CONFIG_FS_LITTLEFS_LOOKAHEAD_SIZE=256
CONFIG_FS_LITTLEFS_FC_HEAP_SIZE=8192
CONFIG_NOCACHE_MEMORY=y

# Flash
CONFIG_FLASH=y
[... 110 lines cut ...]
00000020: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000030: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000040: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000050: 88 88 88 88 88 88 88 88  88 88 88 88 88 88 88 88 |........ ........|
00000060: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
00000070: 88 88 88 88 88 88 88 88  88 88 88 88 88 88 88 88 |........ ........|
00000080: 88 88 88 88 88 88 88 88  88 88 88 88 88 88 88 88 |........ ........|
00000090: 88 88 88 88 88 88 88 88  88 88 88 88 88 88 88 88 |........ ........|
000000A0: ff ff ff ff ff ff ff ff  ff ff ff ff ff ff ff ff |........ ........|
000000B0: 88 88 88 88 88 88 88 88  88 88 88 88 88 88 88 88 |........ ........|
000000C0: 88 88 88 88 88 88 88 88  88 88 88 88 88 88 88 88 |........ ........|
000000D0: 88 88 88 88 88 88 88 88  88 88 88 88 88 88 88 88 |........ ........|
000000E0: 88 88 88 88 88 88 88 88  88 88 88 88 88 88 88 88 |........ ........|
000000F0: 88 88 88 88 88 88 88 88  88 88 88 88 88 88 88 88 |........ ........|
```

**Expected behavior**
I expected zephyrs qspi-nor-flash implementation to work with any qspi-nor-flash, given the configuration was done correctly.

**Impact**
I would have to change to another hardware if i cannot get the Winbond flash up and running with the littleFS.

**Environment:**
 - OS: Ubuntu (Docker Container)
 - Zephyr 3.6 with SDK 0.16.5

ajarmouni-st, 2024-07-03:
@MSeesko Can you try erasing the Flash chip first before running the LittleFS sample?

GeorgeCGV, 2024-07-03:
Used to work back in the day. You have to use `writeoc = "PP_1_1_4";`. 

Could you validate that QE (SR2 1st bit) is set?

MSeesko, 2024-07-05:
> @MSeesko Can you try erasing the Flash chip first before running the LittleFS sample? FYI https://docs.zephyrproject.org/latest/samples/drivers/jesd216/README.html

Yes i tried it manually via the flash shell command
```
uart:~$ flash erase W25Q128 0 1000000
Erase success.
```

and also via the KConfig Option from the LittleFS sample CONFIG_APP_WIPE_STORAGE.

Still the same error.

> Used to work back in the day. You have to use `writeoc = "PP_1_1_4";` (_Instruction 0x32_) and you do.
> 
> Could you validate that QE (SR2 1st bit) is set?

How exactly would i do that? Is there a zephyr API to call a register read for the QSPI?

Also specifically i have the W25Q128JVSSIQ which should have the QE Bit set by default by the factory.

de-nordic, 2024-07-09:
If this fails in the Flash shell to then it is not LittleFS problem. There might be a problem in QSPI driver.

aescolar, 2024-07-09:
CC @erwango

GeorgeCGV, 2024-07-09:
> How exactly would i do that? Is there a zephyr API to call a register read for the QSPI?

It is possible to set a breakpoint in `drivers/flash/flash_stm32_qspi.c`.

Unfortunately, I don't have HW with QSPI at the moment. The `CCR`s seem correct:
- Write - `0x3002532` = 4 data lines, 0 dummy, no alt-bytes, 3-byte address, 1 address line, 1 instruction line, `0x32` (_PP_1_1_4_)
- Erase - `0x2520` = 3-byte address, 1 address line, 1 instruction line, `0x20` (_SPI_NOR_CMD_SE_)
- Write enable - `0x106` =  1 instruction line, `0x06` (_SPI_NOR_CMD_WREN_)

You can modify the driver to read out all status registers:
```diff
+++ b/drivers/flash/flash_stm32_qspi.c
@@ -1508,6 +1508,15 @@ static int flash_stm32_qspi_init(const struct device *dev)
                dev_cfg->flash_size);
 #endif
 
+       uint8_t regs[3];
+       for (unsigned i = 0; i < sizeof(regs); ++i) {
+               if (qspi_read_status_register(dev, i + 1, &regs[i]) < 0) {
+                       LOG_ERR("Read SR%u failed", i);
+               } else {
+                       LOG_INF("SR%u %x", i + 1, regs[i]);
+               }
+       }
+
        return 0;
 }
```

MSeesko, 2024-07-12:
Thanks @GeorgeCGV for your help!

I added your suggestion for the driver and got the following output:

```
[00:00:00.001,000] <dbg> flash_stm32_qspi: qspi_send_cmd: Instruction 0x66
[00:00:00.001,000] <dbg> flash_stm32_qspi: qspi_send_cmd: CCR 0x166
[00:00:00.001,000] <dbg> flash_stm32_qspi: qspi_send_cmd: Instruction 0x99
[00:00:00.001,000] <dbg> flash_stm32_qspi: qspi_send_cmd: CCR 0x199
[00:00:00.001,000] <inf> flash_stm32_qspi: Reading SFDP
[00:00:00.001,000] <inf> flash_stm32_qspi: W25Q128: SFDP v 1.5 AP ff with 1 PH
[00:00:00.001,000] <inf> flash_stm32_qspi: PH0: ff00 rev 1.5: 16 DW @ 80
[00:00:00.001,000] <inf> flash_stm32_qspi: Reading SFDP
[00:00:00.001,000] <inf> flash_stm32_qspi: W25Q128: 16 MiBy flash
[00:00:00.001,000] <dbg> flash_stm32_qspi: spi_nor_process_bfp: Erase 4096 with 20
[00:00:00.001,000] <dbg> flash_stm32_qspi: spi_nor_process_bfp: Erase 32768 with 52
[00:00:00.002,000] <dbg> flash_stm32_qspi: spi_nor_process_bfp: Erase 65536 with d8
[00:00:00.002,000] <dbg> flash_stm32_qspi: spi_nor_process_bfp: Page size 256 bytes
[00:00:00.002,000] <dbg> flash_stm32_qspi: spi_nor_process_bfp: Flash size 16777216 bytes
[00:00:00.002,000] <inf> flash_stm32_qspi: Quad read mode 4 instr [0x6b] supported
[00:00:00.002,000] <inf> flash_stm32_qspi: Quad read mode 7 instr [0xeb] supported
[00:00:00.002,000] <inf> flash_stm32_qspi: Quad read mode 7 instr [0xeb] will be used
[00:00:00.002,000] <inf> flash_stm32_qspi: QE requirement mode: 4
[00:00:00.002,000] <inf> flash_stm32_qspi: Quad mode enabled
[00:00:00.002,000] <dbg> flash_stm32_qspi: setup_pages_layout: layout page 256 not compatible with erase s6
[00:00:00.002,000] <dbg> flash_stm32_qspi: setup_pages_layout: erase size will be used as layout page size
[00:00:00.002,000] <dbg> flash_stm32_qspi: setup_pages_layout: layout 4096 x 4096 By pages
[00:00:00.002,000] <inf> flash_stm32_qspi: NOR quad-flash at 0x90000000 (0x1000000 bytes)
[00:00:00.002,000] <inf> flash_stm32_qspi: SR1 0
[00:00:00.002,000] <inf> flash_stm32_qspi: SR2 2
[00:00:00.002,000] <inf> flash_stm32_qspi: SR3 60
[00:00:00.002,000] <dbg> flash_stm32h7: stm32h7_flash_init: Flash initialized. BS: 32
[00:00:00.002,000] <dbg> flash_stm32h7: stm32h7_flash_init: Block 0: bs: 131072 count: 16
[00:00:00.002,000] <dbg> flash_stm32h7: flash_stm32h7_write_protection: Disable write protection
[00:00:00.002,000] <dbg> fs: fs_register: fs register 0: 0
*** Booting Zephyr OS build v3.6.0 ***
[00:00:00.002,000] <dbg> disk: disk_access_register: disk interface(SD) registered
[00:00:00.002,000] <inf> main:  --- Driver Sensor Control --- 
[00:00:00.002,000] <inf> flashdisk: Initialize device SD
[00:00:00.002,000] <inf> flashdisk: offset 0, sector size 4096, page size 4096, volume size 1048576
[00:00:00.003,000] <inf> fat_fs_test: Block count 256
Sector size 4096
Memory Size(MB) 1
[00:00:00.003,000] <inf> flashdisk: Initialize device SD
[00:00:00.003,000] <inf> flashdisk: offset 0, sector size 4096, page size 4096, volume size 1048576
[00:00:00.005,000] <inf> flashdisk: Initialize device SD
[00:00:00.005,000] <inf> flashdisk: offset 0, sector size 4096, page size 4096, volume size 1048576
[00:00:00.009,000] <dbg> flash_stm32_qspi: qspi_send_cmd: Instruction 0x6
[00:00:00.009,000] <dbg> flash_stm32_qspi: qspi_send_cmd: CCR 0x106
[00:00:00.009,000] <dbg> flash_stm32_qspi: qspi_send_cmd: Instruction 0x20
[00:00:00.009,000] <dbg> flash_stm32_qspi: qspi_send_cmd: CCR 0x2520
... the following 4 log entries repeat about 100 times
[00:00:00.060,000] <dbg> flash_stm32_qspi: qspi_send_cmd: Instruction 0x6
[00:00:00.060,000] <dbg> flash_stm32_qspi: qspi_send_cmd: CCR 0x106
[00:00:00.060,000] <dbg> flash_stm32_qspi: qspi_write_access: Instruction 0x32
[00:00:00.060,000] <dbg> flash_stm32_qspi: qspi_write_access: CCR 0x3002532
... until it ends with:
[00:00:00.385,000] <dbg> flash_stm32_qspi: qspi_send_cmd: Instruction 0x6
[00:00:00.385,000] <dbg> flash_stm32_qspi: qspi_send_cmd: CCR 0x106
[00:00:00.385,000] <dbg> flash_stm32_qspi: qspi_write_access: Instruction 0x32
[00:00:00.385,000] <dbg> flash_stm32_qspi: qspi_write_access: CCR 0x3002532
[00:00:00.386,000] <inf> flashdisk: Initialize device SD
[00:00:00.386,000] <inf> flashdisk: offset 0, sector size 4096, page size 4096, volume size 2097152
[00:00:00.389,000] <err> fs: fs mount error (-19)
```

As you can see i tried switching from littlefs to fatfs, but still it seems to fail at the same write access instruction
```
[00:00:00.060,000] <dbg> flash_stm32_qspi: qspi_send_cmd: Instruction 0x6
[00:00:00.060,000] <dbg> flash_stm32_qspi: qspi_send_cmd: CCR 0x106
[00:00:00.060,000] <dbg> flash_stm32_qspi: qspi_write_access: Instruction 0x32
[00:00:00.060,000] <dbg> flash_stm32_qspi: qspi_write_access: CCR 0x3002532
```

The QE Bit from Statusregister 2 (SR2 1st Bit) seems to be set.
`[00:00:00.002,000] <inf> flash_stm32_qspi: SR2 2`

Also i ordered another QSPI-Flash mx25l25645g, which is also used in the [littlefs sample](https://github.com/zephyrproject-rtos/zephyr/blob/1e20f58c17c1cc8f0d5914cc0e014f74b7c85592/samples/subsys/fs/littlefs/boards/nucleo_h743zi.overlay). Using the exact same devicetree-configuration from that sample and using that QSPI-Flash i still get the same behavior, regardless of using writeoc PP_1_1_4 or PP_1_4_4 (it's either Instruction 0x32 or 0x38, but still fails after trying qspi_write_access).

arnebdr, 2024-07-29:
I also had issues with a Winbond Flash W25Q512 and found a solution for it. Maybe your problem is related to this: #76432

FRASTM, 2024-08-02:
> I also had issues with a Winbond Flash W25Q512 and found a solution for it. Maybe your problem is related to this: #76432

Not sure. here the  **qspi** flash driver is writing to the flash.
However this might be a issue with the qspi_program_addr_4b  command sent to the flash

de-nordic, 2024-09-05:
I have removed the File System label as it more Flash/qspi issue.

FRASTM, 2024-09-18:
@MSeesko can you please check if this: https://github.com/zephyrproject-rtos/zephyr/issues/76432   can fix the pb with that Winbond flash ?

wuhanstudio, 2024-10-06:
The same problem here (**W25q128 QSPI + STM32L475**) and this https://github.com/zephyrproject-rtos/zephyr/issues/76432 couldn't solve the problem:

Board Schematic: https://github.com/RT-Thread/IoT_Board/blob/master/docs/board/Pandora_STM32L4_Board_V2.4_SCH.pdf

[image]

```
&quadspi {
	pinctrl-0 = <&quadspi_clk_pe10 &quadspi_ncs_pe11
			&quadspi_bk1_io0_pe12 &quadspi_bk1_io1_pe13
			&quadspi_bk1_io2_pe14 &quadspi_bk1_io3_pe15>;
	pinctrl-names = "default";

	status = "okay";
	w25q128jv: qspi-nor-flash@90000000 {
			compatible = "st,stm32-qspi-nor";
			reg = <0x90000000 DT_SIZE_M(16)>; /* 128 Mbits */
			qspi-max-frequency = <80000000>;
			jedec-id = [ef 40 18];
			spi-bus-width = <4>;
			status = "okay";
	};
};

&w25q128jv {
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		external_partition: partition@0 {
			label = "storage";
			reg = <0x00000000 DT_SIZE_M(16)>;
		};
	};
};
```

The official example  https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/drivers/spi_flash could erase the flash but could not write to the flash:

```
qspi-nor-flash@90000000 SPI flash testing
==========================

Perform test on single sector
Test 1: Flash erase
Flash erase succeeded!

Test 2: Flash write
Attempting to write 4 bytes
Data read does not match data written!!
000ff000 wrote 55 read ff MISMATCH
000ff001 wrote aa read ff MISMATCH
000ff002 wrote 66 read ff MISMATCH
000ff003 wrote 99 read ff MISMATCH

Perform test on multiple consecutive sectors
Test 1: Flash erase
Flash erase succeeded!

Test 2: Flash write
Attempting to write 4 bytes at offset 0xff000
Data read does not match data written!!
000ff000 wrote 55 read ff MISMATCH
000ff001 wrote aa read ff MISMATCH
000ff002 wrote 66 read ff MISMATCH
000ff003 wrote 99 read ff MISMATCH
Attempting to write 4 bytes at offset 0x100000
Data read does not match data written!!
00100000 wrote 55 read ff MISMATCH
00100001 wrote aa read ff MISMATCH
00100002 wrote 66 read ff MISMATCH
00100003 wrote 99 read ff MISMATCH

[00:00:00.000,000] <inf> flash_stm32_qspi: Reading SFDP
[00:00:00.000,000] <inf> flash_stm32_qspi: qspi-nor-flash@90000000: SFDP v 1.5 AP ff with 1 PH
[00:00:00.000,000] <inf> flash_stm32_qspi: PH0: ff00 rev 1.5: 16 DW @ 80
[00:00:00.000,000] <inf> flash_stm32_qspi: Reading SFDP
[00:00:00.000,000] <inf> flash_stm32_qspi: qspi-nor-flash@90000000: 16 MiBy flash
[00:00:00.000,000] <inf> flash_stm32_qspi: Quad read mode 4 instr [0x6b] supported
[00:00:00.000,000] <inf> flash_stm32_qspi: Quad read mode 7 instr [0xeb] supported
[00:00:00.000,000] <inf> flash_stm32_qspi: Quad read mode 7 instr [0xeb] will be used
[00:00:00.000,000] <inf> flash_stm32_qspi: QE requirement mode: 4
[00:00:00.000,000] <inf> flash_stm32_qspi: Quad mode enabled
[00:00:00.000,000] <inf> flash_stm32_qspi: NOR quad-flash at 0x90000000 (0x1000000 bytes)
*** Booting Zephyr OS build v3.7.0-3953-g1bbeeadfbf43 ***
```

FRASTM, 2024-10-09:
The PR https://github.com/zephyrproject-rtos/zephyr/pull/78663  is for the **flash_stm32_ospi** driver  to  write the flash in 4Bytes mode if the flash says it supports this SPI_NOR_CMD_4BA
Then it could be interesting to trace if the write operation to the flash is achieved in  SPI_NOR_CMD_4BA command
This command is also valid for the QSPI driver.

closed 2024-12-24 by github-actions[bot]