zephyrproject-rtos/zephyr #75350: imx_usdhc: disk_access_init on MIMXRT595 Board fails (Error code -134)

attina, 2024-07-03:
Describe the bug
The disk_access_init failed with error code -134 on the NXP MIMXRT595 Eval Board. 

What target platform are you using?
mimxrt595_evk_cm3 (NXP MIMXRT595 Eval Board)

What have you tried to diagnose or workaround this issue?
None

To Reproduce
Steps to reproduce the behavior:

Following project variables are enabled:

CONFIG_DISK_ACCESS=y
CONFIG_LOG=y
CONFIG_FILE_SYSTEM=y
CONFIG_FAT_FILESYSTEM_ELM=y
CONFIG_PRINTK=y
CONFIG_MAIN_STACK_SIZE=2048
CONFIG_DISK_DRIVER_SDMMC=y
CONFIG_SD_LOG_LEVEL_DBG=y
CONFIG_SDHC_LOG_LEVEL_DBG=y
CONFIG_SDMMC_LOG_LEVEL_DBG=y

Expected behavior
disk_access_init successfully.

Actual behavior
*** Booting Zephyr OS build v3.6.0 ***
*** Booting Zephyr OS build v3.6.0 ***
[00:26:34.261,726] <inf> sd: Card does not support CMD8, assuming legacy card
[00:26:34.262,947] <err> main: Storage init ERROR!
[00:26:35.867,130] <inf> sd: Card does not support CMD8, assuming legacy card
[00:26:35.868,348] <err> fs: fs mount error (-5)
Error mounting disk.
[00:26:35.868,374] <err> fs: fs not mounted (mp == 0x30180fe0)
*** Booting Zephyr OS build v3.6.0 ***
*** Booting Zephyr OS build v3.6.0 ***
[00:38:50.392,247] <dbg> sd: sd_init_io: Host controller support 3.3V max
[00:38:50.392,254] <dbg> sd: sd_init_io: Resetting power to card
[00:38:50.392,286] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 0Hz, card power OFF, voltage 3.3V
[00:38:50.892,391] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 0Hz, card power ON, voltage 3.3V
[00:38:51.392,492] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 400000Hz, card power ON, voltage 3.3V
[00:38:51.392,501] <dbg> usdhc: imx_usdhc_set_io: BUS CLOCK: 386718
[00:38:51.394,317] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:51.394,599] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:51.394,880] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:51.395,162] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:51.395,443] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error tion: SD CMD8 failed with error -5
[00:38:51.396,006] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:51.396,294] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:51.396,578] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:51.396,860] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:51.397,141] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:51.397,146] <inf> sd: Card does not support CMD8, assuming legacy card
[00:38:51.398,349] <err> main: Storage init ERROR!
[00:38:51.398,379] <dbg> sd: sd_init_io: Host controller support 3.3V max
[00:38:51.398,385] <dbg> sd: sd_init_io: Resetting power to card
[00:38:51.398,410] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 0Hz, card power OFF, voltage 3.3V
[00:38:51.898,493] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 0Hz, card power ON, voltage 3.3V
[00:38:52.398,589] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 400000Hz, card power ON, voltage 3.3V
[00:38:52.398,599] <dbg> usdhc: imx_usdhc_set_io: BUS CLOCK: 386718
[00:38:52.400,395] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:52.400,677] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:52.400,959] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:52.401,242] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:52.401,524] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:52.401,806] <dbg> sd: sd_send_interface_condition: SD CMD8 fd_interface_condition: SD CMD8 failed with error -5
[00:38:52.402,374] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:52.402,654] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:52.402,935] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:52.403,217] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:38:52.403,221] <inf> sd: Card does not support CMD8, assuming legacy card
[00:38:52.404,420] <err> fs: fs mount error (-5)
Error mounting disk.
[00:38:52.404,442] <err> fs: fs not mounted (mp == 0x30180fe0)

Impact
Prototype: Annoyance
Series: Showstopper

Environment (please complete the following information):

OS: Windows
Toolchain: Zephyr SDK 
Version used: v3.6.0

dleach02, 2024-07-07:
@attina, thank you for the report. Which Zephyr sample/test did you use to reproduce this? Or is this a custom setup?

attina, 2024-07-08:
Hi David,
The sample on which I found this issue is samples/subsys/fs/fs_sample, no any custom setup.

dleach02, 2024-07-10:
@hakehuang do we have results on our test range for this sample on RT595?

hakehuang, 2024-07-10:
> @hakehuang do we have results on our test range for this sample on RT595?

RT595 use a very old SDHC card socket which I need buy/borrow(so I did not setup on board yet), ButRT685 is OK with below log.
```
2024-07-01 14:23:56,208 - twister - DEBUG - Flash command: ['west', 'flash', '--skip-rebuild', '-d', '/home/ubuntu/nxp/mimxrt685_evk/temp/mimxrt685_evk_mimxrt685s_cm33_twister_test_only/twister-out/mimxrt685_evk_mimxrt685s_cm33/tests/subsys/fs/fs_api/filesystem.api', '--runner', 'jlink', '--tool-opt=-SelectEmuBySN  000720532062']
2024-07-01 14:23:59,446 - twister - DEBUG - DEVICE: *** Booting Zephyr OS build v3.7.0-rc2 ***
2024-07-01 14:23:59,449 - twister - DEBUG - DEVICE: Running TESTSUITE fs_api_dir_file
2024-07-01 14:23:59,455 - twister - DEBUG - DEVICE: ===================================================================
2024-07-01 14:23:59,457 - twister - DEBUG - DEVICE: START - test_file_ops
2024-07-01 14:23:59,457 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,459 - twister - DEBUG - DEVICE: Open tests:
2024-07-01 14:23:59,460 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,461 - twister - DEBUG - DEVICE: Open a file without a path
2024-07-01 14:23:59,463 - twister - DEBUG - DEVICE: E: invalid file name!!
2024-07-01 14:23:59,463 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,466 - twister - DEBUG - DEVICE: Open a file with wrong abs path
2024-07-01 14:23:59,468 - twister - DEBUG - DEVICE: E: invalid file name!!
2024-07-01 14:23:59,469 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,471 - twister - DEBUG - DEVICE: Open a file with wrong path
2024-07-01 14:23:59,473 - twister - DEBUG - DEVICE: E: invalid file name!!
2024-07-01 14:23:59,473 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,476 - twister - DEBUG - DEVICE: Open a file with wrong abs path
2024-07-01 14:23:59,479 - twister - DEBUG - DEVICE: E: mount point not found!!
2024-07-01 14:23:59,479 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,482 - twister - DEBUG - DEVICE: Filesystem has no open functionality
2024-07-01 14:23:59,482 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,483 - twister - DEBUG - DEVICE: Double-open
2024-07-01 14:23:59,484 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,488 - twister - DEBUG - DEVICE: Reopen the same fileOpened file /NAND:/testfile.txt
2024-07-01 14:23:59,488 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,490 - twister - DEBUG - DEVICE: Write tests:
2024-07-01 14:23:59,492 - twister - DEBUG - DEVICE: Write to an unopened file
2024-07-01 14:23:59,496 - twister - DEBUG - DEVICE: Write to filesystem has no write interface
2024-07-01 14:23:59,499 - twister - DEBUG - DEVICE: Write to file from a invalid source
2024-07-01 14:23:59,514 - twister - DEBUG - DEVICE: E: file write error (-22)
2024-07-01 14:23:59,514 - twister - DEBUG - DEVICE: Data written:"hello world!"
2024-07-01 14:23:59,514 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,515 - twister - DEBUG - DEVICE: Data successfully written!
2024-07-01 14:23:59,515 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,515 - twister - DEBUG - DEVICE: Read tests:
2024-07-01 14:23:59,515 - twister - DEBUG - DEVICE: Read an unopened file
2024-07-01 14:23:59,515 - twister - DEBUG - DEVICE: Filesystem has no read interface
2024-07-01 14:23:59,515 - twister - DEBUG - DEVICE: Read to a invalid buffer
2024-07-01 14:23:59,517 - twister - DEBUG - DEVICE: E: file read error (-22)
2024-07-01 14:23:59,518 - twister - DEBUG - -- west flash: using runner jlink
-- runners.jlink: reset after flashing requested
-- runners.jlink: JLink version: 7.96
-- runners.jlink: Flashing file: /home/ubuntu/nxp/mimxrt685_evk/temp/mimxrt685_evk_mimxrt685s_cm33_twister_test_only/twister-out/mimxrt685_evk_mimxrt685s_cm33/tests/subsys/fs/fs_api/filesystem.api/zephyr/zephyr.bin

2024-07-01 14:23:59,520 - twister - DEBUG - DEVICE: Data read:"hello world!"
2024-07-01 14:23:59,521 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,523 - twister - DEBUG - DEVICE: Data read matches data written
2024-07-01 14:23:59,524 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,526 - twister - DEBUG - DEVICE: Truncate tests: max file size is 128byte
2024-07-01 14:23:59,527 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,530 - twister - DEBUG - DEVICE: Truncate, seek, tell an unopened file
2024-07-01 14:23:59,531 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,535 - twister - DEBUG - DEVICE: Truncate, seek, tell fs has no these functionality
2024-07-01 14:23:59,538 - twister - DEBUG - DEVICE: Truncating to size larger than 128byte
2024-07-01 14:23:59,541 - twister - DEBUG - DEVICE: E: file truncate error (-22)
2024-07-01 14:23:59,541 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,543 - twister - DEBUG - DEVICE: Testing shrink to 0 size
2024-07-01 14:23:59,546 - twister - DEBUG - DEVICE: File seek from invalid whence
2024-07-01 14:23:59,548 - twister - DEBUG - DEVICE: E: file seek error (-22)
2024-07-01 14:23:59,551 - twister - DEBUG - DEVICE: Testing write after truncating
2024-07-01 14:23:59,552 - twister - DEBUG - DEVICE: 
2024-07-01 14:23:59,552 - twister - DEBUG - DEVICE: Write tests:
2024-07-01 14:23:59,555 - twister - DEBUG - DEVICE: Write to an unopened file
2024-07-01 14:23:59,559 - twister - DEBUG - DEVICE: Write to filesystem has no write interface
2024-07-01 14:23:59,562 - twister - DEBUG - DEVICE: Write to file from a invalid source
[... 187 lines cut ...]
2024-07-01 14:24:00,039 - twister - DEBUG - DEVICE: TESTSUITE fs_api_register_mount succeeded
[... cut ...]
2024-07-01 14:24:00,131 - twister - DEBUG - DEVICE: - PASS - [fs_api_register_mount.test_mount_unmount] duration = 0.081 seconds
2024-07-01 14:24:00,132 - twister - DEBUG - DEVICE: 
2024-07-01 14:24:00,134 - twister - DEBUG - DEVICE: ------ TESTSUITE SUMMARY END ------
2024-07-01 14:24:00,135 - twister - DEBUG - DEVICE: 
2024-07-01 14:24:00,140 - twister - DEBUG - DEVICE: ===================================================================
2024-07-01 14:24:00,144 - twister - DEBUG - DEVICE: RunID: d535f6c0d99484b60c2ed32741f595cf
2024-07-01 14:24:00,150 - twister - DEBUG - DEVICE: PROJECT EXECUTION SUCCESSFUL
```

hakehuang, 2024-07-10:
@attina , I notice in your log you are using v3.60?

attina, 2024-07-11:
@hakehuang, Yes, the issue was reproduced on SDK v3.60.

dleach02, 2024-07-12:
@attina are you able to reproduce this issue with the current state of main?

attina, 2024-07-15:
@hakehuang Yes, The card is a SD card, part number is "Kingstone sd4/16GB".
I added file `mimxrt595_evk_cm33.overlay` in board folder.
```
&usdhc0 {
	mmc {
		compatible = "zephyr,sdmmc-disk";
	};
};
```
and run the test again, the problem is still there.
```
uart:~$ *** Booting Zephyr OS build v3.6.0 ***
[00:24:43.565,027] <dbg> sd: sd_init_io: Host controller support 3.3V max
[00:24:43.565,033] <dbg> sd: sd_init_io: Resetting power to card
[00:24:43.565,064] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 0Hz, card power OFF, voltage 3.3V
[00:24:44.065,182] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 0Hz, card power ON, voltage 3.3V
[00:24:44.565,280] <dbg> usdhc: imx_usdhc_set_io: SDHC I/O: bus width 1, clock 400000Hz, card power ON, voltage 3.3V
[00:24:44.565,295] <dbg> usdhc: imx_usdhc_set_io: BUS CLOCK: 386718
[00:24:44.567,103] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:24:44.567,393] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:24:44.567,679] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:24:44.567,965] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:24:44.568,248] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:24:44.568,532] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:24:44.568,817] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:24:44.569,104] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:24:44.569,390] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:24:44.569,675] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:24:44.569,961] <dbg> sd: sd_send_interface_condition: SD CMD8 failed with error -5
[00:24:44.569,969] <inf> sd: Card does not support CMD8, assuming legacy card
```

attina, 2024-07-15:
@dleach02 I will try to test on main branch.

hakehuang, 2024-07-15:
> @hakehuang Yes, The card is a SD card, part number is "Kingstone sd4/16GB".

there is a missing in dts, from the log your card is not support 1v8, but the default dts add 1v8 support.

attina, 2024-07-16:
@dleach02 The sample code failed to pass the build on branch main.

hakehuang, 2024-07-17:
> @dleach02 The sample code failed to pass the build on branch main.

the default setting builds ok for me on v3.7.0-rc3-23-g3828c8b86ac, even with below change
```
scripts/twister -p mimxrt595_evk/mimxrt595s/cm33 -T samples/subsys/fs/fs_sample/
ZEPHYR_BASE unset, using "/home/shared/disk/zephyr_project/zephyr_test/zephyr"
Renaming output directory to /home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out.2
INFO    - Using Ninja..
INFO    - Zephyr version: v3.7.0-rc3-23-g3828c8b86ac5
INFO    - Using 'zephyr' toolchain.
INFO    - Building initial testsuite list...
INFO    - Writing JSON report /home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/testplan.json
INFO    - JOBS: 8
INFO    - Adding tasks to the queue...
INFO    - Added initial list of jobs to queue
INFO    - Total complete:   11/  11  100%  skipped:    9, failed:    0, error:    0
INFO    - 11 test scenarios (11 test instances) selected, 9 configurations skipped (9 by static filter, 0 at runtime).
INFO    - 2 of 11 test configurations passed (100.00%), 0 failed, 0 errored, 9 skipped with 0 warnings in 23.98 seconds
INFO    - In total 0 test cases were executed, 11 skipped on 1 out of total 2 platforms (50.00%)
INFO    - 0 test configurations executed on platforms, 2 test configurations were only built.
INFO    - Saving reports...
INFO    - Writing JSON report /home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/twister.json
INFO    - Writing xunit report /home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/twister.xml...
INFO    - Writing xunit report /home/shared/disk/zephyr_project/zephyr_test/zephyr/twister-out/twister_report.xml...
INFO    - Run completed

```

attina, 2024-07-18:
@hakehuang, no rework was mode on the board. I believe this issue is a software or configuration problem.
Because I have another project which is based on freeRTOS, that project can access SD card file system without any trouble.

hakehuang, 2024-07-18:
> rework was mode on the board. I believe this issue is a software or configuration problem.
> Because I have another project which is based on freeRTOS, that project can access SD card file system without any trouble.

without rework the board is connected to MMC card, not SD, as I tested SD card in mimxrt1060_evk which works fine.

attina, 2024-07-19:
@hakehuang checked the board with hardware, the board is reworked to support SD card, current rework match to the schematic.

[image]

[image]

hakehuang, 2024-07-19:
@attina , maybe you can try populate R500, which is a pull up for command. seeing from the code, it try to send CMD8 and it fails then switch card to idle, and all following cmd are not valid though.

attina, 2024-07-20:
@hakehuang Is it possible workaround this by changing code. I really don't good at re-work the board:).

hakehuang, 2024-07-21:
> @hakehuang Is it possible workaround this by changing code. I really don't good at re-work the board:).

this is common code, it would be very hard to resolve, can you find another type of sd card which supports 1v8? or this is a customer requirement?

attina, 2024-07-22:
@hakehuang, It's a customer requirement. And I think add 3.3v legacy card support it's a reasonable compatible requirement.
Could you guide me to do that in customer branch first.

attina, 2024-07-22:
Go through kongston website, 3.3v is a very common used voltage level in all kinds of card.
[image]

hakehuang, 2024-07-22:
> It's a customer requirement. And I think add 3.3v legacy card support it's a reasonable compatible requirement.
> Could you guide me to do that in customer branch first.

in the subsys/sd.c, sd_common_init comment below lines, I do not have kinston card, all I have is sandisk, so I did not give a try.
```
	/* Perform voltage check using SD CMD8 */
	ret = sd_retry(sd_send_interface_condition, card, CONFIG_SD_RETRY_COUNT);
	if (ret == -ETIMEDOUT) {
		LOG_INF("Card does not support CMD8, assuming legacy card");
		return sd_idle(card);
	} else if (ret) {
		LOG_ERR("Card error on CMD 8");
		return ret;
	}
```

attina, 2024-07-22:
> I find the PIO4_0 shall be pull high such as the SDC can power on. As SDHC is not the default option, so you need modify the 
> device tree to enable this pin.
@hakehuang You are right, I should check the schematic carefully, since this is a rework board, the device tree need to update according to the rework. After change the DTS as the following, SD card access is working now. 
```
&usdhc0 {
	mmc {
		status = "disabled";
	};
	pwr-gpios = <&gpio4 0 GPIO_ACTIVE_HIGH>;
	sdmmc {
		compatible = "zephyr,sdmmc-disk";
		status = "okay";
	};
	no-1-8-v;
};
```
Thanks again for your great help.

dleach02, 2024-07-22:
Thank you @hakehuang in helping to identify the resolution. 

@attina I'm going to close this issue based on your response.

closed 2024-07-22 by dleach02

attina, 2024-07-23:
@dleach02 @hakehuang Thank you.