zephyrproject-rtos/zephyr #75389: it8xxx2_evb: tests/drivers/watchdog/wdt_basic_api/drivers.watchdog fails

majunkier, 2024-07-03:
**Describe the bug**
The system outputs an error indicating that it is waiting to restart the MCU. However, the expected restart does not occur. 
Manual reboot during test execution makes the test pass, so the problem could be with the wake-up part of watchdog.

**To Reproduce**
```
twister -p it8xxx2_evb --device-testing --west-flash="<ITE_TOOL_PATH>" --device-serial=<DEV_SERIAL> -s tests/drivers/watchdog/wdt_basic_api/drivers.watchdog
```

**Expected behavior**
Test should pass

**Impact**
Not clear

**Logs and console output**
```
tests/drivers/watchdog/wdt_basic_api/drivers.watchdog  FAILED Timeout (device: A50285BI, 68.712s)
INFO    - twister-out/it8xxx2_evb/tests/drivers/watchdog/wdt_basic_api/drivers.watchdog/handler.log
ERROR   - Running TESTSUITE wdt_basic_test_suite
===================================================================
START - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
```

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.16.8
 - Version : v3.7.0-rc2-66-gbd2fc8009b98

Dino-Li, 2024-07-04:
@GTLin08 can you look into this issue?

Dino-Li, 2024-07-04:
@majunkier 
If the watchdog test is executed right after flashing test firmware image, failure symptom is expected.
Because SOC have to enter the DBGR mode to reflash firmware image.
In this mode, the watchdog function will be disabled until we perform a cold reset for SOC.

Dino-Li, 2024-07-08:
@SzymonRichert
If this watchdog test has been included in the automated tests for the it8xxx2_evb platform, that test should have cold reset the SOC for each refresh test firmware image, otherwise the previous test would fail.

SzymonRichert, 2024-07-11:
@Dino-Li can you submit a PR excluding this platform from execution ?

Dino-Li, 2024-07-12:
@SzymonRichert on this platform, after flashing test firmware image, does autotest no longer assert SOC's WRST# pin to hard reset it?

pkunieck, 2024-07-19:
@Dino-Li  We don't have any documentation and I don't see any implementation for using WRST# pin in code. Can you provide us with documentation for this board or maybe you have working implementation for using WRST# pin on this board for watchdog test? I was trying to contact ITE but I haven't heard back from them yet.

Dino-Li, 2024-07-22:
@pkunieck Which ITE download board are you using?
If you are using the download board as the attached, it will assert GPIO C1 (blue circle) to reset SOC on the EVB after flashing.
Hunter (zhenyu.hu@intel.com) and Hongb (hongbo.wang@intel.com) should know the background of the reset.
Or contact me (dino.li@ite.com.tw) and I will add related personnel into the loop.
[image]
[image]

pkunieck, 2024-07-23:
I created PR for excluding this test for now. If there will be fix, or idea about DBGR mode we will test it and include it again.

closed 2024-08-12 by pkunieck