zephyrproject-rtos/zephyr #75398: [Zephyr 3.7 RC2] MQTT ON HL7800 MODEM DOESNT WORK

zafersn, 2024-07-03:
**Describe the bug**
Previously, mqtt was working with hl7800 with zehyr3.6.0. After migration, i started getting this error:

Any thought?

```
CONNECT
[00:03:42.410,000] <dbg> modem_hl7800:  HANDLE CONNECT (len:0)
[00:03:42.410,000] <dbg> modem_hl7800: sock_read: Socket 1 RX 192 bytes
[00:03:42.410,000] <dbg> modem_hl7800: wait_for_modem_data: cur:0, exp:1
[00:03:42.460,000] <dbg> modem_hl7800: sock_read: Processing RX, buf len: 234
[00:03:42.461,000] <dbg> modem_hl7800: sock_read: Got all data, get EOF and OK (buf len:42)
[00:03:42.461,000] <dbg> modem_hl7800: sock_read: Sock 1 RX done
[00:03:42.461,000] <dbg> modem_hl7800: allow_sleep: Keep sleep
+KUDP_RCV: "10.105.16.254",53
[00:03:42.461,000] <dbg> modem_hl7812:  UNHANDLED RX IN2
                                       2b 4b 55 44 50 5f 52 43  56 3a 20 22 31 30 2e 31 |+KUDP_RC V: "10.1
                                       30 35 2e 31 36 2e 32 35  34 22 2c 35 33          |05.16.25 4",53   
[00:03:42.461,000] <dbg> modem_hl7812: sockreadrecv_cb_work: Sock 1 RX CB (size: 220)
[00:03:46.590,000] <err> mqtt_helper: getaddrinfo() failed, error -101
[00:03:46.590,000] <err> aws_iot: mqtt_helper_connect, error: 101
[00:03:46.590,000] <err> aws_iot_integration: aws_iot_connect, error: 101
[00:03:46.590,000] <err> cloud_module: cloud_connect failed, error: 101
[00:03:46.590,000] <dbg> cloud_module: connect_cloud: Cloud connection establishment in progress
[00:03:46.590,000] <dbg> cloud_module: connect_cloud: New connection attempt in 32 seconds if not successful
```
Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using?
 - custom board combination of HL7800 and stm32u575
zephyr main branch

**To Reproduce**
Steps to reproduce the behavior:
because I am working on custom board, I cant exactly provide steps to reproduce it.

**Expected behavior**
at 3.6.0 version,  I had succesfull getaddrinfo return and mqtt connection

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain Zephyr SDK 0.16.5
 - Commit SHA or Version used

zafersn, 2024-07-04:
Anyone ask for additional information on this issue? I am still trying to fix it but have not been able to do that so far. And also, Anyone who can shed some light on where the possible issue is and how to debug it will be appreciated

rerickson1, 2024-07-05:
@zafersn based on the log message output you are using a modified HL7800 driver and not the one in the tree. Please provide logs from an unmodified driver.

rerickson1, 2024-07-05:
@jukkar can I get the label `platform: Laird Connectivity` changed to `platform: Ezurio`.
Laird Connectivity is no longer valid because we have renamed the company.

zafersn, 2024-07-05:
hi @rerickson1 
here is the logs belongs to the hl7800 driver: 

```
[00:00:00.064,000] <inf> modem_hl7800: Sleep State: UNINITIALIZED
[00:00:00.064,000] <inf> modem_hl7800: Network State: 0 NOT_REGISTERED
[00:00:00.064,000] <inf> modem_hl7800: Startup State: UNKNOWN
[00:00:00.064,000] <inf> modem_hl7800: Modem Run
[00:00:00.074,000] <dbg> modem_hl7800: modem_boot_handler: Initialization
[00:00:06.277,000] <dbg> modem_hl7800: hl7800_rx: HANDLE +CEREG:  (len:1)
[00:00:06.277,000] <dbg> modem_hl7800: on_cmd_network_report: Network status: 2
[00:00:06.277,000] <inf> modem_hl7800: Network State: 2 SEARCHING
[00:00:06.507,000] <dbg> modem_hl7800: hl7800_rx: HANDLE +KSUP:  (len:1)
[00:00:06.507,000] <inf> modem_hl7800: Startup State: READY
[00:00:06.507,000] <inf> modem_hl7800: Sleep State: AWAKE
[00:00:06.507,000] <inf> modem_hl7800: Modem booted!
[00:00:06.507,000] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+CEREG=0]
[00:00:06.521,000] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:06.521,000] <dbg> modem_hl7800: send_at_cmd: OUT: [AT&V]
[00:00:06.589,000] <dbg> modem_hl7800: hl7800_rx: HANDLE ACTIVE PROFILE: (len:0)
[00:00:06.589,000] <dbg> modem_hl7800: wait_for_modem_data: cur:0, exp:64
[00:00:06.639,000] <dbg> modem_hl7800: profile_handler: length: 61: E0 Q0 V1 X4 &C1 &D1 &R1 &S0 +IFC=2,2 &K3 +IPR=115200 +FCLASS0
[00:00:06.639,000] <dbg> modem_hl7800: profile_handler: echo: 0
[00:00:06.639,000] <dbg> modem_hl7800: hl7800_rx: HANDLE STORED PROFILE 0: (len:0)
[00:00:06.639,000] <dbg> modem_hl7800: profile_handler: length: 61: E0 Q0 V1 X4 &C1 &D1 &R1 &S0 +IFC=2,2 &K3 +IPR=115200 +FCLASS0
[00:00:06.639,000] <dbg> modem_hl7800: profile_handler: echo: 0
[00:00:06.639,000] <dbg> modem_hl7800: hl7800_rx: HANDLE STORED PROFILE 1: (len:0)
[00:00:06.639,000] <dbg> modem_hl7800: profile_handler: length: 61: E1 Q0 V1 X4 &C1 &D1 &R1 &S0 +IFC=2,2 &K3 +IPR=115200 +FCLASS0
[00:00:06.639,000] <dbg> modem_hl7800: profile_handler: echo: 1
[00:00:06.639,000] <dbg> modem_hl7800: wait_for_modem_data: cur:59, exp:66
[00:00:06.690,000] <dbg> modem_hl7800: wait_for_modem_data: cur:59, exp:66
[00:00:06.740,000] <dbg> modem_hl7800: wait_for_modem_data: cur:59, exp:66
[00:00:06.790,000] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:06.790,000] <dbg> modem_hl7800: iface_status_work_cb: Wait for driver init, process network state later
[00:00:06.790,000] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+CMEE=1]
[00:00:06.794,000] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:06.794,000] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+CGMR]
[00:00:06.797,000] <dbg> modem_hl7800: hl7800_rx: HANDLE AT+CGMR (len:14)
[00:00:06.797,000] <dbg> modem_hl7800: wait_for_modem_data: cur:15, exp:31
[00:00:06.847,000] <dbg> modem_hl7800: wait_for_modem_data: cur:22, exp:31
[00:00:06.897,000] <dbg> modem_hl7800: wait_for_modem_data: cur:22, exp:31
[00:00:06.947,000] <inf> modem_hl7800: Revision: HL7800.4.6.9.4
[00:00:06.947,000] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:06.947,000] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KSRAT?]
[00:00:06.950,000] <dbg> modem_hl7800: hl7800_rx: HANDLE +KSRAT:  (len:1)
[00:00:06.950,000] <inf> modem_hl7800: +KSRAT: 1
[00:00:06.951,000] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:06.951,000] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KBNDCFG?]
[00:00:06.970,000] <dbg> modem_hl7800: hl7800_rx: HANDLE +KBNDCFG:  (len:24)
[00:00:06.974,000] <dbg> modem_hl7800: hl7800_rx: HANDLE +KBNDCFG:  (len:24)
[00:00:06.974,000] <inf> modem_hl7800: Current band configuration: 0000 00000000 00000800
[00:00:06.975,000] <dbg> modem_hl7800: hl7800_rx: HANDLE +KBNDCFG:  (len:3)
[00:00:06.975,000] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:06.975,000] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+CFUN=4,0]
[00:00:07.034,000] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:07.035,000] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KSLEEP=2]
[00:00:07.196,000] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:07.196,000] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+CPSMS=0]
[00:00:07.290,000] <dbg> modem_hl7800: iface_status_work_cb: Wait for driver init, process network state later
[00:00:07.361,000] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:07.361,000] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+CEDRXS=0]
[00:00:07.365,000] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:07.365,000] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+CGMI]
[00:00:07.370,000] <dbg> modem_hl7800: hl7800_rx: HANDLE AT+CGMI (len:15)
[00:00:07.370,000] <dbg> modem_hl7800: wait_for_modem_data: cur:16, exp:18
[00:00:07.420,000] <inf> modem_hl7800: Manufacturer: Sierra Wireless
[00:00:07.420,000] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:07.420,000] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+CGMM]
[00:00:07.423,000] <dbg> modem_hl7800: hl7800_rx: HANDLE AT+CGMM (len:6)
[00:00:07.423,000] <dbg> modem_hl7800: wait_for_modem_data: cur:7, exp:9
[... 224 lines cut ...]
[00:00:25.384,000] <dbg> modem_hl7800: on_cmd_connect: 4244 hl7800.c
[00:00:25.384,000] <dbg> modem_hl7800: hl7800_rx: UNHANDLED RX
                                       2b 4b 55 44 50 5f 52 43  56 3a 20 22 31 30 2e 31 |+KUDP_RC V: "10.1
                                       30 35 2e 31 36 2e 32 35  34 22 2c 35 33          |05.16.25 4",53   
[00:00:25.384,000] <dbg> modem_hl7800: sockreadrecv_cb_work: Sock 1 RX CB (size: 95)
[... cut ...]
[00:00:25.515,000] <dbg> net_sock: net_pkt_remote_addr_is_unspecified: 1205 socket.c
[00:00:25.515,000] <dbg> net_sock: sock_get_offload_pkt_src_addr: 1244 socket.c
[00:00:25.515,000] <dbg> net_sock: zsock_recv_dgram: (net_socket_): sock_get_offload_pkt_src_addr -134
[00:00:25.516,000] <dbg> net_sock: zsock_recv_dgram: 1631 socket.c
[00:00:25.516,000] <dbg> net_sock: zsock_recv_dgram: 1634 socket.c
[00:00:25.516,000] <err> net_dns_dispatcher: recv failed on IPv4 socket (134)
[00:00:25.516,000] <err> net_dns_dispatcher: DNS recv error (-134)
```

rerickson1, 2024-07-05:
Thanks for the logs, could you share the sample app so I can try to reproduce?

zafersn, 2024-07-05:
@rerickson1  

> Thanks for the logs, could you share the sample app so I can try to reproduce?

[cellular_hl7800-git.zip](https://github.com/user-attachments/files/16113157/cellular_hl7800-git.zip)

this is exactly what i was doing for you :)

rerickson1, 2024-07-05:
@zafersn I cannot get the sample to build, it gets in an endless cmake loop. Also, the sample isnt configured to use the HL7800 driver...
[build.log](https://github.com/user-attachments/files/16113295/build.log)

zafersn, 2024-07-05:
@rerickson1  in my case, I used `nucleo_u575zi_q.overlay `and  `nucleo_u575zi_q.conf.`  So, it must be configured in my case, which I got all the logs shared them with you, and lte nbiot connection. please see the logs here :https://github.com/zephyrproject-rtos/zephyr/issues/75398#issuecomment-2211302024 . ILL have a look at your logs.

zafersn, 2024-07-05:
I see in your logs, I just sent the prj.conf file to you with a different cellular apn config, just in case, wanted to hide some information, I am sharing everything publicly here. Please set this config something else. 

CONFIG_MODEM_CELLULAR_APN="netfe*****"

and rest of the cmake warning is because of you probably didnt  use board.conf file. Sorry for this comlexity, I just built the test app on already exist zephyr sample for modem test

zafersn, 2024-07-05:
@rerickson1  please try this sample app,

[cellular_hl7800_git2.zip](https://github.com/user-attachments/files/16113536/cellular_hl7800_git2.zip)

rerickson1, 2024-07-05:
I have reproduced the dns issue with the http get sample.
Patch:
[hl7800_http_get.patch](https://github.com/user-attachments/files/16113965/hl7800_http_get.patch)
Log:
[http_get_3.7.0-rc2.log](https://github.com/user-attachments/files/16113966/http_get_3.7.0-rc2.log)

It seems the socket packet source address may not be set correctly due to a change in the stack.
@jukkar any tips?

jukkar, 2024-07-08:
> It seems the socket packet source address may not be set correctly due to a change in the stack.

I checked the socket code and the error 134 (ENOTSUP) is returned if the address family is not `AF_INET` nor `AF_INET6`, so this value is not set in some cases even if it is set in hints when calling `getaddrinfo` in `http_get.c`
What stack change are you referring to?

jukkar, 2024-07-08:
There was a somewhat similar issue with esp_at driver which was discussed here
https://discord.com/channels/720317445772017664/733037635194585148/1250097643510698055

Could there be similar issue with hl7800 driver?

rerickson1, 2024-07-08:
> What stack change are you referring to?

Reworked LLMNR and mDNS responders, and DNS resolver to use sockets and socket services API.

> There was a somewhat similar issue with esp_at driver which was discussed here https://discord.com/channels/720317445772017664/733037635194585148/1250097643510698055
> 
> Could there be similar issue with hl7800 driver?

NET_OFFLOAD does not have a recvfrom option:
```
static struct net_offload offload_funcs = {
	.get = offload_get,
	.bind = offload_bind,
	.listen = offload_listen,
	.connect = offload_connect,
	.accept = offload_accept,
	.send = offload_send,
	.sendto = offload_sendto,
	.recv = offload_recv,
	.put = offload_put,
};
```

rerickson1, 2024-07-08:
@jukkar I set remote address in the HL7800 driver and there is still a problem.
[Changes-On-2e956c.patch](https://github.com/user-attachments/files/16132322/Changes-On-2e956c.patch)

```
[00:00:26.283,203] <dbg> net_sock_addr: exec_query: (0x200047a8): Timeout 5000
[00:00:26.283,294] <dbg> net_dns_resolve: dns_resolve_name: (0x200047a8): DNS id will be 53
[00:00:26.283,325] <dbg> net_dns_resolve: dns_write: (0x200047a8): [0] submitting work to server idx 0 for id 53 hash 59079
[00:00:26.283,538] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KUDPSND=1,"8.8.8.8",53,28]
[00:00:26.299,530] <dbg> modem_hl7800: hl7800_rx: HANDLE CONNECT (len:0)
[00:00:26.302,032] <dbg> modem_hl7800: send_data: Sent 28 bytes
[00:00:26.310,699] <dbg> modem_hl7800: hl7800_rx: HANDLE OK (len:0)
[00:00:26.457,977] <dbg> modem_hl7800: hl7800_rx: HANDLE +KUDP_DATA:  (len:4)
[00:00:26.457,977] <dbg> modem_hl7800: start_socket_rx: Start socket RX ID:1 size:44
[00:00:26.458,068] <dbg> modem_hl7800: send_at_cmd: OUT: [AT+KUDPRCV=1,44]
[00:00:26.470,458] <dbg> modem_hl7800: hl7800_rx: HANDLE CONNECT (len:0)
[00:00:26.470,458] <dbg> modem_hl7800: sock_read: Socket 1 RX 44 bytes
[00:00:26.470,489] <dbg> modem_hl7800: wait_for_modem_data: cur:0, exp:1
[00:00:26.520,629] <dbg> modem_hl7800: sock_read: Processing RX, buf len: 93
[00:00:26.520,965] <dbg> modem_hl7800: sock_read: Got all data, get EOF and OK (buf len:49)
[00:00:26.521,026] <dbg> modem_hl7800: sock_read: Sock 1 RX done
[00:00:26.521,118] <dbg> modem_hl7800: hl7800_rx: UNHANDLED RX
                                       2b 4b 55 44 50 5f 52 43  56 3a 20 22 38 2e 38 2e |+KUDP_RC V: "8.8.
                                       38 2e 38 22 2c 35 33                             |8.8",53          
[00:00:26.521,148] <dbg> modem_hl7800: sockreadrecv_cb_work: Sock 1 RX CB (size: 72)
[00:00:26.521,179] <dbg> net_sock: zsock_received_cb: (0x200045a0): ctx=0x20004cac, pkt=0x2001c410, st=0, user_data=0
[00:00:26.634,796] <err> net_dns_dispatcher: recv failed on IPv4 socket (134)
[00:00:26.634,826] <err> net_dns_dispatcher: DNS recv error (-134)
```

Before the above patch there was an issue in `sock_get_offload_pkt_src_addr`:
```
[00:00:26.281,433] <dbg> net_sock: zsock_received_cb: (0x200045a0): ctx=0x20004cac, pkt=0x2001c410, st=0, user_data=0
[00:00:26.395,782] <dbg> net_sock: zsock_recv_dgram: (0x200039e0): sock_get_offload_pkt_src_addr -134
[00:00:26.395,843] <err> net_dns_dispatcher: recv failed on IPv4 socket (134)
[00:00:26.395,843] <err> net_dns_dispatcher: DNS recv error (-134)
```

rerickson1, 2024-07-08:
Address family was not set with `net_pkt_set_remote_address`. That needs to be set separately.

closed 2024-07-10 by nashif