zephyrproject-rtos/zephyr #75402: picolibc: sigevent missing fields

cfriedt, 2024-07-03:
**Describe the bug**

In picolibc (both the module and SDK headers, fwict), `struct sigevent` is missing the `sigev_notify_function` and `sigev_notify_attributes` fields.

This ticket was opened to track the upstream bug and ensure the fix makes its way into Zephyr's picolibc module and the SDK prior to release.

https://github.com/picolibc/picolibc/issues/754

Please also mention any information which could help others to understand
the problem you're facing:
 - What target platform are you using? All
 - What have you tried to diagnose or workaround this issue? Found bugs while running the testsuite
 - Is this a regression? Yes. Please see the picolibc bug.
 - ...

**To Reproduce**
Steps to reproduce the behavior:
1. move signal.h to the minimal libc from posix
2. build just about any POSIX test suite
4. See error about missing fields.

**Expected behavior**
Tests should pass. Assigning a value to `sigev_notify_function` or `sigev_notify_attributes` should pass

**Impact**
It's a showstopper for fixing additional bugs in Zephyr or for using POSIX features from picolibc.

**Logs and console output**

**Environment (please complete the following information):**
 - OS: all
 - Toolchain (e.g Zephyr SDK, ...): all
 - Commit SHA or Version used: `main`

**Additional context**
https://github.com/picolibc/picolibc/issues/754

cfriedt, 2024-07-10:
@keith-packard - wondering if it's best to assign this one to you. What do you think?

keith-packard, 2024-07-10:
Do we need this in LTS? If so, I'll need to do an upstream picolibc release and then we'll need an SDK release and a module update. Is all of that feasible?

keith-packard, 2024-07-10:
I could do a picolibc point release that just addresses that issue, but I'd rather do a full release as there are a few bug fixes which should get included.

keith-packard, 2024-07-10:
synopsis has been running a closed test suite and finding some bugs.

cfriedt, 2024-10-01:
@keith-packard - are you able to confirm this is fixed with recent revisions of picolibc? I think it was likely fixed in 1.8.7 and 1.8.8

keith-packard, 2024-10-01:
Yup, this was fixed in picolibc 1.8.7. As 1.8.8 has been merged to Zephyr, I think this issue can be closed?

closed 2024-10-01 by cfriedt