zephyrproject-rtos/zephyr #75405: esp32s3_devktic sample can't build

overrize, 2024-07-03:
```
(zephyr) USER:~/zephyrproject/zephyr$ west build -b esp32s3_devkitc/esp32s3/procpu samples/hello_world
ERROR: it looks like samples/hello_world is a build directory: did you mean --build-dir samples/hello_world instead?
FATAL ERROR: refusing to proceed without --force due to above error
```
I'm learning to use zephyr on a esp32 board,I found the command to build a hello_world sample in the document cannot work and present these messages.If I do as what it suggested,it would occur these messages.

`(zephyr) USER:~/zephyrproject/zephyr$ west --build-dir -b esp32s3_devkitc/esp32s3/procpu samples/hello_world
usage: west [-h] [-z ZEPHYR_BASE] [-v] [-V] <command> ...
west: unknown command "esp32s3_devkitc/esp32s3/procpu"; workspace /home/rexcon/zephyrproject does not define this extension command -- try "west help"
`
I don't know how to solve these problem.

ASerbinski, 2024-07-03:
Your first try should work. Your second try is definitely wrong for a number of reasons;
1) That's not a build directory (something must be messed up for it to think it is),
2) --build-dir requires a space followed by a parameter, which is the path to the build directory, hence its suggestion of "--build-dir samples/hello_world"

Please show the output of running ```ls -laR samples/hello_world/```

overrize, 2024-07-04:
> Your first try should work. Your second try is definitely wrong for a number of reasons;
>
>     1. That's not a build directory (something must be messed up for it to think it is),
>
>     2. --build-dir [...]

after running` ls -laR samples/hello_world/`shows 
`samples/hello_world/:
total 60
drwxrwxr-x  5 rexcon rexcon 4096 7月   4 00:04 .
drwxrwxr-x 24 rexcon rexcon 4096 7月   2 23:30 ..
-rw-rw-r--  1 rexcon rexcon 4666 7月   3 23:58 CMakeCache.txt
drwxrwxr-x  3 rexcon rexcon 4096 7月   3 23:58 CMakeFiles
-rw-rw-r--  1 rexcon rexcon  194 7月   2 23:30 CMakeLists.txt
drwxrwxr-x  2 rexcon rexcon 4096 7月   3 23:53 Kconfig
-rw-rw-r--  1 rexcon rexcon   15 7月   2 23:30 prj.conf
-rw-rw-r--  1 rexcon rexcon  610 7月   2 23:30 README.rst
-rw-rw-r--  1 rexcon rexcon  322 7月   2 23:30 sample.yaml
drwxrwxr-x  2 rexcon rexcon 4096 7月   2 23:30 src
-rw-rw-r--  1 rexcon rexcon  127 7月   3 23:58 sysbuild_modules.txt
-rw-rw-r--  1 rexcon rexcon 5096 7月   3 23:58 zephyr_modules.txt
-rw-rw-r--  1 rexcon rexcon  748 7月   3 23:58 zephyr_settings.txt

samples/hello_world/CMakeFiles:
total 16
drwxrwxr-x 3 rexcon rexcon 4096 7月   3 23:58 .
drwxrwxr-x 5 rexcon rexcon 4096 7月   4 00:04 ..
-rw-rw-r-- 1 rexcon rexcon   85 7月   3 23:58 cmake.check_cache
drwxrwxr-x 2 rexcon rexcon 4096 7月   3 23:58 pkgRedirects

samples/hello_world/CMakeFiles/pkgRedirects:
total 8
drwxrwxr-x 2 rexcon rexcon 4096 7月   3 23:58 .
drwxrwxr-x 3 rexcon rexcon 4096 7月   3 23:58 ..

samples/hello_world/Kconfig:
total 20
drwxrwxr-x 2 rexcon rexcon 4096 7月   3 23:53 .
drwxrwxr-x 5 rexcon rexcon 4096 7月   4 00:04 ..
-rw-rw-r-- 1 rexcon rexcon 6168 7月   3 23:58 Kconfig.modules
-rw-rw-r-- 1 rexcon rexcon 2538 7月   3 23:58 Kconfig.sysbuild.modules

samples/hello_world/src:
total 12
drwxrwxr-x 2 rexcon rexcon 4096 7月   2 23:30 .
drwxrwxr-x 5 rexcon rexcon 4096 7月   4 00:04 ..
-rw-rw-r-- 1 rexcon rexcon  204 7月   2 23:30 main.c
`

overrize, 2024-07-04:
It works after I use build -p