zephyrproject-rtos/zephyr #75429: Losing bluetooth connection immediality after L2CAP ethablisement - Bluetooth A2DP (Atom Echo - ESP32-Pico-D4)

xG3nesis, 2024-07-04:
Hello everyone!

I'm working on implementing an A2DP Source using the A2DP source sample for my M5Stack Atom Echo, which is based on the ESP32-Pico-D4 architecture.

Everything seems to function correctly up to the AVDTP/A2DP establishment process. However, after my ESP32 connects to the L2CAP layer and opens channels, it immediately loses the connection, displaying the error message 0x13. I understand this indicates "Remote User Terminated Connection," but I'm encountering the same error message on my Kali VM (target on which i'm attempting to connect).

The goal, after establishing the AVDTP connection, is to discover the available "Stream Endpoints." Unfortunately, I'm currently unable to achieve this.

**Logs and console output**
Below, you'll find the logs and console output from my ESP32:
```uart:~$ ␛[8D␛[J[01:04:08.287,000] <dbg> bt_hci_core: bt_hci_cmd_create: opcode 0x0405 param_len 13
uart:~$ ␛[8D␛[J[01:04:08.287,000] <dbg> bt_hci_core: bt_hci_cmd_create: buf 0x3ffd4318
uart:~$ ␛[8D␛[J[01:04:08.287,000] <dbg> bt_hci_core: bt_hci_cmd_send_sync: buf 0x3ffd4318 opcode 0x0405 len 16
uart:~$ ␛[8D␛[J[01:04:08.287,000] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
uart:~$ ␛[8D␛[J[01:04:08.287,000] <dbg> bt_hci_core: tx_processor: TX process start
uart:~$ ␛[8D␛[J[01:04:08.287,000] <dbg> bt_hci_core: hci_core_send_cmd: fetch cmd
uart:~$ ␛[8D␛[J[01:04:08.287,000] <dbg> bt_hci_core: hci_core_send_cmd: Sending command 0x0405 (buf 0x3ffd4318) to driver
uart:~$ ␛[8D␛[J[01:04:08.287,000] <dbg> bt_hci_core: bt_send: buf 0x3ffd4318 len 16 type 0
uart:~$ ␛[8D␛[J[01:04:08.287,000] <dbg> bt_hci_driver_esp32: bt_esp32_send: buf 0x3ffd4318 type 0 len 16
uart:~$ ␛[8D␛[J[01:04:08.287,000] <dbg> bt_hci_driver_esp32: bt_esp32_send: Final HCI buffer:
                                              01 05 04 0d 03 a9 9a 4c  e0 00 18 cc 02 00 00 00 |.......L ........
                                              01                                               |.
uart:~$ ␛[8D␛[J[01:04:08.287,000] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_hci_driver_esp32: hci_esp_host_rcv_pkt: host packet data:
                                              04 0f 04 00 05 05 04                             |.......
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_hci_driver_esp32: bt_esp_evt_recv: len 4
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_hci_driver_esp32: hci_esp_host_rcv_pkt: Calling bt_recv(0x3ffd3c5c)
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_hci_core: bt_recv_unsafe: buf 0x3ffd3c5c len 6
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_hci_core: hci_cmd_status: opcode 0x0405
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_hci_core: hci_cmd_done: opcode 0x0405 status 0x00 buf 0x3ffd3c5c
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_hci_core: hci_cmd_done: sync cmd released
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_hci_core: bt_tx_irq_raise: kick TX
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_hci_core: tx_processor: TX process start
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_conn: bt_conn_tx_processor: start
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_conn: bt_conn_tx_processor: no connection wants to do stuff
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_hci_core: bt_hci_cmd_send_sync: rsp 0x3ffd4318 opcode 0x0405 len 0
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_conn: bt_conn_set_state: disconnected -> initiating
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_conn: bt_conn_ref: handle 0 ref 1 -> 2
uart:~$ ␛[8D␛[J[01:04:08.288,000] <dbg> bt_conn: bt_conn_unref: handle 0 ref 2 -> 1
uart:~$ ␛[8D␛[JConnection pending
uart:~$ ␛[8D␛[Juart:~$ ␛[8D␛[Juart:~$ ␛[8D␛[J[01:04:11.767,000] <dbg> bt_hci_driver_esp32: hci_esp_host_rcv_pkt: host packet data:
                                              04 12 08 00 03 a9 9a 4c  e0 00 01                |.......L ...     
uart:~$ ␛[8D␛[J[01:04:11.767,000] <dbg> bt_hci_driver_esp32: bt_esp_evt_recv: len 8
uart:~$ ␛[8D␛[J--- 107 messages dropped ---
[01:04:11.767,000] <dbg> bt_hci_driver_esp32: hci_esp_host_rcv_pkt: Calling bt_recv(0x3ffd3c5c)
uart:~$ ␛[8D␛[J--- 39 messages dropped ---
[01:04:11.767,000] <dbg> bt_hci_core: bt_recv_unsafe: buf 0x3ffd3c5c len 10
uart:~$ ␛[8D␛[J[01:04:11.767,000] <dbg> bt_hci_core: rx_work_handler: Getting net_buf from queue
uart:~$ ␛[8D␛[J[01:04:11.767,000] <dbg> bt_hci_core: rx_work_handler: buf 0x3ffd3c5c type 1 len 10
uart:~$ ␛[8D␛[J--- 119 messages dropped ---
[01:04:11.767,000] <dbg> bt_hci_core: hci_event: event 0x12
uart:~$ ␛[8D␛[J--- 103 messages dropped ---
[01:04:11.768,000] <dbg> bt_br: bt_hci_role_change: status 0x00 role 1 addr XX:XX:XX:XX:XX:XX
uart:~$ ␛[8D␛[J[01:04:11.768,000] <dbg> bt_conn: bt_conn_ref: handle 0 ref 1 -> 2
uart:~$ ␛[8D␛[J[01:04:11.768,000] <dbg> bt_conn: bt_conn_unref: handle 0 ref 2 -> 1
uart:~$ ␛[8D␛[J--- 85 messages dropped ---
[01:04:11.789,000] <dbg> bt_l2cap_br: l2cap_br_data_pull: last frag, removing 0x3ffd53a4
uart:~$ ␛[8D␛[J[01:04:11.789,000] <dbg> bt_l2cap_br: l2cap_br_data_pull: chan 0x3ffcc5a0 done
uart:~$ ␛[8D␛[J[01:04:11.789,000] <dbg> bt_conn: bt_conn_tx_processor: pop: cb 0 userdata 0
uart:~$ ␛[8D␛[J--- 142 messages dropped ---
[01:04:11.789,000] <dbg> bt_conn: bt_conn_tx_processor: TX process: conn 0x3ffcbee8 buf 0x3ffd53a4 (last)
uart:~$ ␛[8D␛[J[01:04:11.799,000] <dbg> bt_conn: tx_complete_work: conn 0x3ffcbee8
uart:~$ ␛[8D␛[J[01:04:11.799,000] <dbg> bt_conn: tx_notify: conn 0x3ffcbee8
uart:~$ ␛[8D␛[J--- 56 messages dropped ---
[01:04:11.799,000] <dbg> bt_conn: tx_notify: tx 0x3ffcd1f8 cb 0 user_data 0
uart:~$ ␛[8D␛[J[01:04:11.799,000] <dbg> bt_conn: tx_free: 0x3ffcd1f8
uart:~$ ␛[8D␛[J[01:04:11.838,000] <dbg> bt_conn: send_buf: conn 0x3ffcbee8 buf 0x3ffd4350 len 10 flags 0x02
uart:~$ ␛[8D␛[J--- 11 messages dropped ---
[01:04:11.841,000] <dbg> bt_conn: tx_complete_work: conn 0x3ffcbee8
uart:~$ ␛[8D␛[J[01:04:11.841,000] <dbg> bt_conn: tx_notify: conn 0x3ffcbee8
[... 539 lines cut ...]

	k_work_init_delayable(&discover_pipes, call_discover_seps_handler);

}

int main(void)
{
	int err;

	err = bt_enable(bt_ready);
	if (err) {
		printk("Bluetooth init failed (err %d)\n", err);
	}

	return err;
}
```

**Environment (please complete the following information):**
 - OS : Windows 10
 - Zephyr version : 3.6.99
 - SDK version : zephyr-sdk-0.16.8
 - CMake version : 3.29.5

Thank you in advance for your assistance! 😄

sylvioalves, 2024-07-10:
ESP32 only supports BLE, not BT classic. Converting this from BUG to Enhancement accordingly.

xG3nesis, 2024-07-11:
I must confess, @sylvioalves, your answer has left me a bit underwhelmed. So, I have a trio of puzzlers for you:

1. If BT Classic isn't supported, how come my ESP32-PICO-D4 is happily chatting away via a simple Bluetooth RFCOMM serial connection? Magic?
[image]

2. If BT Classic isn't part of the deal, why not shout it from the rooftops in the Bluetooth documentation?
   - https://docs.zephyrproject.org/latest/connectivity/bluetooth/index.html

3. And lastly, if BT Classic is out of the picture, why didn't your post (updated on the 17th of june) include a neon sign making it clear which Bluetooth flavor (Classic or BLE) we're working with?
   - https://github.com/zephyrproject-rtos/zephyr/issues/29394
[image]

sylvioalves, 2024-07-11:
Hi @xG3nesis, nice to hear that it is working as we have not yet tested it.

Let me tell you the background and reasoning that I consider it is not supported (yet).

When we started working on ESP32xx Bluetooth support in Zephyr, BT classic was still marked as "EXPERIMENTAL". We then decided to only add BLE support. It means we implemented and tested only BLE sample codes (and provide necessary changes to make it work). BT classic support was not handled in any way, neither checking nor testing. Up to now, we have not yet worked on it. That is the reason I initially considered this is not implemented. So it is really interesting to see it is already in place as you mentioned before.

For item 3) of the development overview, you are correct, we could have split BT/BLE in different lines to make it clear.

Anyway, thanks for enlightening me with kindly words about BT classic. I still do not have the solution for the issue though.

sylvioalves, 2024-07-11:
It is also worth mentioning that there are some hidden BT configurations in ESP32 adapter. Those configurations (at least a few) could be related to BT classic. Take a look in this file to evaluate as well:
https://github.com/zephyrproject-rtos/hal_espressif/blob/zephyr/zephyr/esp32/include/bt/esp_bt.h#L156

rftafas, 2024-07-11:
> Up to now, we have not yet worked on it. That is the reason I initially considered this is not implemented. So it is really interesting to see it is already in place as you mentioned before.

I'd like also to add that BT Classic is not on our roadmap either. But, being open source, if the community decides to go for it, we will help as much as possible (same apply to several other subsystems/features).