zephyrproject-rtos/zephyr #75443: Compilation errors in "LVGL common input" on multiple instances of same nodes

rexut, 2024-07-04:
https://github.com/zephyrproject-rtos/zephyr/blob/ec601b89df93ee0054c9317e80974a924b65e332/modules/lvgl/input/lvgl_common_input.c#L76

In the case of multiple instantiation and thus multiple extraction of the macro `LV_DEV_INIT()`, the **missing semicolon after the do-while loop** leads to compilation errors

```
.../zephyr/modules/lvgl/input/lvgl_common_input.c: In function 'lvgl_init_input_devices':
.../zephyr/modules/lvgl/input/lvgl_common_input.c:71:9: error: expected ';' before 'do'
   71 |         do {                                                                       \
      |         ^~
```

For example with such kind of Device Tree snippets:

```
… … …
lvgl_axis_x_keypad: lvgl-axis-x-keypad {
	compatible = "zephyr,lvgl-keypad-input";
	input = <&joystick_axis_x_keys>;
	input-codes = <INPUT_KEY_LEFT INPUT_KEY_RIGHT>;
	lvgl-codes = <LV_KEY_LEFT LV_KEY_RIGHT>;
};
… … … 
lvgl_axis_y_keypad: lvgl-axis-y-keypad {
	compatible = "zephyr,lvgl-keypad-input";
	input = <&joystick_axis_y_keys>;
	input-codes = <INPUT_KEY_UP INPUT_KEY_DOWN>;
	lvgl-codes = <LV_KEY_UP LV_KEY_DOWN>;
};
… … …
```

henrikbrixandersen, 2024-07-04:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

rexut, 2024-07-04:
> Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Thanks for pointing that out - I'm familiar with it. For this very very small bug with a simple one liner patch, I choose to ignore it - I found the template for it a little overloaded.

PR #75444 (2024-07-08): modules: lvgl: fix initialization macro syntax
Adding a semicolon after the do-while loop in the LV_DEV_INIT() macro also supports multiple instantiation.

This resolves #75443.
closed 2024-07-08 by aescolar