zephyrproject-rtos/zephyr #75449: tests: arch: common: stack_unwind: arch.common.stack_unwind.riscv_fp fails on nrf54h20 ppr

nordic-piks, 2024-07-04:
**Describe the bug**
The tests:
- tests/arch/common/stack_unwind/arch.common.stack_unwind.riscv_fp

Observed for
- nrf54h20dk/nrf54h20/cpuppr

**To Reproduce**
Steps to reproduce the behavior:
1. have nrf54h20dk/nrf54h20/cpuppr connected
2. go to your zephyr dir
3. call ./scripts/twister -T tests/arch/common/stack_unwind -p nrf54h20dk/nrf54h20/cpuppr --device-testing --device-serial /dev/ttyACM1 -v --inline-logs
4. See console output with error

**Expected behavior**
Valid console output

**Impact**
Not clear

**Logs and console output**
tests/kernel/interrupt/arch.interrupt.minimallibc:
```
 ERROR   - *** Booting Zephyr OS build v3.7.0-rc2-61-g6700a154be3c ***
 Hello World! nrf54h20dk
 1: func1
 2: func2
 3: func1
 4: func2
 5: func1
 6: func2
 E:      a0: 00000003    t0: 00000000
 E:      a1: 2fc05a7b    t1: 00000000
 E:      a2: 2fc07088    t2: 00000009
 E:      a3: 0000000a
 E:      a4: 00000000
 E:      a5: 00000004
 E:      sp: 2fc07094
 E:      ra: 2fc00f1c
 E:    mepc: 2fc00f2a
 E: mstatus: 00001880
 E:
 E:      s0: 2fc062f4
 E:      s1: 00000006
 E:
 E: call trace:
 E:       0:
```

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.1 LTS
 - Toolchain Zephyr SDK 0.16.5
 - Commit SHA or Version used: v3.7.0-rc2-61-g6700a154be3c

katgiadla, 2024-08-08:
The similar symptom occurs for `tests/arch/common/stack_unwind/arch.common.stack_unwind.symtab` for `nrf54l15pdk/nrf54l15/cpuflpr` on version `v3.7.0-679-g3240b034feee`

```
*** Booting Zephyr OS build v3.7.0-679-g3240b034feee ***
Hello World! nrf54l15pdk
1: func1
2: func2
3: func1
4: func2
5: func1
6: func2
E:      a0: 00000003    t0: 00000000
E:      a1: 2003170f    t1: 50104000
E:      a2: 20032e88    t2: 00000009
E:      a3: 00000000
E:      a4: 00000000
E:      a5: 00000004
E:      sp: 20032e90
E:      ra: 20028838
E:    mepc: 20028846
E: mstatus: 00001880
E: 
E:      s0: 200320f8
E:      s1: 00000006
E: 
E: call trace:
E:       0: ra: 20028846 [func2+0x30]
E:       1: ra: 20028838 [func2+0x22]
E: 
E: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
E: Current thread: 0x20031f10 (main)
E: Halting system
```

nordic-piks, 2024-08-30:
nrf_triage_done
NRFX-6328

nordic-piks, 2024-09-04:
Cannot reproduce

closed 2024-09-04 by nordic-piks