zephyrproject-rtos/zephyr #75455: NVS: Possibility of Continuously Running Garbage Collection

jtyao, 2024-07-04:
I've been studying the NVS code and find it simple and straightforward, which is great. However, I have a concern: when performing garbage collection on a sector, if no valid ATE is found, then gc_addr & ADDR_OFFS_MASK will keep increasing and may even exceed sector_size. For a sector with no valid ATE, it should be directly erased. The problematic code snippet is as follows:
`	if (nvs_close_ate_valid(fs, &close_ate)) {
		gc_addr &= ADDR_SECT_MASK;
		gc_addr += close_ate.offset;
	} else {
		rc = nvs_recover_last_ate(fs, &gc_addr);
		if (rc) {
			return rc;
		}
	}`
I suggest making the following modification:
`diff --git a/subsys/fs/nvs/nvs.c b/subsys/fs/nvs/nvs.c
index da35fae89e2..bbbc2efeb36 100644
--- a/subsys/fs/nvs/nvs.c
+++ b/subsys/fs/nvs/nvs.c
@@ -661,6 +661,10 @@ static int nvs_gc(struct nvs_fs *fs)
                if (rc) {
                        return rc;
                }
+
+               if (gc_addr == (sec_addr + fs->sector_size - ate_size)) {
+                       goto gc_done;
+               }
        }

        do {`

henrikbrixandersen, 2024-07-04:
Please submit this proposal as a pull request instead.

Laczen, 2024-07-04:
@jtyao thank you for reporting this.

It might be that this is not happening: when there is no valid ate in a sector the call to `nvs_recover_last_ate()` wil stop at the beginning of the sector. The subsequent call to `nvs_prev_ate()` will increment the address again until the `gc_prev_addr` is equal to the stop condition of the `do {}` loop.

If you would like to make sure that this is the case one could create a 3 sector nvs system. After the nvs system is mounted the third sector could be filled with `~erase_value` (an invalid value). When data is written to the system the first gc call should show the error.

jtyao, 2024-07-05:
> @jtyao thank you for reporting this.
>
> It might be that this is not happening: when there is no valid ate in a sector the call to `nvs_recover_last_ate()` wil stop at the beginning of the sector. The [...]

Thank you for your response. After taking a closer look at the implementation of nvs_recover_last_ate(), I realized that it modifies the gc_addr even if no valid ATE is found. Therefore, the issue I was concerned about won't actually occur. Thanks again!

closed 2024-07-05 by jtyao