zephyrproject-rtos/zephyr #75488: kconfig: Error in Kconfig options MEMC_STM32

miggazElquez, 2024-07-05:
The Kconfig option `MEMC_STM32` is not set correctly:

The option is defined in two files:

-  `drivers/memc/Kconfig.stm32`, wich depends on
    - `MEMC`
    - `DT_HAS_ST_STM32_FMC_ENABLED`
 - `soc/st/stm32/Kconfig.defconfig`, wich depends on
     - `MEMC`
     - `SOC_FAMILY_STM32`

So, if you have `CONFIG_MEMC=y` in your Kconfig options and you are on a STM32 SoC, `CONFIG_MEMC_STM32` will be enabled, even if there is no STM32 FMC enabled.

This Kconfig option causes the driver for the STM32 FMC to be compiled, regardless of the presence of an enabled node for the FMC. However, the driver fails to compile if there is no FMC node in the devicetree. So, if you compile a project with CONFIG_MEMC=y on a board with an STM32 SoC and no enabled FMC, the build will fail.

**To Reproduce**
Add `CONFIG_MEMC=y` in an application.
Build on for a board with a STM32 SoC without enabling the fmc
The build fails

**Expected behavior**
It shouldn't try to compile the driver for the fmc, and the build should work.

miggazElquez, 2024-07-05:
Fixed by https://github.com/zephyrproject-rtos/zephyr/pull/75386

henrikbrixandersen, 2024-07-05:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md (e.g. we have no indication of which revision of Zephyr, you are reporting this against).