zephyrproject-rtos/zephyr #75492: doc: Documentation issue in 'samples/fuel_gauge/max17048/README'

Goyalrahul1516, 2024-07-05:
**Unknown unit of time to empty parameter**

Interfaced MAX17048 sensor with nRF52832 dev kit, got four ouputs
1. time to empty 128
2. time to full 0
3. charge 10%
4. Voltage 3690
But i don't know the units of time to empty 

<< You may also want to update the automatically generated issue title above. >>

**Environment**

* Page: `samples/fuel_gauge/max17048/README`
* Version: main
* SHA-1: 23807100208e988e77b4fc3673ca75b8af4224a3

kartben, 2024-07-05:
Please just refer to the API documentation. It's supposed to be expressed in minutes 
https://docs.zephyrproject.org/latest/hardware/peripherals/fuel_gauge.html#c.fuel_gauge_prop_type.FUEL_GAUGE_RUNTIME_TO_EMPTY

kartben, 2024-07-05:
BTW, it would be most welcome if you're interested in submitting a pull request to update the sample so that the units are included in the console output.

Goyalrahul1516, 2024-07-06:
> Please just refer to the API documentation. It's supposed to be expressed in minutes https://docs.zephyrproject.org/latest/hardware/peripherals/fuel_gauge.html#c.fuel_gauge_prop_type.FUEL_GAUGE_RUNTIME_TO_EMPTY

I just read the documentation, time to discharge unit is resolved but now the doc shows that FUEL_GAUGE_VOLTAGE is shown in micro Volts, the problem is the battery i connected has a rating of 3.6V but the output shows 3690 uV which means 3.69 mV. So am i getting something wrong here? :(

aaronemassey, 2024-07-06:
> I just read the documentation, time to discharge unit is resolved but now the doc shows that FUEL_GAUGE_VOLTAGE is shown in micro Volts, the problem is the battery i connected has a rating of 3.6V [...]

If understand [this datasheet](https://www.analog.com/media/en/technical-documentation/data-sheets/max17048-max17049.pdf) correctly, it looks like that falls well within error for measurement.

Goyalrahul1516, 2024-07-06:
I think there has been a misunderstanding, i said my battery was 3.6V but the output shows 3.69mV according to the documentation. So i just wanted to clear that up. 

Also no problem, Family comes before everything, take your time this can wait.

aaronemassey, 2024-07-07:
My apologies for the misunderstanding.

Looking at [32b27384a68f250a1](https://github.com/zephyrproject-rtos/zephyr/commit/32b27384a68f250a1279246dd50a9455eba86d81) it looks like you're completely correct that that MAX17048 was missed during a conversion to micro-volts. I'm honestly shocked I missed that conversion. Thank you for catching this.

That is a breaking change for current users, but it's something that should've occurred a while ago with a simple short fix. I need to setup my Chromebook I have with me for Zephyr development (should be pretty quick) but if you want this resolved faster you're more than welcome to push a PR to change [here](https://github.com/zephyrproject-rtos/zephyr/blob/a912bb6f14271a921596bd72eb8a1f1f2f11402f/drivers/fuel_gauge/max17048/max17048.c#L93). Current was never implemented so that does not need to be adjusted.

Either way, a PR for this should be pushed this week. Let me know if you want to do it, otherwise I'll go ahead and do it. Afterwards you can cherry-pick it to your Zephyr for or update to top-of-tree.

Goyalrahul1516, 2024-07-08:
I am going to put a PR now, let me know if any further changes are required. I tested the changes and got 7 digit output across voltage. To convert to Voltage just need to divide by 1e6.

Goyalrahul1516, 2024-07-08:
I apologize for making the same PR twice, i am new to git and github and previous CI made me confused so i just made the changes and created a new PR.

aaronemassey, 2024-07-08:
> I apologize for making the same PR twice, i am new to git and github and previous CI made me confused so i just made the changes and created a new PR.

No worries! I appreciate you making the PR!

PR #75573 (2024-07-10): Samples: max17048: updated voltage output and added units 
present sample shows no unit in output.
sample shows voltage in mV.
updated sample shows units.
updated ssample shows voltage in uV.
fixes #75492
closed 2024-07-10 by aescolar