zephyrproject-rtos/zephyr #75504: STM32H750B-dk fails sdram test provided by zephyr

MaazSk, 2024-07-05:
I am using stm32h750b-dk board and whenever I am running test code provided by zephyr in `zephyr/tests/drivers/memc/ram` the code passes for sram1 and sram2 but always fails for sdram2 even though the board has one sdram2. The test results are as follows

[image]

henrikbrixandersen, 2024-07-05:
Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

MaazSk, 2024-07-05:
> Please use our bug template when reporting bugs. You need to edit this issue to include the information requested in https://github.com/zephyrproject-rtos/zephyr/blob/main/.github/ISSUE_TEMPLATE/001_bug_report.md

Done

FRASTM, 2024-07-08:
The FMC of the stm32h750b_dk for the SDRAM controller is configured with 16-bit data bus width STM32_FMC_SDRAM_MWID_16
Due to the SDRAM2 memory organization, the test_ram_rw does not apply as it is.
but writing the SDRAM2 with
*(__IO uint32_t*) (0xd0000000 + 0 + 4*i) = i;
and reading back in a aRxBuffer[uwIndex] = *(__IO uint32_t*) (0xd0000000 + 0 + 4*i);
--> matching

MaazSk, 2024-07-08:
@FRASTM, even after making these changes I am unable to write to the memory where as I can see that I am able to read from that memory

FRASTM, 2024-07-08:
> @FRASTM, even after making these changes I am unable to write to the memory where as I can see that I am able to read from that memory

If writing is not possible, could you try to disable the write protection before the HAL_SDRAM_Init() in the memc_stm32_sdram_init() function of the  drivers/memc/memc_stm32_sdram.c
```patch
diff --git a/drivers/memc/memc_stm32_sdram.c b/drivers/memc/memc_stm32_sdram.c
index d6b7891fe13..1b7d2a740e4 100644
--- a/drivers/memc/memc_stm32_sdram.c
+++ b/drivers/memc/memc_stm32_sdram.c
@@ -46,6 +46,8 @@ static int memc_stm32_sdram_init(const struct device *dev)
 		sdram.State = HAL_SDRAM_STATE_RESET;
 		memcpy(&sdram.Init, &config->banks[i].init, sizeof(sdram.Init));
 
+		sdram.Init.WriteProtection = FMC_SDRAM_WRITE_PROTECTION_DISABLE;
+
 		(void)HAL_SDRAM_Init(
 			&sdram,
 			(FMC_SDRAM_TimingTypeDef *)&config->banks[i].timing);
```

MaazSk, 2024-07-08:
I did this and also made the following changes in the device tree
```
&fmc{
    sdram{
				
				bank@1 {
					reg = < 0x1 >;
					st,sdram-control = < STM32_FMC_SDRAM_NC_8
                                STM32_FMC_SDRAM_NR_12
                                STM32_FMC_SDRAM_MWID_16
                                STM32_FMC_SDRAM_NB_4
                                STM32_FMC_SDRAM_CAS_3
                                STM32_FMC_SDRAM_SDCLK_PERIOD_2
                                STM32_FMC_SDRAM_RBURST_ENABLE
                                STM32_FMC_SDRAM_RPIPE_0>;
					st,sdram-timing = < 0x2 0x7 0x4 0x7 0x2 0x2 0x2 >;
				};
			};
};
```
but still it is not working

FRASTM, 2024-07-09:
Ok, I see a wrong value in the ModeRegisterDefinition that could explain the error.

MaazSk, 2024-07-09:
@FRASTM can you please help me what should be the value in the ModeRegisterDefinition?

FRASTM, 2024-07-09:
> @FRASTM can you please help me what should be the value in the ModeRegisterDefinition? Also, is bank1 correct?
Set mode-register = <0x230>;  in the DTS of the stm32h750b_dk:

```patch
diff --git a/boards/st/stm32h750b_dk/stm32h750b_dk.dts b/boards/st/stm32h750b_dk/stm32h750b_dk.dts
index 2c3f69de188..f124c581db6 100644
--- a/boards/st/stm32h750b_dk/stm32h750b_dk.dts
+++ b/boards/st/stm32h750b_dk/stm32h750b_dk.dts
@@ -208,7 +208,7 @@
 		status = "okay";
 		power-up-delay = <100>;
 		num-auto-refresh = <8>;
-		mode-register = <0x220>;
+		mode-register = <0x230>;
 		refresh-rate = <0x603>;
 		bank@1 {
 			reg = <1>;
```

MaazSk, 2024-07-09:
By making the suggested changes it is working, doing some more tests will update in some time.
Thank you!

MaazSk, 2024-07-11:
Hi @FRASTM,
Sorry for the delay in response I tried the code it passes but if I reset the device constantly it fails the tests.
I am attaching the test result 
Test passed
[image]
Test failed
[image]

FRASTM, 2024-07-15:
> Hi @FRASTM,
That's not the case with my version. (opencod or stm32cube programmer as runner) The testcase always passed after cold or warm reset

MaazSk, 2024-07-15:
After making a small change it is working for me 
```
st,sdram-control = < STM32_FMC_SDRAM_NC_8
                                STM32_FMC_SDRAM_NR_12
                                STM32_FMC_SDRAM_MWID_16
                                STM32_FMC_SDRAM_NB_4
                                STM32_FMC_SDRAM_CAS_3
                                STM32_FMC_SDRAM_SDCLK_PERIOD_2
                                STM32_FMC_SDRAM_RBURST_ENABLE
                                STM32_FMC_SDRAM_RPIPE_1>;
```

I changed `STM32_FMC_SDRAM_RPIPE` value from `STM32_FMC_SDRAM_RPIPE_0` to `STM32_FMC_SDRAM_RPIPE_1`

FRASTM, 2024-07-16:
> After making a small change it is working for me
>
> ```
> st,sdram-control = < STM32_FMC_SDRAM_NC_8
>                                 STM32_FMC_SDRAM_NR_12 [...]

That can be added to the change even if I do not really catch the impact of this parameter (except that it fixes your issue)
Do you run at another sysclock than the DT_FREQ_M(480) ?

MaazSk, 2024-07-16:
I'll check the clock frequency and update on the same. Apart from that there are couple of boards that will be arriving soon so will test on them as well

MaazSk, 2024-07-17:
Hi,
I am using zephyr tests directly and I have not touched the clock settings. Even though can you guide me how to set the clock settings I will set them to what you have and then I can test again.

FRASTM, 2024-07-18:
> Hi, I am using zephyr tests directly and I have not touched the clock settings. Even though can you guide me how to set the clock settings I will set them to what you have and then I can test again.

You can change the clock setting new new properties in the &rcc node (bus clocks) or even the &pll configuration of the stm32h750b_dk.dts. Use the [stm32h750b_dk.dts](https://www.st.com/en/development-tools/stm32cubemx.html) to help defining the right combination.

MaazSk, 2024-07-26:
The changes fixed the issue. Closing it thank you!

closed 2024-07-26 by MaazSk