zephyrproject-rtos/zephyr #75514: cst816s not working with [Waveshare 1.69inch Touch LCD Module 1](https://www.waveshare.com/wiki/1.69inch_Touch_LCD_Module)

ljunquera, 2024-07-05:
**Describe the bug**
I am using a [Waveshare 1.69inch Touch LCD Module 1](https://www.waveshare.com/wiki/1.69inch_Touch_LCD_Module) which states it uses a cst816s driver for touch, yet in my testing swipes and button presses are not being recognized properly. I don't see any one (1) events which seem to indicate a release.

There are numerous examples for other platforms using this Waveshare: Raspberry Pi Pico User Guide, ESP32S3 User Guide, STM32 User Guide, and Arduino User Guide. I was looking to create one for Zephyr. They have samples for each and [cst816 reference documentation](https://files.waveshare.com/upload/c/c2/CST816S_register_declaration.pdf).

Please also mention any information which could help others to understand
the problem you're facing:
 - I am using [nRF5340 Development Kit](https://www.nordicsemi.com/Products/Development-hardware/nRF5340-DK)
 - printk statements from within the cst816s driver
 - not regression

**To Reproduce**
Steps to reproduce the behavior:
1. I have changed the [lvgl sample to test for swipes](https://github.com/ljunquera/lvgl_swipe).
2. I have added printk statements to the cst816 driver:
```
	if (pressed) {
		printk("pressed event: %d, row: %d, col: %d\n", event, row, col);
		input_report_abs(dev, INPUT_ABS_X, col, false, K_FOREVER);
		input_report_abs(dev, INPUT_ABS_Y, row, false, K_FOREVER);
		input_report_key(dev, INPUT_BTN_TOUCH, 1, true, K_FOREVER);
	} else {
		printk("not pressed event: %d, row: %d, col: %d\n", event, row, col);
		input_report_key(dev, INPUT_BTN_TOUCH, 0, true, K_FOREVER);
	}
```
 3. In the driver I have `#define CST816S_CHIP_ID 0xB5` and commented out `//#define CST816S_CHIP_ID 0xB4` which is a recent change.

**Expected behavior**
- I expect a left swipe to register as a left swipe, right as a right, top as a top, bottom as a bottom.
- I expect a button press to register as a button press.

**Impact**
Show stopper currently

**Logs and console output**
This is a button press (with printk in interrupt):
```
cst816s_isr_handler
cst816s_isr_handler
pressed event: 0, row: 86, col: 109
pressed event: 0, row: 86, col: 109
cst816s_isr_handler
cst816s_isr_handler
pressed event: 2, row: 86, col: 109
pressed event: 2, row: 86, col: 109
cst816s_isr_handler
cst816s_isr_handler
pressed event: 2, row: 86, col: 109
[00:00:05.567,932] <err> cst816s: Could not read x data
```

This is a swipe from the left most of the screen to the right in the midpoint vertically of a 240x280 screen (removed the pintk from interrup):
```
*** Booting nRF Connect SDK v3.5.99-ncs1-1 ***
pressed event: 0, row: 129, col: 20
pressed event: 0, row: 129, col: 20
pressed event: 2, row: 129, col: 20
pressed event: 2, row: 129, col: 20
pressed event: 2, row: 129, col: 20
pressed event: 2, row: 129, col: 20
pressed event: 2, row: 129, col: 20
pressed event: 2, row: 129, col: 20
pressed event: 2, row: 129, col: 20
pressed event: 2, row: 129, col: 20
pressed event: 0, row: 189, col: 200
pressed event: 0, row: 189, col: 200
pressed event: 2, row: 189, col: 200
pressed event: 2, row: 189, col: 200
pressed event: 2, row: 189, col: 201
pressed event: 2, row: 189, col: 201
pressed event: 2, row: 188, col: 203
pressed event: 2, row: 188, col: 203
pressed event: 2, row: 187, col: 205
pressed event: 2, row: 187, col: 205
[00:00:01.794,952] <dbg> app: on_lvgl_screen_gesture_event_callback: Gesture event detected 12
[00:00:01.794,982] <dbg> app: handle_screen_gesture: BOTTOM gesture detected
```

**Environment (please complete the following information):**
 - OS: MacOS
 - Toolchain Zephyr

fabiobaltieri, 2024-07-08:
I don't have any hardware to check these, joelguittet@ could you verify that release events work on your boards?

@ljunquera that `[00:00:05.567,932] <err> cst816s: Could not read x data` may be the handler failing on the release event, can you try to figure out more about the error? Is it a NACK from the device?

ljunquera, 2024-07-08:
I can send you a device if you want.

fabiobaltieri, 2024-07-08:
> I can send you a device if you want.

I'd rather you send me a PR with the fix instead. :-)

ljunquera, 2024-07-09:
How can I get you more information? I can use a logic analyzer or add more debug statements?

fabiobaltieri, 2024-07-11:
Sure grab a trace or add some prints, printing the i2c error code in the log could be a start. That said, this is a contributed driver so you'll have to contribute the fix yourself (or find some other contributor to do it).

ljunquera, 2024-07-26:
@fabiobaltieri do you know anybody who would be interested in contracting to do this?

ljunquera, 2024-07-26:
[image]

ljunquera, 2024-07-26:
[image]

ljunquera, 2024-07-26:
Here are some testing results:
[Testing Results 1.pdf](https://github.com/user-attachments/files/16391168/Testing.Results.1.pdf)

ljunquera, 2024-07-26:
Here are some testing results:
[image]

| Gesture | Results |
|---------|--------|
| 1 | pressed event: 0, row: 33, col: 52<br>pressed event: 2, row: 33, col: 52<br>pressed event: 2, row: 33, col: 52<br>pressed event: 2, row: 33, col: 52<br>pressed event: 2, row: 33, col: 52<br>pressed event: 2, row: 33, col: 52<br>pressed event: 2, row: 33, col: 52<br>pressed event: 2, row: 33, col: 52<br>pressed event: 2, row: 33, col: 52<br>pressed event: 2, row: 33, col: 52<br>pressed event: 0, row: 278, col: 69<br>pressed event: 2, row: 278, col: 69<br>pressed event: 2, row: 278, col: 69<br>pressed event: 2, row: 278, col: 69<br>pressed event: 2, row: 278, col: 69<br>pressed event: 2, row: 278, col: 69<br>pressed event: 2, row: 278, col: 69<br>pressed event: 2, row: 278, col: 69<br>pressed event: 2, row: 278, col: 69<br>pressed event: 2, row: 278, col: 68 |
| 2 |  pressed event: 0, row: 33, col: 118<br>pressed event: 2, row: 33, col: 118<br>pressed event: 2, row: 33, col: 118<br>pressed event: 2, row: 33, col: 118<br>pressed event: 2, row: 33, col: 118v<br>pressed event: 2, row: 33, col: 118<br>pressed event: 2, row: 33, col: 118<br>pressed event: 2, row: 33, col: 118<br>pressed event: 2, row: 33, col: 118<br>pressed event: 2, row: 33, col: 118<br>pressed event: 0, row: 278, col: 104<br>pressed event: 2, row: 278, col: 104<br>pressed event: 2, row: 278, col: 104<br>pressed event: 2, row: 278, col: 104<br>pressed event: 2, row: 278, col: 104<br>pressed event: 2, row: 278, col: 104<br>pressed event: 2, row: 278, col: 104<br>pressed event: 2, row: 278, col: 104<br>pressed event: 2, row: 278, col: 104<br>pressed event: 2, row: 278, col: 104 |
| 3 |  pressed event: 0, row: 51, col: 183<br>pressed event: 2, row: 51, col: 183<br>pressed event: 2, row: 51, col: 183<br>pressed event: 2, row: 51, col: 183<br>pressed event: 2, row: 51, col: 183<br>pressed event: 2, row: 51, col: 183<br>pressed event: 2, row: 51, col: 183<br>pressed event: 2, row: 51, col: 183<br>pressed event: 2, row: 51, col: 183<br>pressed event: 2, row: 51, col: 183<br>pressed event: 0, row: 278, col: 188<br>pressed event: 2, row: 278, col: 188<br>pressed event: 2, row: 278, col: 188<br>pressed event: 2, row: 278, col: 188<br>pressed event: 2, row: 278, col: 188<br>pressed event: 2, row: 278, col: 188<br>pressed event: 2, row: 278, col: 188<br>pressed event: 2, row: 278, col: 188<br>pressed event: 2, row: 278, col: 188<br>pressed event: 2, row: 278, col: 188 |
| 4 | pressed event: 0, row: 151, col: 218<br>pressed event: 2, row: 151, col: 218<br>pressed event: 2, row: 151, col: 218<br>pressed event: 2, row: 151, col: 218<br>pressed event: 2, row: 151, col: 218<br>pressed event: 2, row: 151, col: 218<br>pressed event: 2, row: 151, col: 218<br>pressed event: 2, row: 151, col: 218<br>pressed event: 2, row: 151, col: 218<br>pressed event: 2, row: 150, col: 218<br>pressed event: 0, row: 94, col: 157<br>pressed event: 2, row: 94, col: 157<br>pressed event: 2, row: 94, col: 157<br>pressed event: 2, row: 94, col: 157<br>pressed event: 2, row: 94, col: 157<br>pressed event: 2, row: 94, col: 157<br>pressed event: 2, row: 94, col: 157<br>pressed event: 2, row: 93, col: 157<br>pressed event: 2, row: 92, col: 157<br>pressed event: 2, row: 91, col: 156<br>pressed event: 0, row: 81, col: 80<br>pressed event: 2, row: 81, col: 80<br>pressed event: 2, row: 81, col: 80<br>pressed event: 2, row: 81, col: 80<br>pressed event: 2, row: 81, col: 80<br>pressed event: 2, row: 81, col: 80<br>pressed event: 2, row: 80, col: 79<br>pressed event: 2, row: 79, col: 78<br>pressed event: 2, row: 78, col: 77<br>pressed event: 2, row: 77, col: 75 |
| 5 |  pressed event: 0, row: 160, col: 214<br>pressed event: 2, row: 160, col: 214<br>pressed event: 2, row: 160, col: 214<br>pressed event: 2, row: 160, col: 214<br>pressed event: 2, row: 160, col: 214<br>pressed event: 2, row: 160, col: 214<br>pressed event: 2, row: 160, col: 214<br>pressed event: 2, row: 160, col: 214<br>pressed event: 2, row: 160, col: 214<br>pressed event: 2, row: 160, col: 214<br>pressed event: 0, row: 122, col: 125<br>pressed event: 2, row: 122, col: 125<br>pressed event: 2, row: 122, col: 125<br>pressed event: 2, row: 122, col: 125<br>pressed event: 2, row: 123, col: 124<br>pressed event: 2, row: 125, col: 123<br>pressed event: 2, row: 127, col: 122<br>pressed event: 2, row: 129, col: 121<br>pressed event: 2, row: 131, col: 120<br>pressed event: 2, row: 133, col: 119<br>pressed event: 0, row: 160, col: 0<br>pressed event: 2, row: 160, col: 0<br>pressed event: 2, row: 160, col: 0<br>pressed event: 2, row: 160, col: 0<br>pressed event: 2, row: 160, col: 0<br>pressed event: 2, row: 160, col: 0<br>pressed event: 2, row: 160, col: 0<br>pressed event: 2, row: 160, col: 0<br>pressed event: 2, row: 160, col: 0<br>pressed event: 2, row: 160, col: 0 | 
| 6 |  pressed event: 0, row: 241, col: 214<br>pressed event: 2, row: 241, col: 214<br>pressed event: 2, row: 241, col: 214<br>pressed event: 2, row: 241, col: 214<br>pressed event: 2, row: 241, col: 214<br>pressed event: 2, row: 241, col: 214<br>pressed event: 2, row: 241, col: 214<br>pressed event: 2, row: 241, col: 214<br>pressed event: 2, row: 241, col: 214<br>pressed event: 2, row: 241, col: 214<br>pressed event: 0, row: 251, col: 155<br>pressed event: 2, row: 251, col: 155<br>pressed event: 2, row: 251, col: 155<br>pressed event: 2, row: 251, col: 155<br>pressed event: 2, row: 251, col: 155<br>pressed event: 2, row: 251, col: 155<br>pressed event: 2, row: 251, col: 155<br> [... cut ...] : 228, col: 42<br>pressed event: 2, row: 228, col: 42<br>pressed event: 2, row: 228, col: 42 |
| 17 | pressed event: 0, row: 238, col: 154<br>pressed event: 2, row: 238, col: 154<br>pressed event: 2, row: 238, col: 154<br>pressed event: 2, row: 238, col: 154<br>pressed event: 2, row: 238, col: 154<br>pressed event: 2, row: 238, col: 154<br>pressed event: 2, row: 238, col: 154<br>pressed event: 2, row: 238, col: 154<br>pressed event: 2, row: 238, col: 154<br>pressed event: 2, row: 238, col: 154 |

fabiobaltieri, 2024-07-26:
> @fabiobaltieri do you know anybody who would be interested in contracting to do this?

Not really, try asking in #job-posting on http://chat.zephyrproject.org/

ljunquera, 2024-07-27:
It's possible it might be the cst816t. Just a guess but I see that one referenced on one of their pages. [Here is a reference](https://github.com/koendv/cst816t) to a library I found as well.

closed 2024-10-10 by github-actions[bot]